esp8266/Arduino #2118: can we use esp8266/arduino with esp8266-rtos-sdk ?

wuwx, 2016-06-08:
this library seems use esp8266-nonos-sdk
can we use esp8266-rtos-sdk?
if we could, how do it

igrr, 2016-06-08:
Currently this code is heavily dependent on the way non-OS SDK works. So you can not use it with FreeRTOS SDK.

However, the version of Arduino core for the next chip is based on FreeRTOS, and it may be backported to ESP8266 later. There are no plans which I can disclose at this point though.

marvinroger, 2016-06-08:
@igrr do you mean @me-no-dev's version?

igrr, 2016-06-08:
@me-no-dev's version is obviously also based on FreeRTOS, and as such can be backported to the 8266. But I was referring to a different version (not that it matters in the context of this question).

me-no-dev, 2016-06-08:
I have a RTOS version ready for ESP8266, but the current RTOS has issues with C++. If @igrr or someone else there helps, we can have a try :)

malachib, 2016-12-19:
I've had limited success with C++ using [https://github.com/SuperHouse/esp-open-rtos](url).  Not sure how relevant, but there it is =)

stanlaw7, 2017-04-04:
Wonder if @me-no-dev can share more around the specific part of C++ lacked in RTOS version?

hreintke, 2017-04-06:
I had the sming esp framework running on RTOS SDK.
No issues there with c++.

Would also like to have an arduino on top of RTOS.
Is there any work in process available ?

me-no-dev, 2017-04-06:
@hreintke Well I had issues when I attempted this. Other than that I have an Arduino port on RTOS (obviously not working)

hreintke, 2017-04-06:
@me-no-dev 
Checked you repos on github, didn't find it there.
Is that port somewhere available ?

me-no-dev, 2017-04-06:
Nope :) only on my computer. Putting another Arduino for ESP8266 is not a good idea. At least for us. Imagine syncing and supporting two frameworks that do the same thing. I did it as a POC and it did not work (because of C++ not working).

hreintke, 2017-04-06:
Completely understand and agree on your statement on the two frameworks.
Any change for me to get a personal copy to compare to my work on smingRTOS and personal experimenting ?

devyte, 2017-10-17:
Closin per #3655 .

closed 2017-10-17 by devyte