esp8266/Arduino #2120: HTTPClient.setTimout(2000) not taken into account for begin() ?

hallard, 2016-06-08:
#### Hardware
Hardware:		nodemcu ESP12
Core Version:      	latest git

### Description
I think I misunderstood something, with HTTPClient library, I'm calling `http.setTimeout(1000);` it  works once connected to server and when doing a `http.get()`. No pb at all
But whatever value I use if the server is down or not responding `http.begin()` seems to default to 5 seconds

### Settings in IDE

Module:  			NodeMCU V1.0
Flash Size: 		4MB/1MB
CPU Frequency:	160Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		OTA 
Reset Method:		nodemcu

### Sketch

Here my function 

```arduino
bool WS_post(char * host, uint16_t port, char * url)
{
  HTTPClient http;
  bool ret = false;

  // Just in case
  http.setTimeout(2000);

  Serial.printf("[HTTP] %s:%d/%s", host, port, url);

  // configure target server and url
  if ( http.begin(String(host), port, String(url)) )
  {
    // start connection and send HTTP header
    int httpCode = http.GET();

    // httpCode will be negative on error
    if (httpCode > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.printf("[HTTP] GET... code: %d\r\n", httpCode);

      // file found at server
      if (httpCode == HTTP_CODE_OK) {
        String payload = http.getString();
        
        Serial.printf("'%s' (%d ms)\r\n",payload.c_str(), millis()-start);
        ret = true;
      }
    } else {
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }
  } else {
    Serial.printf("[HTTP] Begin... error\r\n");
  }

  http.end();
  return ret;
}
```

### Debug Messages

```
[HTTP] 80.11.xxx.xxx:82//?blablabla
after 5 seconds and not 2
[HTTP] GET... failed, error: connection refused
```

josep112, 2016-07-19:
Hi,Hallard 
You can solve this? I have the same problem

Humancell, 2016-08-24:
I'm curious about the status of this?  Particularly, I'm wanting to know what the default timeout is when using the WiFiClient client; / client.connect() call?  And if I know the default, can I change this?

Likewise (slightly off topic) what is the default timeout on ESPhttpUpdate.update()?  And is this also settable?

mkeyno, 2017-03-02:
for reliable performance or periodic remote HTTPClient   request , I think HTTPClient  should be rewritten in async mode, because  sync stack has no correct idea about status of remote server and if you run the websocket server probably it will be failed

igrr, 2017-05-17:
Will track this in https://github.com/esp8266/Arduino/issues/1420.

closed 2017-05-17 by igrr