esp8266/Arduino #2121: Issue loading files w/SPIFFS (ERR_CONTENT_LENGTH_MISMATCH)

Th3On3Fr33man, 2016-06-08:
### Basic Info

#### Hardware
Hardware:			ESP-8266
Core Version:      	2.2.0

### Description

Alright so I've been looking into this for the past two days and I still feel like I'm getting nowhere.

I recently started using the **SPIFFS File System** for Arduino development on an **HUZZAH ESP8266** like the **FSBrowser.ino** example, and while it's been great in terms of separating code, as my code continues to grow it has not been great in terms of stability.

Ever since I began adding more and more javascript, I began to have errors pop up for various files, whether it's my HTML/CSS/JS, and the primary error I see is **ERR_CONTENT_LENGTH_MISMATCH**.
[image]

```
//File Read for File System
bool handleFileRead(String path)
{
  if(mySerial)  
    mySerial.println("handleFileRead: " + path);
  if(path.endsWith("/")) path += "index.htm";
  String contentType = getContentType(path);
  String pathWithGz = path + ".gz";
  if(SPIFFS.exists(pathWithGz) || SPIFFS.exists(path))
  {

    if(SPIFFS.exists(pathWithGz))
      path += ".gz";
    File file = SPIFFS.open(path, "r");

    if(mySerial)
        mySerial.println("TEST: " + path + " FILE OPEN: " + file.size());

    size_t sent = server.streamFile(file, contentType);

    if(mySerial)
        mySerial.println("TEST: " + path + " SIZE: " + file.size());

    file.close();

    if(mySerial)
        mySerial.println("TEST: " + path + " FILE CLOSE");

    return true;
  } //end if 
  return false;
} //end function handleFileRead
```

So after that happened I began examining the handleFileRead() function that I copied from the FSBrowser.h example and after testing various print statements it seems the hangup occurs within this line:

**size_t sent = server.streamFile(file, contentType);**

```
template<typename T> size_t streamFile(T &file, const String& contentType){
  setContentLength(file.size());
  if (String(file.name()).endsWith(".gz") &&
      contentType != "application/x-gzip" &&
      contentType != "application/octet-stream"){
    sendHeader("Content-Encoding", "gzip");
  }
  send(200, contentType, "");
  return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE);
}
```

So from there, I went into **ESP8266WebServer.h** and found the **streamFile()** function. Once inside this function, I discovered this line:

**return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE);**
[... 1137 lines cut ...]
      } //end else if
      else 
      {
        if(mySerial)
            mySerial.println("Unknown command");
      } //end else
          
      // send data to all connected clients
      webSocket.broadcastTXT(payload, length);
      break;
    case WStype_BIN:
      if(mySerial)
        mySerial.printf("[%u] get binary length: %u\r\n", num, length);
      hexdump(payload, length);

      // echo data back to browser
      webSocket.sendBIN(num, payload, length);
      break;
    default:
      if(mySerial)
        mySerial.printf("Invalid WStype [%d]\r\n", type);
      break;
  } //end switch
} //end function webSocketEvent
```

kiralikbeyin, 2016-06-14:
Hi,

Very nice code. 

I have some suggestions;

1-sometimes mDNS makes trouble , reboot your router,pc and esp ---> because "ws://myword.local" in your webpage kills your connection so try to use;

`var connection = new WebSocket('ws://192.168.1.23:81/', ['arduino']);
`

2-Dont use font css - remove them from header
3-If you will use fonts like font awesome never gz them
4-try to make js files .js.gz 

I hope this helps.

Th3On3Fr33man, 2016-06-16:
@kiralikbeyin  alright so I ended up switching over to that modified version of the library in the meantime since I made this post, and unfortunately I don't think I can apply your current solution directly since it relies on the original library. 

Sorry but is there any chance that we could find a way to apply your solution to my new code? 

Here is the relevant set of functions: 

`// WEB HANDLER IMPLEMENTATION (Modified Server Library Code)
class SPIFFSEditor: public AsyncWebHandler {
  private:
    String _username;
    String _password;
    bool _uploadAuthenticated;
  public:
    SPIFFSEditor(String username=String(), String password=String()):_username(username),_password(password),_uploadAuthenticated(false){}
    bool canHandle(AsyncWebServerRequest *request){
      if(request->method() == HTTP_GET && request->url() == "/edit" && (SPIFFS.exists("/edit.htm") || SPIFFS.exists("/edit.htm.gz")))
        return true;
      else if(request->method() == HTTP_GET && request->url() == "/list")
        return true;
      else if(request->method() == HTTP_GET && (request->url().endsWith("/") || SPIFFS.exists(request->url()) || (!request->hasParam("download") && SPIFFS.exists(request->url()+".gz"))))
        return true;
      else if(request->method() == HTTP_POST && request->url() == "/edit")
        return true;
      else if(request->method() == HTTP_DELETE && request->url() == "/edit")
        return true;
      else if(request->method() == HTTP_PUT && request->url() == "/edit")
        return true;
      return false;
    }

    //Function used to serve webpage to user (Modified Server Library Code)
    void handleRequest(AsyncWebServerRequest *request){
      if(_username.length() && (request->method() != HTTP_GET || request->url() == "/edit" || request->url() == "/list") && !request->authenticate(_username.c_str(),_password.c_str()))
        return request->requestAuthentication();

      if(request->method() == HTTP_GET && request->url() == "/edit"){
        request->send(SPIFFS, "/edit.htm");
      } else if(request->method() == HTTP_GET && request->url() == "/list"){
        if(request->hasParam("dir")){
          String path = request->getParam("dir")->value();
          Dir dir = SPIFFS.openDir(path);
          path = String();
          String output = "[";
          while(dir.next()){
            File entry = dir.openFile("r");
            if (output != "[") output += ',';
            bool isDir = false;
            output += "{\"type\":\"";
            output += (isDir)?"dir":"file";
            output += "\",\"name\":\"";
            output += String(entry.name()).substring(1);
            output += "\"}";
            entry.close();
          }
          output += "]";
          request->send(200, "text/json", output);
          output = String();
        }
        else
          request->send(400);
      } else if(request->method() == HTTP_GET){
        String path = request->url();
        if(path.endsWith("/"))
          path += "index.htm";
        request->send(SPIFFS, path, String(), request->hasParam("download"));
      } else if(request->method() == HTTP_DELETE){
        if(request->hasParam("path", true)){
          ESP.wdtDisable(); SPIFFS.remove(request->getParam("path", true)->value()); ESP.wdtEnable(10);
          request->send(200, "", "DELETE: "+request->getParam("path", true)->value());
        } else
[... 231 lines cut ...]
            mySerial.printf("BodyStart: %u\n", total);
        } //end if
    } //end if
    
    if(mySerial)
    {
        mySerial.printf("%s", (const char*)data);
    } //end if
      
    if(index + len == total)
    {
        if(mySerial)
        {
            mySerial.printf("BodyEnd: %u\n", total);
        } //end if
    } //end if
      
  });
  server.begin();

  //Setup a timer for all WiFi status updates
  statusTimerId = timer.setInterval(5000, sendInfoWifi);
}`

Sorry again and thank you so much for your assistance.

kiralikbeyin, 2016-06-16:
There is nothing to do with esp code. I think all problem is reading a lot of files (size not important) More count more trouble.

Merge all of your js files to only one js file and also css files to one css file this will absolute solution.
(I merged 4 js files to 2 js files and also 5 css to 1 css file)

If you are using font awesome or gly of bootstrap they are raising the count of read files when startup. So don't use them now.(ERR_CONTENT_LENGTH_MISMATCH) +Some links in css code may search other css files or font files in dir, this is why you got this error.

Also try safari,chrome and firefox for console logs, they are giving different logs. Remember to clear browser cache before trusting your code.

+I am not using Async lib but it will work for you.

Th3On3Fr33man, 2016-06-16:
@kiralikbeyin Alright I merged three of my css files into one and I took a gauge library I was using in js and merged it with my general js file. 

However, there's another js file that I've had to include at the end of the file because it references elements in the html and creates elements that load at the start of the page. I'm not sure how to include this with my other js because it has to load later? 

`var voltageGauge = new Gauge({
renderTo    : 'voltageGauge',
width       : 300,
height      : 300,
glow        : false,
units       : 'V',
title       : 'Voltage',
minValue    : 0,
maxValue    : 100,
majorTicks  : ['0','10','20','30','40','50','60','70','80','90','100'],
minorTicks  : 10,
strokeTicks : false,
highlights  : [
{ from : 0, to : 50, color : 'rgba(0,   0, 255, .3)' },
{ from : 50, to : 100, color : 'rgba(255, 0, 0, .3)' }
],
colors      : {
plate: '#222',
majorTicks: '#f5f5f5',
minorTicks: '#ddd',
title: '#fff',
units: '#ccc',
numbers: '#eee',
needle     : {
start : 'rgba(240, 128, 128, 1)',
end : 'rgba(255, 160, 122, .9)',
circle: {
outerStart: '#333',
outerEnd: '#111',
innerStart: '#111',
innerEnd: '#222'
},
shadowUp: false,
shadowDown: false
},
circle: {
shadow: false,
outerStart: '#333',
outerEnd: '#111',
middleStart: '#222',
middleEnd: '#111',
innerStart: '#111',
innerEnd: '#333'
},
valueBox: {
rectStart: '#222',
rectEnd: '#333',
background: '#babab2',
shadow: 'rgba(0, 0, 0, 1)'
}
},
animation : {
delay : 25,
duration: 1000,
fn : 'bounce'
}
});

voltageGauge.draw();`

kiralikbeyin, 2016-06-16:
[image]

Here Jquery before bootstrap,

at the end some custom code.

You can also directly paste js code before body close

```
        </div>
      </div>
    </div>
  </div>
  <script type="text/javascript" src="js/mycode.js"></script>
</body>
```

If you can't merge It is not a rule to use 1 js file, use it like;
```
  <script type="text/javascript" src="js/mycode.js"></script>
  <script type="text/javascript" src="js/mycode2.js"></script>
```

less file more performance.

Th3On3Fr33man, 2016-06-20:
Alright so I seemed to have mostly solved this issue by using the modified library, compressing my code, and removing jQuery entirely. I still get errors every once in a while, but these appear to be unavoidable and everything works most of the time.

@kiralikbeyin I compressed all my scripts into one and posted it at the end of the body as you suggested, which seems to work. Thanks for all your help.

devyte, 2017-10-17:
@Th3On3Fr33man your description starts with ESP8266Webserver, but then later you discuss ESPAsyncWebserver.
Given your original code, and from personal experience, I think your original issue is in ESP8266Webserver.
Is the issue still relevant with 
- latest git
- PR #3362  (lwip2)
?

devyte, 2018-02-28:
Closing due to lack of feedback and age.

closed 2018-02-28 by devyte

air18ru, 2019-04-19:
I have a slightly different problem, but its solution may suit you. I use material design on my web server. This involves downloading large js and css files. The page took a long time to load - up to 7 seconds. I wondered why. And found your issues. I noticed in your message the HTTP_DOWNLOAD_UNIT_SIZE parameter, which is set to 1460 bytes. I thought, why 1460? Perhaps this is to ensure that the data blocks fit well into the protocol data unit (PDU). But the low channel layer is able to split the data into units by itself. And there is no need to do it with a web server. I set the value HTTP_DOWNLOAD_UNIT_SIZE to 16384 bytes. A miracle! The page began to load 4 times faster.
So, you may have a problem because some bytes are lost due to incorrect fragmentation, first by the web server, then by the channel layers.