esp8266/Arduino #2122: how to link static library

ewfian, 2016-06-09:
### Basic Infos

#### Hardware
Hardware:		ESP-8266-12F

### Description

a have a static library(mpl libraries form InvenSense, Inc ) build by [arm-none-eabi-gcc-4.9.3] and i want use it in my project ,i attempt to use but it seems does not work 

[Arduino-IDE-1.5-3rd-party-Hardware-specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification)

```
ARM MPL Libaries -

All libaries compiled using GCC NONE-ARM-EABI 4.9.3 compiler options

1. M0, M0+ Options -

arm-none-eabi-gcc-4.9.3 -DEMPL -DUSE_DMP -DMPU9150 -DREMOVE_LOGGING -DMPL_LOG_NDEBUG=1 -I"C:\eclipseWS\libmplmpu\mpl" -I"C:\eclipseWS\libmplmpu\mllite" -I"C:\eclipseWS\libmplmpu\driver\msp430\bsp\USB_API\USB_Common" -I"C:\eclipseWS\libmplmpu\driver\include" -O2 -Wall -Wa,-adhlns="$@.lst" -c -fmessage-length=0 -mcpu=cortex-m0 -mthumb -g -gdwarf-2

2. M3 Compilers Options -

arm-none-eabi-gcc-4.9.3 -DEMPL -DUSE_DMP -DMPU9150 -DREMOVE_LOGGING -DMPL_LOG_NDEBUG=1 -I"C:\eclipseWS\libmplmpu\mpl" -I"C:\eclipseWS\libmplmpu\mllite" -I"C:\eclipseWS\libmplmpu\driver\msp430\bsp\USB_API\USB_Common" -I"C:\eclipseWS\libmplmpu\driver\include" -O2 -Wall -Wa,-adhlns="$@.lst" -c -fmessage-length=0 -mcpu=cortex-m3 -mthumb -g -gdwarf-2

4. M4 Compiler Options -

arm-none-eabi-gcc-4.9.3 -DEMPL -DUSE_DMP -DMPU9150 -DREMOVE_LOGGING -DMPL_LOG_NDEBUG=1 -I"C:\eclipseWS\libmplmpu\mpl" -I"C:\eclipseWS\libmplmpu\mllite" -I"C:\eclipseWS\libmplmpu\driver\msp430\bsp\USB_API\USB_Common" -I"C:\eclipseWS\libmplmpu\driver\include" -O2 -Wall -Wa,-adhlns="$@.lst" -c -fmessage-length=0 -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -g -gdwarf-2

arm-none-eabi-gcc-4.9.3 -DEMPL -DUSE_DMP -DMPU9150 -DREMOVE_LOGGING -DMPL_LOG_NDEBUG=1 -I"C:\eclipseWS\libmplmpu\mpl" -I"C:\eclipseWS\libmplmpu\mllite" -I"C:\eclipseWS\libmplmpu\driver\msp430\bsp\USB_API\USB_Common" -I"C:\eclipseWS\libmplmpu\driver\include" -O2 -Wall -Wa,-adhlns="$@.lst" -c -fmessage-length=0 -mcpu=cortex-m4 -mthumb -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -g -gdwarf-2

arm-none-eabi-gcc-4.9.3 -DEMPL -DUSE_DMP -DMPU9150 -DREMOVE_LOGGING -DMPL_LOG_NDEBUG=1 -I"C:\eclipseWS\libmplmpu\mpl" -I"C:\eclipseWS\libmplmpu\mllite" -I"C:\eclipseWS\libmplmpu\driver\msp430\bsp\USB_API\USB_Common" -I"C:\eclipseWS\libmplmpu\driver\include" -O2 -Wall -Wa,-adhlns="$@.lst" -c -fmessage-length=0 -mcpu=cortex-m4 -mthumb -g -gdwarf-2
```

### Settings in IDE

Module:  			NodeMCU 1.0(ESP 12 F)
Flash Size: 		4MB
CPU Frequency:	80Mhz

### Debug Messages

```
c:/users/name/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\name\AppData\Local\Temp\build398b4761e2daba412071276124ee61b0.tmp/liblibmplmpu.a(compass_vec_cal.o): Relocations in generic ELF (EM: 40)

c:/users/name/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\name\AppData\Local\Temp\build398b4761e2daba412071276124ee61b0.tmp/liblibmplmpu.a(compass_vec_cal.o): Relocations in generic ELF (EM: 40)

c:/users/name/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\name\AppData\Local\Temp\build398b4761e2daba412071276124ee61b0.tmp/liblibmplmpu.a(compass_vec_cal.o): Relocations in generic ELF (EM: 40)

c:/users/name/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\name\AppData\Local\Temp\build398b4761e2daba412071276124ee61b0.tmp/liblibmplmpu.a(compass_vec_cal.o): Relocations in generic ELF (EM: 40)

c:/users/name/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\name\AppData\Local\Temp\build398b4761e2daba412071276124ee61b0.tmp/liblibmplmpu.a(compass_vec_cal.o): Relocations in generic ELF (EM: 40)

c:/users/name/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\name\AppData\Local\Temp\build398b4761e2daba412071276124ee61b0.tmp/liblibmplmpu.a(compass_vec_cal.o): Relocations in generic ELF (EM: 40)

C:\Users\name\AppData\Local\Temp\build398b4761e2daba412071276124ee61b0.tmp/liblibmplmpu.a: error adding symbols: File in wrong format

collect2.exe: error: ld returned 1 exit status
```

igrr, 2016-06-09:
ESP8266 has an Xtensa lx106 CPU, not an ARM CPU. You can not link binary code generated for ARM into a program for Xtensa.

closed 2016-06-12 by igrr

Saums, 2016-10-26:
Do you know the file extensions for Linux?