esp8266/Arduino #2124: Is there any way to use GPIO06 to GPIO11 on NodeMCU 1.0(ESP8266-12e) module

Naveed-Akhtar, 2016-06-09:
### Basic Infos

#### Hardware
Hardware:			ESP8266-12e
Core Version:      	2.2.0

### Description

I am trying to use the GPIO06 to GPIO11 for simple blink example. The pins are mentioned as in figure: http://learn.acrobotic.com/uploads/esp8266_esp12e_pinout.png.

Can any body help me using these pins for simple blink example? or Is it possible to use these pins as GPIO?

Thanks for your precious time in advance.

WereCatf, 2016-06-10:
They are in use by the internal Flash, so no.

igrr, 2016-06-10:
I think in DIO mode you should be able to use two of them.

torntrousers, 2016-06-10:
I've wondered about this too and would like to have a go. Any help or guidance?

I thought go add a new definition in boards.txt to have an ESP using DIO, was going to copy the NodeMCU 1.0 definition but looking there that already has DIO (nodemcuv2.build.flash_mode=dio), the NodeMCU 0.9 definition has nodemcuv2.build.flash_mode=qio.

Does that mean it should work already when using NodeMCU 1.0? How can I find which are the two GPIO pins freed up when using DIO mode?

igrr, 2016-06-10:
GPIO9 and GPIO10 are shared with SD_DATA2 and SD_DATA3, which are unused in DIO mode. You can find this information in ESP8266 pin map: http://www.esp8266.com/wiki/lib/exe/detail.php?id=esp8266_gpio_pin_allocations&media=pin_functions.png

torntrousers, 2016-06-10:
I just tried a quick hack of soldering an LED between GND and GPIO 9 / 10 on a NodeMCU 1.0 board and running the blink sketch using that. GPIO-10 seems to work fine and blinks the LED, but with GPIO-9 it gives a WDT reset when doing the pinMode(9, OUTPUT).

WereCatf, 2016-06-10:
@igrr @torntrousers GPIO9 and GPIO10 are still physically connected to the SPI-flash. You'd have to remove the RF-shield, physically cut the connection to those pins from the SPI-flash, solder a jumper-wire there to force the chip to DIO-mode permanently and then put the RF-shield back.

igrr, 2016-06-10:
In that case, maybe we need to check the data sheet of the flash chip used in Nodemcu.
I have tested ESP8285 a few days ago (it has built-in flash connected in DOUT mode) and was able to use both 9 and 10.

torntrousers, 2016-06-10:
Interesting that GPIO10 seems to work ok though, even just having the one extra GPIO could be handy sometimes. From the wiring [here](http://smarpl.com/content/esp8266-esp-201-module-freeing-gpio9-and-gpio10) and a flash chip datasheet [here](https://www.google.co.uk/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=0ahUKEwiuwOGRu53NAhXLAcAKHUE8DHsQFgghMAA&url=https%3A%2F%2Fwww.winbond.com%2Fresource-files%2Fw25q40bw%2520revf%2520101113.pdf&usg=AFQjCNEdK4qq_mFAazZq9XtAWDG7QPn07A&sig2=RSeYg77exrGvumnFUscG8Q&cad=rja) it seems like GPIO10 is connected to the flash Write Protect which might only be enabled when certain status bits are also set so maybe that isn't done and so GPIO10 is safe to be used as it is without any hardware mods?

torntrousers, 2016-06-12:
And just in case anyone is wondering does using DIO make things really slow, [here's a test](http://www.esp8266.com/viewtopic.php?p=49057#p49057) to find out.

WereCatf, 2016-06-12:
@torntrousers A rather flawed test, considering that your sketch is also run from the Flash: the slower the access to the Flash the slower the code will also run if it doesn't all fit in the cache.

torntrousers, 2016-06-13:
Hi @WereCatf, does that really matter? Its not trying to come up with anything like a figure on bits per millisecond just to give an idea of the relative difference between DIO and QIO. And thats the point of showing the code - how would you rewrite the sketch?

WereCatf, 2016-06-13:
@torntrousers No, for most needs and purposes I doubt it matters. It was just an observation and may be worth keeping in mind if one does something that is very performance-sensitive.

devyte, 2017-10-17:
Closing this as resolved.

closed 2017-10-17 by devyte