esp8266/Arduino #2128: Changing Wifi Server Port at runtime

thedroid, 2016-06-10:
Is there anyway to change the port that Wifi Server is listening on at run time.

i.e. Initially start the server on a default port, but give the user the ability to change from the default to a port of their choosing at run time?

devyte, 2017-10-17:
It looks to me like assigning a new temp object like below causes a mem leak:
```
ESP8266WebServer server(80);
...
server = ESP8266WebServer(12345);
```

Also, close() doesn't free internal resources.

devyte, 2018-01-12:
This is already implemented. Usage:
```
WiFiServer server(80);
server.begin();
...

server.begin(8080); //closes, then opens again on the new port.
```

Please test latest git.

devyte, 2018-02-07:
Pulled into 2.4.1'as it's already implemented, and closing.

closed 2018-02-07 by devyte