esp8266/Arduino #2129: HTTP_POST request

enzosagretti, 2016-06-11:
Hello!

I am new in esp8266. I am traing to use this library to send from server HTTP_POST request to esp8266, but i cant read parameters(i only can read if i send parameters into url). How can i read parameters from HTTP_POST (parameter in body http request)?

Tanks in advance!

igrr, 2016-06-12:
Could you give an example of request you are sending, what are the headers and payload?

enzosagretti, 2016-06-12:
I report the bug in #2131.

I am sending by HTTP_POST  request:

`http://192.168.0.16/1`

with payload 

`val={"val":"prue"}`

Source message

```
POST /1 HTTP/1.1
 HOST: 192.168.0.16
 content-length: 18
 
 val={"val":"prue"}
```

The error i think is that in "Parsing.cpp" (line 164) compare

else if (headerName == "**C**ontent-**L**ength"

and headerName  in this case is **c**ontent-**l**ength

igrr, 2016-06-12:
Currently ESP8266WebServer will parse POST request body in three cases:
- for file uploads — appropriate callbacks are called, see FSBrowser sample
- if `Content-Type` is `multipart/form-data` — form data is parsed and values are made available as request arguments
- if POST data starts with `{` or `[` and doesn't contain `=`, it is placed into `plain` request argument
So even with #2131 fixed your request body will not be parsed into arguments array.
Therefore I suggest to change payload to `{"val":"prue"}`, and get it via `server.arg("plain")`. Of course you will also need to change `content-length` to `Content-Length` in your request until #2131 is fixed.

enzosagretti, 2016-06-12:
I think that with #2131 fixed solve it will work fine, because i change case sensitive compare and work for the next example:

**HTTP_POST request**
`http://192.168.0.16/1`

**HTTP_POST Header**

```
POST /1 HTTP/1.1
 HOST: 192.168.0.16
 accept: application/json
 accept-encoding: gzip, deflate
 accept-language: en-US,en;q=0.8
 content-type: application/json
 user-agent: Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/51.0.2704.84 Safari/537.36
 content-length: 18
 
 **val={"val":"prue"}**
```

In the ESP8266 Server side, I parse the request, and return the date payload in Json, and work fine.
ESP8266 return me:

```
{
"val": "prue"
}
```

igrr, 2016-06-12:
Ah, like that. I haven't figured out that `val=` part of your payload was to specify the argument name. That will work as well.
Closing the issue as you have figured out the solution :)

closed 2016-06-12 by igrr

noisemaker00, 2017-06-17:
I cannot get anything on the `server.arg("plain");`.
When I perform POST request there's no args.