esp8266/Arduino #2134: Using esp8266(Nodemcu) in the slave mode for SPI communication with a arduino uno

cop663, 2016-06-12:
Hi everyone!

I am trying to use a nodemuc in the slave mode to do the SPI communication with a arduino uno.

I don't know how to write the code for the slave part. I did some research work, and [find the codes for the arduino uno. ](http://arduino.stackexchange.com/questions/16348/how-do-you-use-spi-on-an-arduino)

Since it uses the AVR mcu, the code below can't compiled for the nodemcu:

`#include <SPI.h>
 
char buf [100];
volatile byte pos;
volatile boolean process_it;
 
void setup (void)
{
  Serial.begin (9600);
 
  // have to send on master in, *slave out*
  pinMode(MISO, OUTPUT);
 
  SPCR |= _BV(SPE);
 
  pos = 0; 
  process_it = false;
 
  SPI.attachInterrupt();
}
 
ISR (SPI_STC_vect)
{
  byte c = SPDR;  
  if (pos < sizeof(buf))
  {
    buf [pos++] = c;
    if (c == '\n')
      process_it = true;
  }
}
 
void loop (void)
{
  if (process_it)
  {
    buf [pos] = 0;  
    Serial.println (buf);
    pos = 0;
    process_it = false;
  } 
}`

I guess the main problem comes from:
`ISR (SPI_STC_vect)
{
  byte c = SPDR;  
  if (pos < sizeof(buf))
  {
    buf [pos++] = c;
    if (c == '\n')
      process_it = true;
  }
}
 `

Anyone knows how to solve it?

Thank you!!

Best

### Debug Messages
[image]

kapyaar, 2016-07-03:
I am looking for the same thing. I am familiar with ESP8266 and arduino core. I wish to connect an ESP8266 (nodemcu module) to an arduino due. How should I go about with it? Should I program ESP with arduino core, and Arduino due seperately, and let them talk using uart? I really like the arduino core for ESP.  Is there a way to use this on an ESP as a slave? 

thanks much for any hints.

me-no-dev, 2016-07-03:
I have spi slave implementation almost ready. Stay tuned

kapyaar, 2016-07-03:
great! testers waiting :)

littleboot, 2016-07-14:
I'm also looking for this. thanks in advance

me-no-dev, 2016-07-14:
it's already merged! Use git version of the core and update to the latest. Examples included
https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave/examples

Mr-Mayhem, 2016-10-11:
I got the master and slave examples to work on Teensy 3.6 and ESP8266 respectively.
But, I am confused on how to modify the master and slave example sketches for my application. 

My project uses a linear photodiode array sensor with 256 pixels. Each pixel integer value is contained inside two bytes.

So, I have an array of 512 bytes every so often that I want to send from the Teensy 3.6 master to a ESP8266 slave. 

I know I need to break this into chunks of 32 bytes and send each chunk in subsequent SPI transfers. But I have no idea how to change the code from all that string manipulation "*char" stuff to send and receive my bytes. (I am getting lots of conversion errors.)

 I'd like to short-circuit it bypassing as much of the string-based code or "cute feature" code as possible, and copy in my bytes to the input somehow, keeping the 32 byte SPI limit in mind. I want to tune it for raw speed, with a flow control bit to skip sending now and then because the ESP8266 will not be able to keep up with the incoming data since it is sinking it to WiFi on the ESP8266, which is slower than the rate it comes in from the Teensy 3.6 running the sensor.

In a second confusion, I see the MOSI and MISO on my oscilloscope, which is properly decoding the SPI protocol (I see the decoded text appearing on MOSI and MISO on the oscope) . I see the question pop in one frame, and the answer appear in the next frame, and then it repeats.  Each frame in my terminology is the period when the slave select is low; the common understanding.

I wonder if this half-duplex behavior is a consequence of the example code being what it is, a simple ping-pong example, or if Full Duplex capability was lost altogether in the way the library was written. If I need to do some kind of flow control, to avoid overloading the ESP with too much data too fast, I imagine I would prefer to make use of full-duplex for sake of speed and/or efficiency. By full-duplex I mean that during each transfer, data is exchanged in both directions. In the example code, data goes one direction in one transfer, and the opposite direction in the next transfer, and repeats, so half the speed overall. 

Although, I think my project's ideal arrangement if possible, would be to use most of the SPI time window for sending master to slave, and one byte reserved for the opposite direction, for use as flow control "slow down" byte. I suspect that is what the current example is getting at, but am uncertain. I am happy to PayPal $100.00 for new SPI Master/Slave code examples tuned for my byte array situation.

BiplavOraon, 2017-05-23:
@me-no-dev in your code SPISlave_Test.ino, shouldn't the logic be inside void loop() rather than void setup()? When I used the codes, the slave(ESP8266) was not able to send the data to master(Arduino). Please help. Thanks in advance.

devyte, 2017-10-17:
The implementation fo SPISlave is merged. Closing.

closed 2017-10-17 by devyte

phil123456, 2018-01-03:
the example code makes no sens is the esp master or slave ?

https://github.com/esp8266/Arduino/blob/master/libraries/SPISlave/examples/SPISlave_Master/SPISlave_Master.ino

regards

pcliu624, 2018-06-01:
@me-no-dev hey im using mega2560  to be master
pin connect:       mega2560   nodemcu
                  sck          52              D5
                  mosi       51              D7
                  miso       50              D6
                   SS          53              D8
also 3v3 and GND
I keep monitor the output window of mega2560 (right?)
It just output Master: Are you alive?
                       Slave:
that mean nodemcu doesn't respond , also the nodemcu onboard leds not working too.
please help !

cheers

wan420, 2018-11-27:
Can I use this for communication between Nodemcu32S and esp8266 ?

tarbear123, 2019-04-30:
can I transfer jpeg/gif through spi?

phil123456, 2019-05-02:
> can I transfer jpeg/gif through spi?

rofl...no you can't .... SPI have a special spies inside it and it does not like jpegs, nor porn videos
the other files will be accepted though....ha ha ha

dude spi is a low level electronic protocol, it's a serial communication, it does not even know what a jpeg is, it's meant to transfer anything, as long as it's bits

you'll have to write yourself the applicative/handshake part (just a state machine, put the length of the buffer you transmit in the beginning for instance, then the bytes you wanna transfer) and transfer chunks of data until you transfered everything