esp8266/Arduino #2135: ESP8266 reset cycles and flash read/write problem

heroseller17, 2016-06-12:
Hello
i have a question 
i use esp8266 (esp-07) with arduino ide
i program it one time and as you know it resets alot....i change my power supply it helps me alot but still i reset!
my question is that is there any problem with esp8266 reset many times?
a ask espressif about it and they answer if you dont read/write flash every time after reset ..there is no problem.
i dont know if i use flash read/write.
i just know every time that esp reset the program start from begining!
please help
thanks

igrr, 2016-06-13:
With core versions above 2.2.0, there are no extra flash writes happening on repeated restarts.
So flash wear shouldn't be an issue, unless you explicitly use SPIFFS or EEPROM functions in your sketch.

heroseller17, 2016-06-13:
Hi igrr thanks for yor reply...i dont use EEPROM in my sketch ...my worry is ,one time program esp without using EEPROM and use your library dosent have any effect on flash memory and cause esp8266 Reset limitation?

igrr, 2016-06-13:
If you aren't using EEPROM you should be fine.

heroseller17, 2016-06-13:
thanks alot...and how about the previous versions of the library?

devyte, 2017-10-17:
Closing per #3655 .

closed 2017-10-17 by devyte