esp8266/Arduino #2138: Espressif SDK 1.5.4 released

mic159, 2016-06-12:
I know you currently have an RC with 1.5.3 in it, but it looks like espressif have released yet another version, 1.5.4
http://bbs.espressif.com/viewtopic.php?f=46&t=2198

The main item I am keen on is this:
> Revised the issue that API system_adc_read and system_get_vdd33 may return wrong value.

supersjimmie, 2016-06-14:
Could that ADC issue be related to this? https://github.com/esp8266/Arduino/issues/2070

Gorkde, 2016-06-16:
probably, got the same issue

supersjimmie, 2016-07-04:
Doesn't solve the adc issue.

mic159, 2016-07-04:
They have released a patch for this version: http://bbs.espressif.com/viewtopic.php?f=46&t=2376

osos, 2016-10-27:
I am interested in this update, as I might encounter this resolved issue in the SDK:

> The ESP8266 maybe reset if connected to a special router.

any timeline?

devyte, 2017-10-17:
The target for release 2.4.0 is to move straight to SDK 2.1.0. Closing.

closed 2017-10-17 by devyte