esp8266/Arduino #2145: Frequent resets of WifiServer especially in softap mode

boguslawb, 2016-06-13:
### Basic Infos

#### Hardware
Hardware:			Nodemcu 1.0
Core Version:      	Arduino 1.6.8, code from github 13.06.2016

### Description

Problem description

Code like posted below (simple echo server) cause frequent wdt resets, exception  especially in softap mode of module. The similar code in lua 5.1.4 runs without a single problem for hours with the same client connected (which is other ESP12E module programed in Arduino using wificlient ).

What am I doing wrong ? It is not about power issue because using the same board with Nodemcu Lua it works reliably.

### Settings in IDE
Arduino 1.6.8
Module:  			?Nodemcu 0.9?
Flash Size: 		?4MB/1MB(SPIFFS)?
CPU Frequency:		?160Mhz?
Upload Using:		 SERIAL115200kb/s?
Reset Method:		?ck / nodemcu?

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 

const char *ssid = "ssid";
const char *password = "12345678";
const char *host = "192.168.7.1";
int channel = 5;

WiFiServer server(8008);

void setup() {
  WiFi.persistent(false);
  Serial.begin(9600);
  delay(10);
  Serial.println();
  initWiFi();
  server.begin();
}

void loop() {
  yield();
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("new client");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        currentLine = client.readStringUntil('\r');
        Serial.println(currentLine);
        ESP.wdtFeed();
        yield();
      }
    }
    // close the connection:
    
    client.flush();
    client.stop();
    Serial.println("client disonnected");
  }  
}

void initWiFi(void) {
  
  IPAddress Ip(192, 168, 1, 1);
  IPAddress NMask(255, 255, 255, 0);

  WiFi.mode(WIFI_AP);
  Ip.fromString(host);
  WiFi.softAPConfig(Ip, Ip, NMask);
 if (!WiFi.softAP( ssid, password,channel,0))
  {
   Serial.println("WiFi.softAP failed.(Password too short?)");
   return;
  }
  IPAddress myIP = WiFi.softAPIP();
  Serial.println();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");  
  
}

igrr, 2016-06-13:
Could you please share the code for the client side as well, or describe data sending pattern (how much data, intervals between data, and so on)?
Also if you see a crash dump, please record it, decode the result, and paste it here. Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

boguslawb, 2016-06-14:
First stack dump:
Decoding 10 results
0x4020be94: ieee80211_decap at ?? line ?
0x401004d8: malloc at /home/boguslaw/.arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/umm_malloc/umm_malloc.c line 1664
0x40209fa4: ieee80211_hostap_attach at ?? line ?
0x4020a4ae: hostap_input at ?? line ?
0x40105e76: igmp_timer at /Users/igrokhotkov/e/arduino/tools/sdk/lwip/src/core/timers.c line 222
0x402053d2: pp_tx_idle_timeout at ?? line ?
0x40204d1b: ppPeocessRxPktHdr at ?? line ?
0x402032f3: loop_task at /home/boguslaw/.arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/core_esp8266_main.cpp line 43

boguslawb, 2016-06-14:
Client code. It is starting code for communication between two ESP12 modules with JSON data.

`

#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <OneWire.h>
#include <DallasTemperature.h>

const char *ssid = "ssid";
const char *password = "12345678";
const char *host = "192.168.7.1";

#define ONE_WIRE_BUS D2

OneWire oneWire(ONE_WIRE_BUS);

DallasTemperature sensors(&oneWire);

struct HTSensorData {
 const char* name;  // HT0
 int time;          
 int status;      
 int error;        
 float temperature;
 int mode;          
};

struct HouseData {
 float temperature; 
 float requested;  
 float tmax;        
 float tmin;       
 int   rmode;     
};

#define HTSENSORDATA_JSON_SIZE (JSON_OBJECT_SIZE(6))
#define HOUSEDATA_JSON_SIZE (JSON_OBJECT_SIZE(5))

bool isConnected = false;//Czy podłączono do sieci WiFi

bool deserialize(HTSensorData& data, char* json);
void serialize(const HTSensorData& data, char* json, size_t maxSize);

bool deserialize2(HouseData& data, char* json);
void serialize2(const HouseData& data, char* json, size_t maxSize);

WiFiClient client;
const int httpPort = 8008;

char json[200];

HTSensorData htdata;
HouseData house;

void setup() {
  int count=0;
  Serial.begin(115200);
  delay(10);
  Serial.println();
  sensors.begin();
  Serial.print("Connecting to ");
  Serial.println(ssid); 
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    count++;
    if (count>30) {
      Serial.println();
      Serial.print("Cannot connect to ");
      Serial.println(ssid);
      return;
    }
  }
  isConnected = true;
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  
  
}

void loop() {
  delay(100);
  if (!isConnected) return;  
  Serial.print("Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println("DONE");
  // After we got the temperatures, we can print them here.
  // We use the function ByIndex, and as an example get the temperature from the first sensor only.
  Serial.print("Temperature for the device 1 (index 0) is: ");
  htdata.temperature = sensors.getTempCByIndex(0);
  Serial.println(htdata.temperature);  

  Serial.print("Connecting to host ");
  Serial.println(host);
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
  }
  else
  {

  htdata.name = "HT0";
  htdata.time = millis();  
  htdata.mode = 1;
  htdata.status = 0;
  htdata.error = 0;
  htdata.mode = 0;
  serialize(htdata,json,sizeof(json));
  client.println(json);
  Serial.println(json);
  
  while(client.available()){
    Serial.println("Data available...");
    String line = client.readStringUntil('\r');
    Serial.println(line);
    line.toCharArray(json, 200);
    deserialize2(house,json);
  }
  
  
  Serial.println();
  if (client.connected()) {
    Serial.println();
    Serial.println("disconnecting from host.");
    client.stop();
  }
  
  }

}

bool deserialize(HTSensorData& data, char* json)
{
    StaticJsonBuffer<HTSENSORDATA_JSON_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(json);
    htdata.name = root["name"];
    htdata.time = root["time"];
    htdata.status = root["status"];
    htdata.error = root["error"];
    htdata.temperature = root["temperature"];
    htdata.mode = root["mode"];
    return root.success();
}

void serialize(const HTSensorData& data, char* json, size_t maxSize)
{
    StaticJsonBuffer<HTSENSORDATA_JSON_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["name"] = htdata.name;
    root["time"] = htdata.time;
    root["status"] = htdata.status;
    root["error"] = htdata.error;
    root["temperature"] = double_with_n_digits(htdata.temperature,1);
    root["mode"] = htdata.mode;
    root.printTo(json, maxSize);
}

bool deserialize2(HouseData& data, char* json)
{
    StaticJsonBuffer<HOUSEDATA_JSON_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(json);
    house.temperature = root["temperature"];
    house.requested = root["requested"];
    house.tmax  = root["tmax"];
    house.tmin = root["tmin"];
    house.rmode = root["rmode"];
    return root.success();
}

void serialize2(const HouseData& data, char* json, size_t maxSize)
{
    StaticJsonBuffer<HOUSEDATA_JSON_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["temperature"] = house.temperature;
    root["requested"] = house.requested;
    root["tmin"] = house.tmin;
    root["tmax"] = house.tmax;
    root["rmode"] = house.rmode;
    root.printTo(json, maxSize);
}

`

boguslawb, 2016-06-14:
I don't know how to turn on syntax coloring

boguslawb, 2016-06-14:
Resets are very frequent. Another one:
Decoding 9 results
0x4020a127: hostap_input at ?? line ?
0x4020a282: hostap_input at ?? line ?
0x4020a395: hostap_input at ?? line ?
0x40106f64: vPortFree at /home/boguslaw/.arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/heap.c line 18
0x402053d2: pp_tx_idle_timeout at ?? line ?
0x40204d1b: ppPeocessRxPktHdr at ?? line ?
0x402032f3: loop_task at /home/boguslaw/.arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/core_esp8266_main.cpp line 43

boguslawb, 2016-06-14:
Is there any way to get also stack trace for my program ? Any way to add debug flag into Arduino IDE ?

boguslawb, 2016-06-14:
Similar exception just after establishing AP.
Exception (0):
epc1=0x4020e08c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff00:  3ffeca1e 00000006 4020e02e 00000008  
3fffff10:  4021206d 000000b1 000000b1 4010020c  
3fffff20:  3fff02ec 4020a14b 3ffe9b80 3ffec6c8  
3fffff30:  3ffeff2c 3ffeff40 00000020 40106f64  
3fffff40:  402057a0 00000001 3ffe9b8a 3ffe9b80  
3fffff50:  0000001c 3ffe8a94 3fff02ec 3ffe9b74  
3fffff60:  00000000 00000025 402053d2 3ffec6c8  
3fffff70:  3ffe9b74 3fffdcc0 3ffe8cd0 3ffe8cd0  
3fffff80:  3ffe9b84 00000040 00000000 3ffee610  
3fffff90:  40204d1b 3fffdab0 00000000 402032f3  
3fffffa0:  3ffe8cd0 40000f49 3fffdab0 40000f49  
<<<stack<<<

Decoding 11 results
0x4020e02e: ieee80211_send_deauth at ?? line ?
0x4021206d: cnx_node_search at ?? line ?
0x4010020c: _umm_free at /home/boguslaw/.arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x4020a14b: hostap_input at ?? line ?
0x40106f64: vPortFree at /home/boguslaw/.arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/heap.c line 18
0x402057a0: DefFreqCalTimerCB at ?? line ?
0x402053d2: pp_tx_idle_timeout at ?? line ?
0x40204d1b: ppPeocessRxPktHdr at ?? line ?
0x402032f3: loop_task at /home/boguslaw/.arduino15/packages/esp8266/hardware/esp8266/2.3.0-rc1/cores/esp8266/core_esp8266_main.cpp line 43

boguslawb, 2016-06-14:
I've removed the whole package from .Arduino15 subdirectory and installed again esp8266 core from github. Then I changed client code and moved json coding before connection with server. Now it is much more stable, but after some time server module just hang without any response to serial port and nothing can connect to its softAP. When it hangs it throw into serial monitor something like :
client disonnected
new client
{"name":"HT0","time":7807157,"status":0,"error":0,"temperature":24.4,"mode":0}

client disonnected
new client
ü!m!ÿæÊṁýŷü

(using gtkterm)

boguslawb, 2016-06-15:
With debug output turned on there is when it hangs:

client disonnected
new client
{"name":"HT0","time":156297,"status":0,"error":0,"temperature":24.1,"mode":0}
add 2
aid 2
station: 00:24:d6:a7:a7:9e join, AID = 2
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1 
...... and so on repeating all the time

boguslawb, 2016-06-15:
After simplify server code to this loop:

void loop() {
  yield();
  WiFiClient client = server.available();   // listen for incoming clients
  
  if (!client) {
    return;
  }

    Serial.println("new client");           // print a message out the serial port
    //while (client.connected()) {            // loop while the client's connected
    //  if (client.available()) {             // if there's bytes to read from the client,
        String currentLine = client.readStringUntil('\r');
        Serial.println(currentLine);
        client.flush();
      
    Serial.println("client disonnected");
 // }  

    
    
}

Much more stable but after 15 minutes....

client disonnected

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
�
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.7.1,mask:255.255.255.0,gw:192.168.7.1)
bcn 100

AP IP address: 192.168.7.1
signal strength (RSSI):31 dBm
add 1
aid 1
station: 18:fe:34:e0:36:d2 join, AID = 1
new client
{"name":"HT0","time":1188123,"status":0,"error":0,"temperature":24.0,"mode":0}
client disonnected

boguslawb, 2016-06-15:
What else can I do to resolve this problem ? Is this related to my code ?

boguslawb, 2016-06-15:
How is the Arduino code different then Lua example below (working with the same Wificlient module)  ? In Lua this echo server  works for hours without any resets or exception. It seems a bit slower in response time then Arduino code  but rock solid.

wifi.setmode(wifi.SOFTAP);  
wifi.ap.config({ssid="Hobbit2",pwd="012345678"});
wifi.ap.setip({ip="192.168.7.1",netmask="255.255.255.0",gateway="192.168.7.1"})
print("Server IP Address:", wifi.ap.getip())
srv=net.createServer(net.TCP)
srv:listen(8008,function(conn)
    conn:on("receive", function(client,request)
        print("received: " .. request);
        client:send("echo: " .. request);
        client:close();
        collectgarbage();
    end)
end)

boguslawb, 2016-06-18:
Seems that module is damaged. With other one (not Nodemcu board)  all works fine. Memory chip corruption ??

closed 2016-06-18 by boguslawb

k2ksk2k, 2019-12-25:
Try to set in the arduino IDE, tools-> Erase Flash: "Sketch + WiFi Settings".