esp8266/Arduino #2147: MQTT Publish Exception

Pragadheeshraj, 2016-06-14:
Hello I have a ESP8266-12E module. I get the following exception when I try to send messages over MQTT.

This happens only for certain messages which are longer than length 250 or so.

Exception Info
Exception (9):
epc1=0x40203555 epc2=0x00000000 epc3=0x00000000 excvaddr=0x2c7d7b3a depc=0x00000000

Stack Decoded Info
Decoding 9 results
0x40203555: PubSubClient::connected() at C:\Users\Pragadheesh Raj\Documents\Arduino\libraries\arduino_191215\src/PubSubClient.cpp line 586
0x402036fc: Print::write(unsigned char const*, unsigned int) at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Print.cpp line 42
0x402029a5: PubSubClient::publish(char const*, unsigned char const*, unsigned int, unsigned char) at C:\Users\Pragadheesh Raj\Documents\Arduino\libraries\arduino_191215\src/PubSubClient.cpp line 586
0x402033b5: Print::write(char const*) at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Print.h line 60
0x40202a41: PubSubClient::publish(char const*, char const*) at C:\Users\Pragadheesh Raj\Documents\Arduino\libraries\arduino_191215\src/PubSubClient.cpp line 586
0x4020383c: Print::println(char const*) at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Print.cpp line 143
0x40201a41: loop at C:\Program Files (x86)\Arduino/mqtt_esp8266.ino line 119
0x40202b8d: loop_wrapper at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100114: cont_norm at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/cont.S line 109

igrr, 2016-06-14:
Do you mind posting your code?
I'd guess that you have a buffer allocated for the message payload, and you are overruning it (writing beyond its size). See excvadr value in the dump? It likely is the value should contain your PubSubClient instance address. Instead it contains 0x2c7d7b3a, which is (if you decode hex to ASCII and take endianness into account) `:{},` character sequence. So, check your buffer size and the amount of data you are trying to put there.

Pragadheeshraj, 2016-06-14:
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

const char* ssid = "NEXUS 5";
const char* password = "1234567890";
const char* mqtt_server = "broker.mqtt-dashboard.com";

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

void setup() {
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  //client.setCallback(callback);
}

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

/*void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  }*/

//}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      //client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, 512, "{\"switches\":[{\"schedule\":{},\"switch_id\":\"6\"},{\"schedule\":{},\"switch_id\":\"7\"},{\"schedule\":{},\"switch_id\":\"8\"},{\"schedule\":{\"start_time\":\"2016061512300\",\"end_time\":\"2016061512310\",\"schedule_status\":0,\"priority\":0},\"switch_id\":\"9\"}],\"mac_address\":\"5c:cf:7f:18:c8:ab\"}");
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
}

Pragadheeshraj, 2016-06-14:
It just sends a JSON text message every 2s.

Pragadheeshraj, 2016-06-14:
I also tried using a String variable to store and publish the payload. This is the result which I get

igrr, 2016-06-14:
You need to fix either

    This happens only for certain messages which are longer than length 250 or so.

or
```c++
char msg[50];
```

and
```c++
snprintf (msg, 512, ...
```

Where did that 512 come from, anyway?
The way `snprintf` is normally used with static arrays is as follows:
```c++
snprintf(buf, sizeof(buf), format_string, ...);
```

closed 2016-06-14 by igrr

Pragadheeshraj, 2016-06-14:
I had tried this 
char arr[512];
String msg = ""

msg.toCharArray(arr,msg.length());
publish(topic,arr);

I got this
Soft WDT reset

ctx: cont 
sp: 3ffefe40 end: 3fff00b0 offset: 01b0

>>>stack>>>
3ffefff0:  3fff0015 00000000 00000000 40203c7e  
3fff0000:  00000292 00000000 3ffeed20 00003a97  
3fff0010:  00000001 00000000 3fff00fc 4020376d  
3fff0020:  3fff00fc 00000001 3ffe8885 3ffeeb0c  
3fff0030:  00000000 00000108 3ffeed1c 402031b8  
3fff0040:  3ffeed20 3ffe84f0 3fff00fc 40203b65  
3fff0050:  3ffe8884 3fff00fc 3fff00fc 3ffef084  
3fff0060:  3ffeed1c 3ffeeb0c 3ffeed1c 402031f1  
3fff0070:  3ffe84f0 3ffeeb0c 3ffeed0c 40204019  
3fff0080:  3ffeed1c 3ffeeb0c 3ffeed0c 402021e5  
3fff0090:  3fffdc20 00000000 3ffef07c 4020333d  
3fff00a0:  00000000 00000000 3ffef090 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

igrr, 2016-06-14:
Why don't you pass msg.length() to `publish`?
Your string in `arr` is not zero terminated, so how is PubSubClient supposed to know where it ends, i.e. how many characters to send?

Pragadheeshraj, 2016-06-14:
I did this 
publish(topic,arr,msg.length());

I get the same error

Pragadheeshraj, 2016-06-14:
What does this actually mean

Soft WDT reset

ctx: cont 
sp: 3fff0030 end: 3fff02a0 offset: 01b0

igrr, 2016-06-14:
This sketch works for me without any issues: http://hastebin.com/cuvunevipo.axapta

Here's the output:
```

Connecting to #!/test/bin
..
WiFi connected
IP address: 
192.168.1.13
Attempting MQTT connection...connected
Publish message: 
Publish message: 
Publish message: 
Publish message: 
pm open,type:2 0
Publish message: 
Publish message: 
Publish message: 
Publish message: 
Publish message: 
Publish message: 
Publish message: 
Publish message: 
```

If I change the sketch to the following: http://hastebin.com/jejubetupi.axapta

Then the output is
```

Connecting to #!/test/bin
..
WiFi connected
IP address: 
192.168.1.13
Attempting MQTT connection...connected
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
```

Soft WDT reset means that there is a loop somewhere which takes too long to finish. You can decode the rest of the stack dump same way you did in your original post.

Pragadheeshraj, 2016-06-14:
Actually the JSON is dynamic data and is stored in s String variable. Any suggestions on how I can publish it ?

Pragadheeshraj, 2016-06-14:
Decoding 10 results
0x40203c6a: PubSubClient::write(unsigned char, unsigned char*, unsigned short) at C:\Users\Pragadheesh Raj\Documents\Arduino\libraries\arduino_191215\src/PubSubClient.cpp line 586
0x402032e0: esp_yield at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40203759: HardwareSerial::write(unsigned char) at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/HardwareSerial.cpp line 541
0x402031a4: PubSubClient::publish(char const*, unsigned char const*, unsigned int, unsigned char) at C:\Users\Pragadheesh Raj\Documents\Arduino\libraries\arduino_191215\src/PubSubClient.cpp line 586
0x40203b51: Print::write(char const*) at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Print.h line 60
0x402031dd: PubSubClient::publish(char const*, char const*) at C:\Users\Pragadheesh Raj\Documents\Arduino\libraries\arduino_191215\src/PubSubClient.cpp line 586
0x4020402c: Print::println(char const*) at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/Print.cpp line 143
0x402021cd: loop at C:\Program Files (x86)\Arduino/sJSON.ino line 122
0x40203329: loop_wrapper at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100114: cont_norm at C:\Users\Pragadheesh Raj\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/cont.S line 109

Pragadheeshraj, 2016-06-14:
The Decoded Stack

igrr, 2016-06-14:
Like this:
```c++
String payload = "{\"something\":42}"; // customize as necessary
client.publish("outTopic", payload.c_str());
```

As I mentioned above, I'm not able to reproduce the issue with the sketch you have posted.
If you are having an issue with different sketch, please post the exact sketch you are using.

igrr, 2016-06-14:
For the record, I'm using this PubSubClient: https://github.com/knolleary/pubsubclient
Not sure what your `arduino_191215` library is, maybe you need to check the issue with it's author.

Pragadheeshraj, 2016-06-14:
`#include <ESP8266WiFi.h>
#include <PubSubClient.h>

const char* ssid = "NEXUS 5";
const char* password = "1234567890";
const char* mqtt_server = "52.86.34.244";

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char arr[512];
int value = 0;
String msg="";
void setup() {
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  //client.setCallback(callback);
}

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

/*void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  }*/

//}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    String id = "ESP8266Client";
    id += String(random(0xffff), HEX);
    if (client.connect(id.c_str())) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      //client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    msg += "{\"switches\":[{\"schedule\":{},\"switch_id\":\"6\"},{\"schedule\":{},\"switch_id\":\"7\"},{\"schedule\":{},\"switch_id\":\"8\"},{\"schedule\":{\"start_time\":\"2016061512300\",\"end_time\":\"2016061512310\",\"schedule_status\":0,\"priority\":0},\"switch_id\":\"9\"}],\"mac_address\":\"5c:cf:7f:18:c8:ab\"}";
    Serial.print("Publish message: ");
//    msg.toCharArray(arr, msg.length());
//    client.publish("outTopic", arr, msg.length());
//    snprintf (arr, sizeof(arr), "{\"switches\":[{\"schedule\":{},\"switch_id\":\"6\"},{\"schedule\":{},\"switch_id\":\"7\"},{\"schedule\":{},\"switch_id\":\"8\"},{\"schedule\":{\"start_time\":\"2016061512300\",\"end_time\":\"2016061512310\",\"schedule_status\":0,\"priority\":0},\"switch_id\":\"9\"}],\"mac_address\":\"5c:cf:7f:18:c8:ab\"}");
    Serial.println(msg);
    client.publish("outTopic", msg.c_str());
  }
}`

I am using the same PubSubClient library

igrr, 2016-06-14:
Does it crash on the first publish attempt, or later on?
You are adding same string to `msg` every time, so `msg` becomes longer on every `loop` iteration, this may be causing issues.
Also I assume you know that PubSubClient has a limited buffer size, right? It's 128 bytes by default:
https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h#L25-L27

Pragadheeshraj, 2016-06-14:
Yeah I am aware of that. I missed this while posting
``
long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
   msg="";
    msg += "{\"switches\":[{\"schedule\":{},\"switch_id\":\"6\"},{\"schedule\":{},\"switch_id\":\"7\"},{\"schedule\":{},\"switch_id\":\"8\"},{\"schedule\":{\"start_time\":\"2016061512300\",\"end_time\":\"2016061512310\",\"schedule_status\":0,\"priority\":0},\"switch_id\":\"9\"}],\"mac_address\":\"5c:cf:7f:18:c8:ab\"}";
    Serial.print("Publish message: ");
//    msg.toCharArray(arr, msg.length());
//    client.publish("outTopic", arr, msg.length());
//    snprintf (arr, sizeof(arr), "{\"switches\":[{\"schedule\":{},\"switch_id\":\"6\"},{\"schedule\":{},\"switch_id\":\"7\"},{\"schedule\":{},\"switch_id\":\"8\"},{\"schedule\":{\"start_time\":\"2016061512300\",\"end_time\":\"2016061512310\",\"schedule_status\":0,\"priority\":0},\"switch_id\":\"9\"}],\"mac_address\":\"5c:cf:7f:18:c8:ab\"}");
    Serial.println(msg);
    client.publish("outTopic", msg.c_str());
  }
}
``
It fails at the very first attempt
I have also changed the buffer size to 512

Pragadheeshraj, 2016-06-14:
Tried the same code you posted  http://hastebin.com/jejubetupi.axapta
The same error occurs

igrr, 2016-06-14:
So I compiled the sketch from the two bits you posted: http://hastebin.com/cufapizemo.axapta

Here's the output:
```
connected
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
pm open,type:2 0
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab" [... cut ...]
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
Publish message: {"switches":[{"schedule":{},"switch_id":"6"},{"schedule":{},"switch_id":"7"},{"schedule":{},"switch_id":"8"},{"schedule":{"start_time":"2016061512300","end_time":"2016061512310","schedule_status":0,"priority":0},"switch_id":"9"}],"mac_address":"5c:cf:7f:18:c8:ab"}
```

Works fine as far I can tell. Your issue is not reproduced here.

Pragadheeshraj, 2016-06-14:
Have you change the buffer size of PubSub Client Library ?

chaeplin, 2016-06-14:
@Pragadheeshraj this line https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h#L26

Pragadheeshraj, 2016-06-14:
Hello @chaeplin I had changed it to 512

chaeplin, 2016-06-14:
@Pragadheeshraj Hello.
I have tested http://hastebin.com/cufapizemo.axapta with 'MQTT_MAX_PACKET_SIZE 450' to my local mosquitto. Can publish and no err.

vicarious11, 2018-11-21:
Hey,

I am publishing packet size of 400 bytes. Changed the library and works fine as well. But after sometime, the ESP crashes. I have declared the variables locally too. I still cannot figure this out.

marekhalmo, 2019-09-05:
This happens to me as well if i'm connected to a public MQTT server and the request takes to long.. Might that be the issue?
There should be some hard limit on how long the MQTT call can take. But i'm not sure that you can break that in the Arduino