esp8266/Arduino #2148: ESP07 Timeout problem on CMD

heroseller17, 2016-06-14:
Hi
i have a problem with esp-07 
in my home every thing is good and i have no problem with esp
but in office i have a problem! and i always send 5bit packet but the time is app about~200ms -400ms and many times timeout!
the same board and the same circuit i just change the ssid and password and ip...thats it
but i have this problem ....
why its happend?

heroseller17, 2016-06-16:
is it possible on range of wifi ?

devyte, 2017-10-17:
@heroseller17 The amount of wifi traffic dictates how much work must be done servicing the wifi stack. Most likely there is much higher traffic at work, which means you're allowed less CPU time for your app.
Most likely your app needs to be redesigned and broken down into pieces that have shorter execution time.
Not the right place to discuss. Closing per #3655 .

closed 2017-10-17 by devyte