esp8266/Arduino #2150: HTTPS GET Request returns -1

djoele, 2016-06-14:
Hello everybody,

today I updated the ESP8266 code from Github. 

My latest update was:
commit 32bd42b0281237d29e18bb769afad02bc6782a64, June 6th

Now  I have:
commit da17d5425a275bca2d891603ac5764b4fb6fbaff, Jun 14th

I have the function below which is responsible for doing a HTTPS GET Request.
The behaviour that I see now is that http=>GET now always returns -1.
When I do the request from a browser it is working and returns 200.

Anybody sees what I am doing wrong? Were there any changes in the ESP8266 codebase that forces me to change something?

  ```
const char* trigger_stack = "https://....";
const String host = ".....";
const int httpsPort = 443;

callURL(trigger_stack, host, httpsPort);

  void callURL(String url, String host, const int port) {
      HTTPClient * http = new HTTPClient();
      http->begin(host, port, url, fingerprint);
      http->setAuthorization(www_username,www_password);
      http->addHeader("X-ESP8266-IP", ipadres);

      int httpCode = http->GET();
      Serial.println((String("[HTTP] url: ") + url + ", return code: " + httpCode));
      http->end();
     http->~HTTPClient();
     delete http;
}
```

djoele, 2016-06-15:
Thank you for the fast response and possible fix. Tonight I will check if it works and report back.

djoele, 2016-06-15:
I tried with the latest fix but I still get the same result: -1.
I will try to see if I can turn on debugging in Arduino IDE.

igrr, 2016-06-16:
Which server are trying to connect to in your test?

djoele, 2016-06-16:
I am coonecting to a Nginx server running on port 443.

Behind this server there is a Domoticz server running on port 8080.

HTDemon, 2016-06-17:
Hi, I got similar issue connect to Slack HTTPS API and tried 2.3.0-rc2, 2.3.0-rc1, 2.2.0 and 2.1.0.

I tried http.begin(String url, String httpsFingerprint) and http.begin(String url) with 2.3.0-rc2
`http.begin("https://slack.com/api/rtm.start?token=" SLACK_BOT_TOKEN, httpsFingerprint);
  int httpCode = http.GET();`

debug log:
> [HTTP-Client][begin] url: https://slack.com/api/rtm.start?token=...
[HTTP-Client][begin] host: slack.com port: 443 url: /api/rtm.start?token=...
[HTTP-Client][begin] httpsFingerprint: ....
[hostByName] request IP for: slack.com
[hostByName] Host: slack.com IP: ....
[HTTP-Client] connected to slack.com:443
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/json; charset=utf-8'
[HTTP-Client][handleHeaderResponse] RX: 'Transfer-Encoding: chunked'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Origin: *'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: private, no-cache, no-store, must-revalidate'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Security-Policy: referrer no-referrer;'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Fri, 17 Jun 2016 08:09:33 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Expires: Mon, 26 Jul 1997 05:00:00 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Pragma: no-cache'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache'
[HTTP-Client][handleHeaderResponse] RX: 'Strict-Transport-Security: max-age=31536000; includeSubDomains; preload'
[HTTP-Client][handleHeaderResponse] RX: 'Vary: Accept-Encoding'
[HTTP-Client][handleHeaderResponse] RX: 'X-Accepted-OAuth-Scopes: rtm:stream,client'
[HTTP-Client][handleHeaderResponse] RX: 'X-Content-Type-Options: nosniff'
[HTTP-Client][handleHeaderResponse] RX: 'X-OAuth-Scopes: identify,read,post,client,apps'
[HTTP-Client][handleHeaderResponse] RX: 'X-XSS-Protection: 0'
[HTTP-Client][handleHeaderResponse] RX: 'X-Cache: Miss from cloudfront'
[HTTP-Client][handleHeaderResponse] RX: 'Via: 1.1 35436fa36ed8ec21d3998280f7181e8b.cloudfront.net (CloudFront)'
[HTTP-Client][handleHeaderResponse] RX: 'X-Amz-Cf-Id: a6Z4JVv6s10ya6L3I3C0xpLr6tgk-JLmjA2aJrUA62148qwMv-A9dQ=='
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] Transfer-Encoding: chunked
[HTTP-Client][end] still data in buffer (2121), clean up.
check_poison_block is called for free block 0x3fff5b48
Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40039030 depc=0x00000000

`http.begin("https://slack.com/api/rtm.start?token=" SLACK_BOT_TOKEN);
  int httpCode = http.GET();`

debug log:
>[HTTP-Client][begin] url: https://slack.com/api/rtm.start?token=....
[HTTP-Client][begin] unexpected protocol: https, expected http
[HTTP-Client] connect: HTTPClient::begin was not called or returned error
[HTTP-Client][returnError] error(-1): connection refused
[HTTP-Client][end] tcp is closed
HTTP GET failed with code -1

igrr, 2016-06-17:
I think this is different issue, you are receiving Exception 3.
Doing a request to HTTPS URL without fingerprint (your second test) should not work, the error you are getting is "unexpected protocol: https, expected http". This is because no fingerprint was supplied.
Please open a new issue for your first

closed 2016-06-23 by igrr

akshaykankal, 2018-10-22:
@djoele How did you solve it?
Can you post the solution here?

djoele, 2018-10-23:
See my comment on Jun 15, 2016:

I tried with the latest fix and it is working.
Issue resolved.

Thank you.

See the commit before from igrr in which he fixed it:

https://github.com/esp8266/Arduino/commit/e64e32b329a0c83b0437d7c4d70ff688df1e7b1f

[https://avatars1.githubusercontent.com/u/4349050?s=200&v=4]<https://github.com/esp8266/Arduino/commit/e64e32b329a0c83b0437d7c4d70ff688df1e7b1f>

Fix regression in WiFiClientSecure, update HTTPS test case (#2150) · esp8266/Arduino@e64e32b<https://github.com/esp8266/Arduino/commit/e64e32b329a0c83b0437d7c4d70ff688df1e7b1f>
github.com
ESP8266 core for Arduino. Contribute to esp8266/Arduino development by creating an account on GitHub.