esp8266/Arduino #2151: mDNS library fails to find non-arduino DNS-SD services on the network

brendanmatkin, 2016-06-15:
### Basic Infos

Mac OS 10.11.5

#### Hardware
Hardware:       ESP-12E
Core Version:  2.2.0

### Description

mDNS library fails to find DNS-SD services on the network, other than those which it creates. If I load this sketch onto several ESP modules, they all find each other's advertised service. I've picked through the library and am not quite sure what's going on - it seems to me like the library might be creating & looking for non-standard DNS-SD service records, but I have been unable to find out how.

I uncommented the debugs in the mDNS library for the debug output below, as well as enabling Debug level "ALL" in the Arduino IDE.  The sketch I'm loading is just the example sketch with MDNS.queryService() looking for various services. The one included below (ipp) is a printer sharing advertisement.

Bonjour Browser shows 4 different services (all of which show "can't resolve link-local name", but do show titles, ip+port v4 & v6, and text records):

1. _ipps._tcp. (my printer)
2. _afpovertcp._tcp. (apple file sharing)
3. _ipp._tcp. (also my printer)
4. _smb._tcp. (also file sharing - Samba)

Although finding a printer or whatever is trivial, I would love to use this to find a computer on the network (with a dynamic IP) running an OTA server. The library is obviously finding SOMETHING, I think it's either just throwing errors when it shouldn't, or looking for the wrong service record structure or something...

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	160Mhz & 80Mhz (same behavior with both)
Flash Mode:		qio
Flash Frequency:	40Mhz & 80Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```/*
  ESP8266 mDNS-SD responder and query sample

  This is an example of announcing and finding services.
  
  Instructions:
  - Update WiFi SSID and password as necessary.
  - Flash the sketch to two ESP8266 boards
  - The last one powered on should now find the other.
 */

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

const char* ssid     = "<my network SSID>";
const char* password = "<my network password>";
char hostString[16] = {0};

void setup() {
  Serial.begin(115200);
  delay(100);
  Serial.println("\r\nsetup()");

  sprintf(hostString, "ESP_%06X", ESP.getChipId());
  Serial.print("Hostname: ");
  Serial.println(hostString);
  WiFi.hostname(hostString);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(250);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

 if (!MDNS.begin(hostString)) {
    Serial.println("Error setting up MDNS responder!");
  }
  Serial.println("mDNS responder started");
  MDNS.addService("esp", "tcp", 8080); // Announce esp tcp service on port 8080

  Serial.println("Sending mDNS query");
  int n = MDNS.queryService("ipp", "tcp"); // Send out query for esp tcp services
  Serial.println("mDNS query done");
  if (n == 0) {
    Serial.println("no services found");
  }
  else {
    Serial.print(n);
    Serial.println(" service(s) found");
    for (int i = 0; i < n; ++i) {
      // Print details for each service found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(MDNS.hostname(i));
      Serial.print(" (");
      Serial.print(MDNS.IP(i));
      Serial.print(":");
      Serial.print(MDNS.port(i));
      Serial.println(")");
    }
  }
  Serial.println();
  
  Serial.println("loop() next");
}

void loop() {
  // put your main code here, to run repeatedly:
  MDNS.update();            // I added this but it has the same behaviour with or without
}
```

### Debug Messages

I've seen a few different patterns (I think the first one is more accurate, the second or something similar seems to happen if I switch networks after the sketch has started). Also not sure why it keeps showing airport, the router used for the below output is an older D-Link (although I've had the same result with at least 3 different routers). 

These tests are all done with a single active ESP8266 module.

```
Connected to <my network>
IP address: 192.168.8.199
mDNS responder started
Sending mDNS query
queryService ipp tcp
Waiting for answers..
Reading answers RX: REQ, ID:0, Q:0, A:1, NS:0, ADD:7
Expected a packet with 4 answers, returning
ERR_NO_SERVICE: airport
mDNS query done
no services found

loop() next
pm open,type:2 0
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 53
:urd 13, 53, 13
ERR_NO_HOST: 192-168-8-198
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 53
:urd 13, 53, 13
ERR_NO_HOST: 192-168-8-198
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 53
:urd 13, 53, 13
ERR_NO_HOST: 192-168-8-198
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 47
Reading answers RX: REQ, ID:0, Q:0, A:1, NS:0, ADD:0
Not expecting any answers right now, returning
:urn 47
Reading answers RX: REQ, ID:0, Q:0, A:1, NS:0, ADD:0
Not expecting any answers right now, returning
```

Second Style: 

```
Connected to <my network>
IP address: 192.168.8.199
mDNS responder started
Sending mDNS query
queryService ipp tcp
Waiting for answers..
mDNS query done
no services found

loop() next
pm open,type:2 0
:urn 240
Reading answers RX: REQ, ID:0, Q:0, A:9, NS:0, ADD:0
Not expecting any answers right now, returning
:urn 400
:urd 39, 400, 13
ERR_NO_HOST: Samsung M2835 @ <my computer name>
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 400
:urd 39, 400, 13
ERR_NO_HOST: Samsung M2835 @ <my computer name>
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 538
:urd 8, 538, 13
:urd 4, 538, 22
:urd 5, 538, 27
ERR_NO_SERVICE: airport
:urn 400
:urd 39, 400, 13
ERR_NO_HOST: Samsung M2835 @ <my computer name>
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 1451
Reading answers RX: REQ, ID:0, Q:0, A:18, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 380
Reading answers RX: REQ, ID:0, Q:0, A:9, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 1451
Reading answers RX: REQ, ID:0, Q:0, A:18, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 380
Reading answers RX: REQ, ID:0, Q:0, A:9, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 156
Reading answers RX: REQ, ID:0, Q:0, A:5, NS:0, ADD:0
Not expecting any answers right now, returning
:urn 1451
Reading answers RX: REQ, ID:0, Q:0, A:18, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 380
Reading answers RX: REQ, ID:0, Q:0, A:9, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 1451
Reading answers RX: REQ, ID:0, Q:0, A:18, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 380
Reading answers RX: REQ, ID:0, Q:0, A:9, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 670
:urd 8, 670, 13
:urd 4, 670, 22
:urd 5, 670, 27
ERR_NO_SERVICE: airport
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 37
:urd 8, 37, 13
:urd 4, 37, 22
:urd 5, 37, 27
ERR_NO_SERVICE: arduino
:urn 53
:urd 13, 53, 13
ERR_NO_HOST: 192-168-8-198
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 53
:urd 13, 53, 13
ERR_NO_HOST: 192-168-8-198
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 53
:urd 13, 53, 13
ERR_NO_HOST: 192-168-8-198
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 47
Reading answers RX: REQ, ID:0, Q:0, A:1, NS:0, ADD:0
Not expecting any answers right now, returning
:urn 47
Reading answers RX: REQ, ID:0, Q:0, A:1, NS:0, ADD:0
Not expecting any answers right now, returning
:urn 515
:urd 8, 515, 13
:urd 4, 515, 22
:urd 5, 515, 27
ERR_NO_SERVICE: airport
:urn 400
:urd 39, 400, 13
ERR_NO_HOST: Samsung M2835 @ <my computer name>
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 400
:urd 39, 400, 13
ERR_NO_HOST: Samsung M2835 @ <my computer name>
hostname: esp_e1a6cf
instance: ESP_E1A6CF
:urn 833
:urd 8, 833, 13
:urd 4, 833, 22
:urd 5, 833, 27
ERR_NO_SERVICE: airport
:urn 1451
Reading answers RX: REQ, ID:0, Q:0, A:18, NS:0, ADD:1
Not expecting any answers right now, returning
:urn 380
Reading answers RX: REQ, ID:0, Q:0, A:9, NS:0, ADD:1
Not expecting any answers right now, returning

// stops here (forever?), unless there is a change in services 

```

igrr, 2016-06-15:
/сс @larsenglund

brendanmatkin, 2016-06-15:
Followup: this library can find the above mentioned services on  my network, on the ESP, although it is pretty raw: https://github.com/mrdunk/esp8266_mdns

larsenglund, 2016-06-15:
The library does indeed only support "ideal" responses with one of each of the four response types in the packet. That was all I needed at the moment so I didn't implement handling of more complex or segmented answers.

larsenglund, 2016-06-15:
Sift through the two RFCs linked in the code to get a better understanding of the packets that get sent around when making requests. Also sniffing a request and answer with something like Wireshark is usefull if one would like to add support for parsing more types of responses.

lorneb, 2016-06-24:
What I found was that 4 responses were needed ( PTR,TXT,SRV,A ) before an answer was created.
In my case the TXT is optional.

If can craft your replies to have this then it works,  if not you need to modify the code.

ESP8266mDNS.cpp 
```C
//around 480
   if (numAnswers < 3 || numAnswers > 4 ) {

//around line 509
	int answerCount = numAnswers;

//around line 621

if ( ( 
			  (partsCollected == 0x0F && answerCount == 4) || 
			  (partsCollected == 0x0D && answerCount == 3)
			)	
			&& serviceMatch) {

Hope that helps

me-no-dev, 2016-07-28:
Can you guys test [this PR](https://github.com/esp8266/Arduino/pull/2333)?
```
mDNS query _arduino._tcp
2 service(s) found
1: (192.168.254.182:8266) esp-test
2: (192.168.254.156:22) piduino1
mDNS query _http._tcp
2 service(s) found
1: (192.168.254.182:80) esp-test
2: (192.168.254.194:80) HP LaserJet CP1025nw
mDNS query _ipp._tcp
1 service(s) found
1: (192.168.254.194:631) HP LaserJet CP1025nw
mDNS query _afpovertcp._tcp
1 service(s) found
1: (192.168.254.49:548) Hristo's Time Capsule
mDNS query _smb._tcp
1 service(s) found
1: (192.168.254.49:445) Hristo's Time Capsule
mDNS query _ftp._tcp
0 service(s) found
```

me-no-dev, 2016-07-28:
@lorneb how is your TXT optional? Your device should return TXT even if empty

lorneb, 2016-07-31:
@me-no-dev  I was mucking around with MDNS,  when testing I was not sending a TXT record.  I should have read the RFC first as you are correct a TXT record is NOT optional.  

I'm sending TXT records now :-)

devyte, 2017-08-30:
Hi, what's the status of this? Is the issue resolved? was a PR made and/or merged?

devyte, 2018-01-03:
The referenced PR was merged. 
Closing as no feedback in over 2 months.

closed 2018-01-03 by devyte