esp8266/Arduino #2152: Digital pin reset

markusschweitzer, 2016-06-15:
### Basic Infos

#### Hardware
Hardware:		esp-03
Core Version:      	latest

### Description

Hi,
should the digital pin be resettet everytime?
Now when i set it to HIGH (connect it to 3.3v), it stays high until I set it to LOW (ground).

Is it possible to set it to LOW everytime, and only switch to HIGH if 3.3v are connected?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:	       qio
Flash Frequency:   40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <Arduino.h>

void setup() {
Serial.begin(115200);
pinMode(12, INPUT);
}

void loop() {
Serial.println(digitalRead(12));
}
```

igrr, 2016-06-15:
You can add a pull-down resistor (10k, for example) to achieve this.

markusschweitzer, 2016-06-15:
Thanks for the fast answer. Isn't there one included? (like in the arduinos. eg nano)

igrr, 2016-06-15:
Well you can use `pinMode(12, INPUT_PULLUP)`, but that will work other way around — return 1 unless you pull pin low.
What you need is `INPUT_PULLDOWN`, but ESP8266 doesn't have internal pull-downs on GPIOs, except for GPIO16.

closed 2016-06-15 by markusschweitzer