esp8266/Arduino #2155: Services only bind to AP when in WIFI_AP_STA mode !

pgScorpio, 2016-06-15:
Core Version:      	2.2.0

When both AP and STA are enabled. All services like mdns, webserver and FTP server only bind to the AP interface, rendering the STA connection almost useless.

At least it should be possible to select which service binds to which interface....
Even better if it would be possible to enable services on both interfaces.

igrr, 2016-06-16:
For web server, you can set IP address to listen on as follows:

```c++
ESP8266WebServer server;

void setup() {
  // connect to WiFi
  // get STA ip
  IPAddress ip = WiFi.localIP();
  // bind server to this address and port 80
  server = ESP8266WebServer(ip, 80);
  // initialize server, add handlers
  // call server.begin();
}

Regarding binding to both interfaces, this needs some investigation.

mkeyno, 2016-06-16:
@igrr  when we create webserver instant as ` ESP8266WebServer server(80)` , it is equal to `ESP8266WebServer(WiFi.localIP(), 80);`

pgScorpio, 2016-06-16:
So the binding can only be set at declaration time ???

I am reading my configuration from eeprom...., so the needed parameters are not yet available at declaration time. 

This means I would have to declare the server dynamically, and so having to use a pointer (having to change all references in code) .

I've looked into the code and, as far as I can see, the parameters are just stored and not used until begin(), but there is no way to change them between declaration and begin. The same problem goes for any server since WifiServer (used by all servers) also has this limitation....

Why can't we give the parameters at begin(...) or change them somehow  before begin() ?

Also when (re-)connecting to an AP and using dhcp the IP address is unknown until connected,
and therefore the IP address of the server should be changed accordingly....

**BTW**: servers should not bind with an ip-address, but they should bind to an **interface** and specify it's **variable** part of the ip address, so the server.begin() should be: server.begin(WIFI_AP, {0.0.0.2}, 80), and the interface will fill in the fixed part of it's ip-address according it's netmask.
And since the interface should (re-)apply it's ip and netmask every time they change, you don't have any ip-address change problems this way...

For now it would probably be the best solution to wrap the services in a separate class so I can setup and connect wifi before starting them, and stop them when disconnected  (destruct  the "services" class, and then re-create the "services" class when connected again). But in this case I have another problem since the Wifi classes do not have appropriate destructors.... (i.e. any open ClientContext's will **not** be deleted.)

igrr, 2016-06-16:
Not at declaration time, but at definition time.
You can create server object once you have loaded IP address from EEPROM, just assign new instance to server variable
`server = WiFiServer(ip, 80);`
This can be done anywhere in the code, like in setup or loop.

By default server pcb's bind to INADDR_ANY, so they actually should be accessible from all interfaces. The fact that they aren't means that there may be a bug somewhere. Need to go through LwIP code and check how listening pcb's are handled w.r.t. packets coming from different interfaces.

pgScorpio, 2016-06-17:
Thanks igrr !

 I finally have got the webserver working on both interfaces when in AP+STA mode....

But I still can't get MDNS working on AP. 
At least 
`MDNS.begin("myDeviceName", IPAddress(IPADDR_ANY));` 
does not work... (mdns still only on STA)

And the FTP server (FTP_SERVER_VERSION "FTP-2016-01-14") only works half on AP, you can log in, but the server reports the wrong ip for the data connection:

```
Status:	Connecting to 192.168.100.1:21...                      <<<<<< this is the AP ip
Status:	Connection established, waiting for welcome message...
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Retrieving directory listing of "\"...
Command:	CWD \
Response:	250 Ok. Current directory is /
Command:	PWD
Response:	257 "/" is your current directory
Command:	TYPE I
Response:	200 TYPE is now 8-bit binary
Command:	PASV
Response:	227 Entering Passive Mode (192,168,19,20,195,89).   <<<<<< this is the STA ip!!!
Command:	MLSD
Response:	425 No data connection MLSD
Error:	Failed to retrieve directory listing
Response:	530 Timeout
Response:	221 Goodbye
Error:	Connection closed by server
```

PS:
I changed ESP8266FtpServer.cpp line 272
`dataIp = WiFi.localIP();`
to:
`dataIp = client.localIP();`
Now it works...

igrr, 2016-06-20:
ESP8266FtpServer library is not maintained here, so can not help you on that one.

Regarding mDNS library, currently it ignores the IP address you pass in. Furthermore UDP multicasts are not sent to all interfaces even if you bind to INADDR_ANY, they are only sent to the default interface, which is AP interface in AP+STA mode. This is something which has to be changed in LwIP stack.

syndycat, 2016-08-17:
Hello, 

I read the above comments but I didn't found the solution. I have NodeMCU v2 and I built a sketch where the wifi is configured WIFI_AP_STA. For AP I set IP configuration to be 192.168.50.1 and for client connection the DHCP is allocating 192.168.20.1. 

The WifiServer is set as (WiFiServer server(80)) ... When I try to get data from 192.168.50.2 network (a mobile device connected to NodeMCU)  is working but from the connected netw (another mobile device connected to my home Wifi where the NodeMCU is connected)  is not working. How you guys work out this issue? 

Thank you,
Cata

pgScorpio, 2016-08-18:
@syndycat

Check your netmask...
If dhcp provides a netmask of 255,255,255,0 the ip 192.168.50.0 is invalid, and you should use an  ip in the range of 192.168.20.x or change the netmask of your dhcp server to 255.255.0.0

pgScorpio, 2016-08-18:
P.s. @syndycat 
This is not related to this issue....
If you need further assistance please check/ask in the esp8266 forums...

syndycat, 2016-08-18:
Thanks pgScorpio for the answer. The AP mask is 255.255.0.0. (sorry for omitting this info). 
I tried also ESP8266WebServer and I have the same issue: is available only from AP connection and not from CNT connection.

swarren, 2017-01-15:
https://github.com/esp8266/Arduino/pull/2864 helps MDNS with both AP+STA active.

devyte, 2017-10-16:
As far as I can tell, the webserver and mDNS are working correctly in latest git (mdns via a notify method that needs to get called on wifi events).
Closing as resolved.

closed 2017-10-16 by devyte