esp8266/Arduino #2156: Adafruit ESP8266E12 HUZZAH. Does it work with Arduino IDE 1.6.9 ?

bphick, 2016-06-15:
Back between Arduino IDE 1.6.5 and 1.6.6, we lost the ability to talk to the Adafruit ESP8266 Huzzah board.  Figuring they would fix the problem in subsequent releases, I upgraded from 1.6.5 to 1.6.9 this week, did all the library and boards updates, and when trying to compile for the Huzzah, following the settings from https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide.  (and I DO have the board selected as the Adrafruit HUZZAH ESP8266, and not the generic one initially shown in the tutorial, but have tried both with no differing results. 

I get the following compiling error:

Board huzzah (platform esp8266, package esp8266) is unknown

Error compiling for board Adafruit HUZZAH ESP8266.

Do I need to revert back to A. IDE 1.6.5 to be able to talk to my boards again??  This was to be a quick code update.  If it matters, I am using Windows 7.

Juppit, 2016-06-15:
Try to rename your arduino directory in something like Arduino15.old and 
reload it.
You should find it in C:\Users\[user_name]\AppData\Local\Arduino15

bphick, 2016-06-15:
Thanks for the quick reply.  Tried a quick fix. Renamed it to C:\Program Files (x86)\Arduino2
and tried booting the Arduino.exe.  No change in my result.  I still get:
 I get the following compiling error:

 Board huzzah (platform esp8266, package esp8266) is unknown

 Error compiling for board Adafruit HUZZAH ESP8266.

I tried my backup of 1.6.5 and no such error, however it is telling me that it doesn't like something I compiled before and was warned about back in January/16.  Which was the use of a const Char *.  But that is unrelated to the 1.6.6 - 1.6.9 problem I am experiencing.

igrr, 2016-06-16:
I think you need to rename the folder @Juppit mentioned, not the one you renamed...
`C:\Users\[user_name]\AppData\Local\Arduino15` not `C:\Program Files (x86)\...`.
The thing in Program Files is the application. The other folder (under AppData) is application data.

bphick, 2016-06-16:
Oops, sorry I misunderstood the first time. Your solution worked. I was not aware of the files in the User directory and naturally gravitated towards the one's I had been manipulating prior to my request.
Your help is very much appreciated!  I'm up and running.

closed 2016-06-16 by igrr