esp8266/Arduino #2161: HTTPS GET Request got Exception (3)

HTDemon, 2016-06-17:
### Basic Infos

#### Hardware
Hardware:		ESP-12f
Core Version:      	2.3.0-rc2

### Description

An issue about connect to Slack HTTPS API.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

I tried http.begin(String url, String httpsFingerprint) and http.begin(String host, uint16_t port, String uri, bool https, String httpsFingerprint)

### Debug Messages

debug log:
> [HTTP-Client][begin] url: https://slack.com/api/rtm.start?token=...
[HTTP-Client][begin] host: slack.com port: 443 url: /api/rtm.start?token=...
[HTTP-Client][begin] httpsFingerprint: ....
[hostByName] request IP for: slack.com
[hostByName] Host: slack.com IP: ....
[HTTP-Client] connected to slack.com:443
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/json; charset=utf-8'
[HTTP-Client][handleHeaderResponse] RX: 'Transfer-Encoding: chunked'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Origin: *'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: private, no-cache, no-store, must-revalidate'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Security-Policy: referrer no-referrer;'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Fri, 17 Jun 2016 08:09:33 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Expires: Mon, 26 Jul 1997 05:00:00 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Pragma: no-cache'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache'
[HTTP-Client][handleHeaderResponse] RX: 'Strict-Transport-Security: max-age=31536000; includeSubDomains; preload'
[HTTP-Client][handleHeaderResponse] RX: 'Vary: Accept-Encoding'
[HTTP-Client][handleHeaderResponse] RX: 'X-Accepted-OAuth-Scopes: rtm:stream,client'
[HTTP-Client][handleHeaderResponse] RX: 'X-Content-Type-Options: nosniff'
[HTTP-Client][handleHeaderResponse] RX: 'X-OAuth-Scopes: identify,read,post,client,apps'
[HTTP-Client][handleHeaderResponse] RX: 'X-XSS-Protection: 0'
[HTTP-Client][handleHeaderResponse] RX: 'X-Cache: Miss from cloudfront'
[HTTP-Client][handleHeaderResponse] RX: 'Via: 1.1 35436fa36ed8ec21d3998280f7181e8b.cloudfront.net (CloudFront)'
[HTTP-Client][handleHeaderResponse] RX: 'X-Amz-Cf-Id: a6Z4JVv6s10ya6L3I3C0xpLr6tgk-JLmjA2aJrUA62148qwMv-A9dQ=='
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] Transfer-Encoding: chunked
[HTTP-Client][end] still data in buffer (2121), clean up.
check_poison_block is called for free block 0x3fff5b48
Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40039030 depc=0x00000000

> [HTTP-Client][begin] host: slack.com port: 443 url: https://slack.com/api/rtm.start?token=..... httpsFingerprint: ab f0 5b a9 1a e0 ae 5f ce 32 2e 7c 66 67 49 ec dd 6d 6a 38
[hostByName] request IP for: slack.com
[hostByName] Host: slack.com IP: 54.230.215.48
[HTTP-Client] connected to slack.com:443
check_poison_block is called for free block 0x3fff5b20
Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40039030 depc=0x00000000
ctx: cont 
sp: 3fff32f0 end: 3fff3710 offset: 01a0

igrr, 2016-06-20:
I was not able to reproduce this with git version.
Please find my test sketch below.
Could you please provide the complete sketch which is failing for you?
```c++
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "#!/bin/test";
const char* password = "";

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  String completeURI = "https://slack.com/api/rtm.start?token=0";
  HTTPClient http;
  http.begin(completeURI, "ab f0 5b a9 1a e0 ae 5f ce 32 2e 7c 66 67 49 ec dd 6d 6a 38");
  Serial.println(http.GET());
}

void loop() {
}
```

Output was:
```
¿.?å.ú;...rﬂ€ín¿..‡b«.í`.ƒû¿l...á..á√å.r..¸Çnƒb‡
connecting to #!/bin/test
sta config unchangedscandone
f 0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with #!/bin/test, channel 6
dhcp client start...
.ip:192.168.77.105,mask:255.255.255.0,gw:192.168.77.1
.
WiFi connected
IP address: 
192.168.77.105
:ref 1
please start sntp first !
State:.sending Client Hello (1)
:wr
:sent 72
:rn 1460
:ww
:rd 5, 1460, 0
:rdi 1460, 5
:rd 48, 1460, 5
:rdi 1455, 48
State:.receiving Server Hello (2)
:rd 5, 1460, 53
:rdi 1407, 5
:rd 1402, 1460, 58
:rdi 1402, 1402
:c0 1402, 1460
:rn 993
:rd 984, 993, 0
:rdi 993, 984
State:.receiving Certificate (11)
:rd 5, 993, 984
:rdi 9, 5
:rd 4, 993, 989
:rdi 4, 4
:c0 4, 993
State:.receiving Server Hello Done (14)
State:.sending Client Key Exchange (16)
:wr
:sent 267
:ww
:wr
:sent 6
:ww
State:.sending Finished (16)
:wr
:sent 69
:rn 75
:ww
:rd 5, 75, 0
:rdi 75, 5
:rd 1, 75, 5
:rdi 70, 1
:rd 5, 75, 6
:rdi 69, 5
:rd 64, 75, 11
:rdi 64, 64
:c0 64, 75
State:.receiving Finished (16)
:ref 2
domain name: 'slack.com'
SAN 0: '*.slack.com', no match
:ur 2
:wr
:sent 213
:ww
:rn 693
:rd 5, 693, 0
:rdi 693, 5
:rd 688, 693, 5
:rdi 688, 688
:c0 688, 693
:wcs ra 645:rn 53
:rch 53, 53
:rcl
:abort
200
:ur 1
:del
Alert: close notify
pm open,type:2 0
```

HTDemon, 2016-06-20:
I referenced sample code from https://github.com/urish/arduino-slack-bot/blob/master/slackbot/slackbot.ino and modify it to this
```
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WebSocketsClient.h>
#include <ArduinoJson.h>
#define SLACK_BOT_TOKEN "xoxb-"
#define WORD_SEPERATORS "., \"'()[]<>;:-+&?!\n\t"

//
WiFiServer server(80);
//
ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
//
int setTemp;
int setWindSpeed;
int setACStatus;
long nextCmdId = 1;
bool connected = false;
const char* ssid = "#!/bin/test";
const char* password = "";

// Sends a ping message to Slack. Call this function immediately after establishing
// the WebSocket connection, and then every 5 seconds to keep the connection alive.
void sendPing() {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["type"] = "ping";
  root["id"] = nextCmdId++;
  String json;
  root.printTo(json);
  webSocket.sendTXT(json);
}

void processSlackMessage(char *payload) {
  char *nextWord = NULL;
  for (nextWord = strtok(payload, WORD_SEPERATORS); nextWord; nextWord = strtok(NULL, WORD_SEPERATORS)) {
    if (strcasecmp(nextWord, "a25") == 0) {
      setTemp = 25;
      setWindSpeed = 1;
      setACStatus = 1;
    }
  }
}

/**
  Called on each web socket event. Handles disconnection, and also
  incoming messages from slack.
*/
void webSocketEvent(WStype_t type, uint8_t *payload, size_t len) {
  switch (type) {
    case WStype_DISCONNECTED:
      Serial.printf("[WebSocket] Disconnected :-( \n");
      connected = false;
      break;

    case WStype_CONNECTED:
      Serial.printf("[WebSocket] Connected to: %s\n", payload);
      sendPing();
      break;

    case WStype_TEXT:
      Serial.printf("[WebSocket] Message: %s\n", payload);
      processSlackMessage((char*)payload);
      break;
  }
}

bool connectToSlack() {
  // Step 1: Find WebSocket address via RTM API (https://api.slack.com/methods/rtm.start)
  HTTPClient http;
  //http.begin("https://slack.com/api/rtm.start?token=" SLACK_BOT_TOKEN);
  http.begin("slack.com", 443, "https://slack.com/api/rtm.start?token=" SLACK_BOT_TOKEN, true, "ab f0 5b a9 1a e0 ae 5f ce 32 2e 7c 66 67 49 ec dd 6d 6a 38");
  int httpCode = http.GET();
  http.end();
  
  if (httpCode != HTTP_CODE_OK) {
    Serial.printf("HTTP GET failed with code %d\n", httpCode);
    return false;
  }

  WiFiClient *client = http.getStreamPtr();
  client->find("wss:\\/\\/");
  String host = client->readStringUntil('\\');
  String path = client->readStringUntil('"');
  path.replace("\\/", "/");

  // Step 2: Open WebSocket connection and register event handler
  Serial.println("WebSocket Host=" + host + " Path=" + path);
  webSocket.beginSSL(host, 443, path, "", "");
  webSocket.onEvent(webSocketEvent);
  return true;
}

void setup() {
  configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");
  Serial.begin(115200);
  // Connect to your AP
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFiMulti.addAP(ssid, password);
  while (WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println(("WiFi connected"));

  // Start the server
  server.begin();
  Serial.println("Server started");

  // Print the IP address
  Serial.println(WiFi.localIP());

  connectToSlack();
}

unsigned long lastPing = 0;

//
void loop()
{
  unsigned long currentMillis = millis();
  if (connected) {
    // Send ping every 5 seconds, to keep the connection alive
    if (currentMillis - lastPing > 5000) {
      sendPing();
      lastPing = millis();
    }
  } else {
    // Try to connect / reconnect to slack
    connected = connectToSlack();
    if (!connected) {
      delay(500);
    }
  }
}
```

Seems it will print exception when using fingerprint parameter.

igrr, 2016-06-20:
You attempt to use `http.getStreamPtr()` after doing `http.end();`.
`http.end()` closes the connection, so `getStreamPtr()` returns a null pointer, which you dereference without checking. I have removed `http.end()` line and it no longer crashes for me where it used to crash (on `client->find("wss:\\/\\/")` line). It doesn't do anything useful of course because I don't have a valid token (`host` and `path` variables end up empty).

devyte, 2017-10-16:
Closing due to user error.

closed 2017-10-16 by devyte