esp8266/Arduino #2162: I2C hang with very slow I2C device (pic12f840@500khz)

far5893, 2016-06-17:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-07
Core Version:      	2.2.0

### Description

On very slow device i2c hang SDA and give strange results.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K/64K
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz?
Upload Using:		SERIAL?
Reset Method:		ck

### Sketch
Wire.begin(12,13); //I2C pin
Wire.setClockStretchLimit(2000);

In my software(l is not important) : 
	Wire.beginTransmission(Addr);
	Wire.write(0b00111100);
	Serial.println("I2C write---"+String(l));
	Serial.println(Wire.endTransmission(1));

Serial Output:

I2C write---4
3

[image]

me-no-dev, 2016-06-20:
any particular reason why your slave does not release SDA?

far5893, 2016-06-20:
I disconnect physically only sda from device but bus remain low .

devyte, 2017-10-16:
@igrr this has milestone 2.4.0. However, I don't fully understand the issue, let alone what needs to be fixed in Wire.
Is the issue still relevant?

far5893, 2017-10-16:
If you have different clock stretch need on devices  is relevant!

enjoyneering, 2017-11-07:
check this - https://github.com/enjoyneering/ESP8266-I2C-Driver

d-a-v, 2018-05-24:
For reference ([link](https://github.com/esp8266/Arduino/issues/2330#issuecomment-391303255))
I will make a PR for that

devyte, 2019-11-08:
CC @Tech-TX

closed 2019-12-01 by devyte