esp8266/Arduino #2164: can upload in SPIFF , can't upload sketch

mkeyno, 2016-06-17:
#### Hardware
Hardware:			?ESP-12F
Core Version:      	?2.2.0-rc2?

### Description

 I made breakout board with ESP12F, after assembly the board , I could upload sketches couple of times ,but some garbage and exception error things come out in my last tries, but now, I can’t upload any sketch to my module whereas easily can upload files in the SPIFF,
I thought it was hardware problem and related to UART resistor divider but as I said I could upload files in any flash size configuration but received following error when I try upload the any sketch
`
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
    setting board to ck
    setting baudrate from 115200 to 115200
    setting port from COM1 to COM23
    setting address from 0x00000000 to 0x00000000
    espcomm_upload_file
    espcomm_upload_mem
    setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
    flush start
    setting serial port timeouts to 1 ms
    setting serial port timeouts to 1000 ms
    flush complete
    espcomm_send_command: sending command header
    espcomm_send_command: sending command payload
    read 0, requested 1
trying to connect
    flush start
    setting serial port timeouts to 1 ms
    setting serial port timeouts to 1000 ms
    flush complete
    espcomm_send_command: sending command header
    espcomm_send_command: sending command payload
    espcomm_send_command: receiving 2 bytes of data
    espcomm_send_command: receiving 2 bytes of data
    espcomm_send_command: receiving 2 bytes of data
    espcomm_send_command: receiving 2 bytes of data
    espcomm_send_command: receiving 2 bytes of data
    espcomm_send_command: receiving 2 bytes of data
    espcomm_send_command: receiving 2 bytes of data
    espcomm_send_command: receiving 2 bytes of data
Uploading 325376 bytes from to flash at 0x00000000
    erasing flash
    size: 04f700 address: 000000
    first_sector_index: 0
    total_sector_count: 80
    head_sector_count: 16
    adjusted_sector_count: 64
    erase_size: 040000
    espcomm_send_command: sending command header
    espcomm_send_command: sending command payload
    setting serial port timeouts to 15000 ms
    read 0, requested 1
    setting serial port timeouts to 1000 ms
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
    flush start
    setting serial port timeouts to 1 ms
    setting serial port timeouts to 1000 ms
    flush complete
error: espcomm_upload_mem failed
`
### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		all possible configuration 
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

mkeyno, 2016-06-21:
dear @igrr  do you have any idea , why  I can upload the sketch files whereas can't upload sketch on my module? is it broke? is there any way to restore?

devyte, 2017-10-16:
Not the right place to debug your project. Closing per #3655 .

closed 2017-10-16 by devyte