esp8266/Arduino #2166: trying to work with RF24 and nodemcu having problem sending

dudilugasi, 2016-06-18:
### Basic Infos

#### Hardware
Hardware:			?ESP-12? esp8266MOD
Core Version:      	?2.1.0-rc2?

### Description
i am trying to listen to a message but also be able to send but when im stop listening i am getting a reset.

### Settings in IDE

Module:  			?Generic ESP8266 Module? NodeMCU 1.0 (ESP 12E - Module)
Flash Size: 		?4MB/1MB? 4MB
CPU Frequency:		?80Mhz?  80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL? SERIAL
Reset Method:		?ck / nodemcu? nodemcu

### Sketch

#include <Arduino.h>
#include <RF24.h>

RF24 radio(2, 15);

const byte rxAddr[6] = "00001";
const byte rxAddr2[6] = "00002";

void setup()
{
    Serial.begin(115200);
    radio.begin();
    radio.setRetries(15, 15);
    radio.openWritingPipe(rxAddr);
    radio.openReadingPipe(1, rxAddr2);
    radio.printDetails();

    radio.startListening();

}

void loop()
{

    char text[32]="hello";
    Serial.println(text);

   radio.stopListening();
   bool ok =  radio.write(&text, sizeof(text));
   if(ok){
     Serial.println("send seccess");
   }
   else{
        Serial.println("send failed");
  }
  radio.startListening();

  delay(1000);
}

### Debug Messages

Soft WDT reset

ctx: cont 
sp: 3ffefaf0 end: 3ffefd40 offset: 01b0

>>>stack>>>
3ffefca0:  3fffdad0 000000ff 3ffeebf8 4020250c  
3ffefcb0:  3fffdad0 000000ff 3ffeebf8 40202737  
3ffefcc0:  3fffdad0 00000030 3ffeebf8 40202788  
3ffefcd0:  00000002 00000030 3ffeebf8 40202a0a  
3ffefce0:  3fffdad0 3ffeecec 3ffeebf8 40202a4f  
3ffefcf0:  0000001d 3ffeecec 3ffeebf8 40202440  
3ffefd00:  6c6c6568 726f6877 00000065 00000000  
3ffefd10:  00000000 00000000 00000000 00000000  
3ffefd20:  feefeffe 00000000 3ffeed09 40203634  
3ffefd30:  feefeffe feefeffe 3ffeed20 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

kireol, 2016-08-19:
Same here.

nodemcu esp8266 12e
arduino ide 1.6.8
rf24 1.1.7
esp8266 library 1

devyte, 2017-10-16:
@dudilugasi assuming this issue is still valid, I suggest requesting help from the lib author first. If an issue in the core is found, please open a new issue with the relevant details.
Closing.

closed 2017-10-16 by devyte