esp8266/Arduino #2168: Newbie Const Char * question

bphick, 2016-06-18:
As a total neophyte newbie I spent a week trying to get my code to work in multiple locations.  I’m sure a more experienced person will find this an easy fix and any help would be very much appreciated.  Be nice, I’m a newbie…

In January, this code would work with a warning about “const char *” not being good code… but it compiled and worked great.  Alas, now I need to upgrade a couple of Adafruit ESP8266 Huzzah’s with this code but it won’t compile.  I put the whole code here, but have highlighted the offending parts of the code between 2 sections marked by something like: **//  _______________________  NEED HELP STARTING HERE __________________” or ENDING similar.**

It says “Cannot convert ‘String’ to Const Char* in assignment:

Arduino: 1.6.9 (Windows 7), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\Owner\Documents\Arduino\Huzzah_Sensor_MultiWiFiLocation_Unit_wPIR_Photo_1_1_16\Huzzah_Sensor_MultiWiFiLocation_Unit_wPIR_Photo_1_1_16.ino: In function 'void setup()':

Huzzah_Sensor_MultiWiFiLocation_Unit_wPIR_Photo_1_1_16:146: error: cannot convert 'String' to 'const char*' in assignment

     WLAN_SSID = WiFi.SSID();

               ^
exit status 1
cannot convert 'String' to 'const char*' in assignment

_______________________________________________________________________________

/************** Starts here*************************************
  PIR and Light sensor added to this version.  Publishes data to 4 feeds every 10 sec.
  and also looks for up to 6 WiFi Networks to log into.
  
  Adafruit ESP8266 Sensor Module
  
  Must use ESP8266 Arduino from:
    https://github.com/esp8266/Arduino
  Works great with Adafruit's Huzzah ESP board:
  ----> https://www.adafruit.com/product/2471
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!
  Written by Tony DiCola for Adafruit Industries.
  MIT license, all text above must be included in any redistribution

Modified for multi-location by Bryan Hickman, a hobbyist and a wantabie. 
 ****************************************************/

// Libraries
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include "DHT.h"

// DHT 22 sensor
#define DHTPIN 5
#define DHTTYPE DHT22 

**//  _______________________  *NEED HELP STARTING HERE *__________________**

// WiFi parameters
const char * WLAN_SSID;
const char * WLAN_PASS;

     // WiFi's and followed by their passwords
const char * myWLAN_SSIDStrings[]={"Location 1", "Location2", "Location3", "Location4", "Location5","Location6"};   // I tried the unprotected xfinitywifi router with no security password, and while it connects to the web, it won't connect to Adafruit.IO
const char * myWLAN_PASSStrings[]={"PW_Location1", " PW_Location2", " PW_Location3"," PW_Location4"," PW_Location5"," PW_Location6"};
    // Particulars on what you said regarding the WiFi Arrays above.
int nbrOfWiFis = 6;  // ******Number of WiFi's to look for sequentially
int StartOnWiFiNo = 3;  //Which one do you want it to start on before sequentially looking through the WiFi Array.  Helps if you know location where unit will spend majority of its time.

**//  _________ * NEED HELP ABOVE HERE and a similarly marked section BELOW*___________**

// Adafruit IO
#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883
#define AIO_USERNAME    "My_UserName"
#define AIO_KEY         "MyAIO_Key"

[... 205 lines cut ...]
  int8_t ret;

  while ((ret = mqtt.connect()) != 0) {

    switch (ret) {
      case 1: Serial.println(F("Wrong protocol")); break;
      case 2: Serial.println(F("ID rejected")); break;
      case 3: Serial.println(F("Server unavail")); break;
      case 4: Serial.println(F("Bad user/pass")); break;
      case 5: Serial.println(F("Not authed")); break;
      case 6: Serial.println(F("Failed to subscribe")); break;
      default: Serial.println(F("Connection failed")); break;
    }

    if(ret >= 0)
      mqtt.disconnect();

    Serial.println(F("Retrying connection..."));
    delay(5000);

  }

  Serial.println(F("Adafruit IO Connected!"));

}

igrr, 2016-06-20:
Dear @bphick, this issue tracker is not a good place for coding questions. Its main purpose is to keep track of issues and feature requests related to esp8266 Arduino core and libraries. Please use esp8266.com community forum for coding help. Thank you.

closed 2016-06-20 by igrr

greg-ware, 2016-06-20:
@bphick, if you ask on esp8266.com community, post the link here.
(@igrr, BTW excellent job you're doing on this git, kudos!)
Greg