esp8266/Arduino #2169: MDNS queryservice results in crashes and wdts

sticilface, 2016-06-19:
Using MDNS.queryservice results in wdt when there are more than one (maybe 2+) esps advertising and querying services on the network.  average uptime is about 1 min. 

advertising using this
```
MDNS.addService("melvanimate", "tcp", 8270);
```

searching using this in loop
```
                static uint32_t mdns_search_timeout = 0;

                if (millis() - mdns_search_timeout > 10000) {
                        mdns_search_timeout = millis();

                        int number = MDNS.queryService("melvanimate", "tcp");

                        if (number > 0) {

                                for (uint8_t result = 0; result < number; result++) {
                                        int port = MDNS.port(result);
                                        String host = MDNS.hostname(result);
                                        IPAddress IP = MDNS.IP(result);
                                        DebugMelvanimatef("Service Found [%u] %s (%u.%u.%u.%u) port = %u\n", result, host.c_str(), IP[0], IP[1], IP[2], IP[3], port);
                                }
                        }
                }
        }
``` 

several issue i think here:
1)  1 seconds time out waiting for mDNS responses. https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/ESP8266mDNS.cpp#L355
should maybe be moved async and handled in the handle(), and packets dealt with as they arrive and removed from the buffer, to stop the buffer being filled before processing starts.. 
2)  when one device queries service all ESPs get the response, so when the are multiple devices querying the service.. you get your own DoS attack against ourselves...  a few stats...  4esps, with one device performing queryService every 10 seconds, generates 3 packets every 10 seconds.  the same devices all performing a query service every 10 seconds results in 490 packets in 540 seconds.  this also represents all the resets that occur... randomly but usually between 5 - 60 seconds.  not sure if this is the cause or not... time between incoming packets it also very narrow...  0, 0.42 and +1.42 ms

interestingly there are stack traces occurring within the wdt resets... that are in _parsePacket()

```
0x40224209: MDNSResponder::_parsePacket() at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:656
0x40224015: MDNSResponder::_parsePacket() at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:539
0x4000b62e: ?? ??:0
0x4000b6c4: ?? ??:0
0x40107366: NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::i2s_slc_isr() at /Users/amelvin/Dropbox/Sketchbook/libraries/Melvanimate/src/Melvanimate.cpp:234
0x4000a333: ?? ??:0
0x4000050c: ?? ??:0
0x40102202: pp_post at ??:?
0x401054df: lmacRxDone at ??:?
0x40107366: NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::i2s_slc_isr() at /Users/amelvin/Dropbox/Sketchbook/libraries/Melvanimate/src/Melvanimate.cpp:234
0x4000050c: ?? ??:0
0x40102f9e: trc_NeedRTS at ??:?
0x40107366: NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::i2s_slc_isr() at /Users/amelvin/Dropbox/Sketchbook/libraries/Melvanimate/src/Melvanimate.cpp:234
0x40107366: NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::i2s_slc_isr() at /Users/amelvin/Dropbox/Sketchbook/libraries/Melvanimate/src/Melvanimate.cpp:234
0x4000050c: ?? ??:0
0x40107366: NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::i2s_slc_isr() at /Users/amelvin/Dropbox/Sketchbook/libraries/Melvanimate/src/Melvanimate.cpp:234
0x4000050c: ?? ??:0
0x4023194d: ieee80211_ht_updateparams at ??:?
0x40100300: umm_make_new_block$constprop$2 at umm_malloc.c:?
0x4000050c: ?? ??:0
0x40100459: check_poison_block at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:861
0x4010054f: check_poison_all_blocks at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:882
0x4020a0ec: UdpContext::_s_recv(void*, udp_pcb*, pbuf*, ip_addr*, unsigned short) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:225
0x401003ec: check_poison_block at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:835
0x4010054f: check_poison_all_blocks at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:882
0x402321cc: ieee80211_decap at ??:?
0x4010020c: _umm_free at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:1287
0x40106801: _UserExceptionVector_1 at ??:?
0x4000050c: ?? ??:0
0x4022b81b: pp_tx_idle_timeout at ??:?
0x40224545: MDNSResponder::update() at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:219
0x40229423: std::_Function_handler<void (), std::_Bind<std::_Mem_fn<void (MDNSResponder::*)()> (MDNSResponder*)> >::_M_invoke(std::_Any_data const&) at /Users/amelvin/Arduino/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional:2073
0x4020a118: UdpContext::_s_recv(void*, udp_pcb*, pbuf*, ip_addr*, unsigned short) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:225
0x4024a2a4: udp_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/udp.c:343
0x40107640: pvPortMalloc at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/heap.c:13
0x4024b9e4: ip_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip.c:553
0x4024ade9: ethernet_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c:1379
0x4022b81b: pp_tx_idle_timeout at ??:?
0x4022b78e: pp_tx_idle_timeout at ??:?
0x4023f86b: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
```

kaeferfreund, 2016-06-19:
have the same problem. Will also look into this...

igrr, 2016-06-20:
Possibly related to #2020?

igrr, 2016-06-20:
@sticilface Could you please enable debug logging in ESP8266mDNS.cpp and paste the output please?

sticilface, 2016-06-28:
Sorry i seemed to have missed your request for more info.  will do. (i actually wrote my own lib to do service discovery.. with a semi-sort of group back off feature to reduce traffic.)

me-no-dev, 2016-07-28:
Can you guys test [this PR](https://github.com/esp8266/Arduino/pull/2333)?
Let me know if you still get resets
```
mDNS query _arduino._tcp
2 service(s) found
1: (192.168.254.182:8266) esp-test
2: (192.168.254.156:22) piduino1
mDNS query _http._tcp
2 service(s) found
1: (192.168.254.182:80) esp-test
2: (192.168.254.194:80) HP LaserJet CP1025nw
mDNS query _ipp._tcp
1 service(s) found
1: (192.168.254.194:631) HP LaserJet CP1025nw
mDNS query _afpovertcp._tcp
1 service(s) found
1: (192.168.254.49:548) Hristo's Time Capsule
mDNS query _smb._tcp
1 service(s) found
1: (192.168.254.49:445) Hristo's Time Capsule
mDNS query _ftp._tcp
0 service(s) found
```

n8henrie, 2016-09-11:
I am still getting the wdt reset on first run followed by an unrecoverable crash loop after reset with the new mdns -- I have to re-flash to stop the crash loop.

devyte, 2017-10-16:
@sticilface The PR referenced above was merged. Is this issue still valid in latest git?

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte