esp8266/Arduino #2170: Exception 9

alxpy, 2016-06-19:
### Basic Infos

#### Hardware
Hardware:		ESP-01
Core Version:      	2.2

### Description

I connected my ESP-01 to the computer via USB-TTL converter, upload test script and got the error.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp
/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
[... 196 lines cut ...]
0x4021e588: dns_tmr at ?? line ?
0x40225ab4: tcp_timer_needed at ?? line ?
0x40105e98: _UserExceptionVector_1 at ?? line ?
0x40105ec8: _UserExceptionVector_1 at ?? line ?
0x40105ef0: _UserExceptionVector_1 at ?? line ?
0x40105f18: _UserExceptionVector_1 at ?? line ?
0x40225a24: tcp_zero_window_probe at ?? line ?
0x402105fc: cnx_connect_timeout at ?? line ?
0x40221004: ethernet_input at ?? line ?
0x40220c9c: etharp_output at ?? line ?
0x4020c024: ieee80211_output_pbuf at ?? line ?
0x40219ea8: wifi_station_dhcpc_event at ?? line ?
0x40221004: ethernet_input at ?? line ?
0x40220c9c: etharp_output at ?? line ?
0x4020c024: ieee80211_output_pbuf at ?? line ?
0x4021dd1c: node_remove_from_list at ?? line ?
0x40230720: chip_v6_unset_chanfreq at ?? line ?
0x40100030: _KernelExceptionVector at ?? line ?
0x40203484: fpm_attach at ?? line ?
0x4010502a: wdt_feed at ?? line ?
0x40202c95: __udivmoddi4 at /home/igrokhotkov/xtensa/crosstool-NG/.build/src/gcc-4.8.2/libgcc/libgcc2.c line 965
:  (inlined by) __umoddi3 at /home/igrokhotkov/xtensa/crosstool-NG/.build/src/gcc-4.8.2/libgcc/libgcc2.c line 1168

...
```

alxpy, 2016-06-19:
And with this code:
```cpp
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
```
I got:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

Exception (0):
epc1=0x402063f0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffff6e0 end: 3fffffb0 offset: 01a0

>>>stack>>>
(Decoding 33 results)
0x40101ae9: wdt_feed at ?? line ?
0x40101ae6: wdt_feed at ?? line ?
0x402169c1: tx_pwr_backoff at ?? line ?
0x40217f99: ram_rfcal_pwrctrl at ?? line ?
0x40217e85: ram_rfcal_pwrctrl at ?? line ?
0x4021699f: tx_pwr_backoff at ?? line ?
0x402168e8: cal_rf_ana_gain at ?? line ?
0x4021364a: tx_pwctrl_init_cal at ?? line ?
0x40213a43: tx_pwctrl_init at ?? line ?
0x40215298: chip_v6_initialize_bb at ?? line ?
0x402152ad: chip_v6_initialize_bb at ?? line ?
0x40211093: phy_init at ?? line ?
0x401024be: register_phy_ops at ?? line ?
0x40215f35: register_chipv6_phy at ?? line ?
0x40204d3a: wifi_station_set_default_hostname at ?? line ?
0x40204d25: wifi_station_set_default_hostname at ?? line ?
0x40201941: __wrap_register_chipv6_phy at /home/alx/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/cores/esp8266/core_esp8266_phy.c line 261
0x4022e040: rijndaelKeySetupEnc at ?? line ?
0x40101fe6: spi_flash_read at ?? line ?
0x40206103: system_get_sdk_version at ?? line ?
0x40206790: user_uart_wait_tx_fifo_empty at ?? line ?
0x4020676b: user_uart_wait_tx_fifo_empty at ?? line ?
0x40101d6e: wdt_feed at ?? line ?
0x40100030: _KernelExceptionVector at ?? line ?
0x40203484: system_upgrade_reboot at ?? line ?
0x4010502a: trc_NeedRTS at ?? line ?
0x40202c95: system_get_os_print at ?? line ?
<<<stack<<<

...
```

igrr, 2016-06-20:
Just ran a test with a NodeMCU board and with ESP-LAUNCHER, WiFiScan runs fine.

Usually such issues with ESP-01 come from failure to power and bootstrap the chip properly.
Make sure you are not using FTDI to supply 3.3V power to the ESP. An external power supply which is able to deliver at least 300mA is required. Having decoupling capacitors close to ESP01 pins is essential. GPIO0, GPIO2, and CH_PD need pull-up resistors to 3.3V (10k for GPIOs and 2k for CH_PD).
See notes in the documentation regarding running this software on bare ESP modules:
http://esp8266.github.io/Arduino/versions/2.2.0/doc/boards.html#generic-esp8266-modules

alxpy, 2016-06-21:
Thank you! Pull-up resistors are not helped, but I still use unstable power by USB. I'll try fix it...
Last question: I ran sketch which just reads the data from DHT, but I see prints of the old program in serial monitor. It can also be caused by an unstable connection?
Code:
```cpp
#include "DHT.h"

#define DHTPIN 2 
#define DHTTYPE DHT22  

DHT dht(DHTPIN, DHTTYPE);

void setup() { 
  Serial.begin(115200); 
  dht.begin(); 
}

void loop() {

  float h = dht.readHumidity(); 
  float t = dht.readTemperature();

  if (isnan(t) || isnan(h)) { 
    Serial.println("Failed to read from DHT"); 
  } 
  else { 
    Serial.print("Humidity: "); 
    Serial.print(h); 
    Serial.print(" %\t"); 
    Serial.print("Temperature: "); 
    Serial.print(t); 
    Serial.println(" *C");
  }

  delay(1000);
}
```
I got:
```
connected with HYIFI, channel 2
dhcp client start...
Humidity: 60.60 %	Temperature: 28.50 *C
Humidity: 60.60 %	Temperature: 28.50 *C
chg_B1:-40
Humidity: 61.40 %	Temperature: 29.60 *C
ip:192.168.0.83,mask:255.255.255.0,gw:192.168.0.1
Humidity: 61.40 %	Temperature: 29.60 *C
Humidity: 62.10 %	Temperature: 29.90 *C
Humidity: 62.10 %	Temperature: 29.90 *C
Humidity: 62.00 %	Temperature: 30.00 *C
Humidity: 62.00 %	Temperature: 30.00 *C
```
**HYIFI** - my wifi

igrr, 2016-06-23:
FYI, by default ESP8266 WiFi stack will save WiFi configuration to flash memory, and automatically reconnect to the same network on restart. This happens even if you aren't using WiFi functions.
You can disable persistence by adding `WiFi.persistent(false);` in the beginning of your WiFi initialization code. To disable WiFi in a sketch where you don't need it you can use `WiFi.mode(WIFI_OFF);`.

alxpy, 2016-06-23:
thank you very much!

closed 2016-06-23 by alxpy

iSevenDays, 2016-06-26:
@igrr thank you, it helped me too.

Without  WiFi.mode(WIFI_OFF); I have a lot of crashes (maybe because 1.5.3 SDK is used, not 1.5.4)

jbernibe, 2020-04-07:
I puted this code WiFi.persistent(false) in my sketch but dont working, still present the trouble : Exception (9):