esp8266/Arduino #2173: Esp mesh network

au190, 2016-06-19:
I want ot create a mesh network, communication behind the router. My question is is it secure this mesh network ? Can anyone hack the communication ?

Thx R.

Matthew-Bradley, 2016-06-28:
When you say they're behind a router, do you mean that they're communicating through the router by being all connected to it, or that they're all connected to each other? (and also the router?)

Regardless, If the router isn't connected to the internet, and the wifi network used is secure (WPA2 with strong password) I imagine it might be fairly secure. If the router has an internet connection, then the answer would probably involve firewalls and such.

Perhaps someone with more knowledge can answer this better?

battuashwik, 2016-07-12:
I have a router and two esp's are connected to it. Now for some reason one of the esp cannot connect to the router but can reach the other esp signal, so can it communicate to the router via the other esp ?

Makuna, 2016-09-11:
I think the original question is in reference to Esp8266WifiMesh library that is included.  If not then ignore the rest of this.

This mesh implementation uses the SoftAP features, and relies on each a SSID prefix to "participate"; but does NOT use any sort of security (no passwords support exposed to set password for the SoftAP).

So as implemented it is easy to hack by simply having a node nearby the mesh.

The Esp8266WifiMesh library is not really a good implementation; due to this, how synchronous it is written, and that it requires string messages (memory fragmentation from string class and binary data transmission requires conversion to text).

vitorsilvalima, 2017-03-08:
I think the implementation would have some features that are described in the mesh user guide from Espressif. It would be great to add support for password and to add methods that allows sending messages to specific nodes within the mesh rather than just sending to all nodes. The way it is implemented now, it looks more like a UDP network, where a message is sent to all listeners listening in a specific port. I would love to help to implement more specific methods for the mesh library.

sabas1080, 2017-03-08:
There is a library that works well with mesh
https://gitlab.com/BlackEdder/painlessMesh

devyte, 2017-03-08:
Not sure how relevant the following is for this topic, but I thought I'd
mention it.
PainlessMesh is only for string messages, and is not a true TCP/IP mesh. As
far as I know, there is no true TCP/IP mesh implemented for the ESP.
Some time ago Andreas Reischle (araresearch) tinkered with the nodemcu lua
firmware. He enabled IP_FORWARDING inside lwip as an experiment. This
allows packets to flown through from station to softap interfaces, and vice
versa. I followed his instructions, and connected an ESP to my router, and
my laptop to the ESP. With some minor modifications of my own I was able to
stream netflix to my laptop through the ESP for over half an hour, without
issues.
This makes me believe that it is quite possible to implement a true TCP/IP
mesh for the ESP, maybe with routing enabled in lwip.
I don't know enough about mesh protocols or routing to attack this, but my
hope is that there is someone reading this who does.

https://gitlab.com/BlackEdder/painlessMesh

or mute
the thread

tirithen, 2017-04-24:
Just thinking of the possibilities with solar powered nodes that could create a wide area network that anyone could connect to, would be so cool.

I have a use case (that is probably common) where I would like several nodes with sensors light control and so on in a home to join up in a network, Would be extra nice if the application also is totally distributed so that any node (if one fails/disconnects) could be responsible for ensuring an internet connection to the outside world.

The problem I find is that how would I limit the network so that only some nodes are allowed to join without lots of external parts like screens/keyboards? Otherwise my neighbur would start to control my lamps and so on for as long as anyone would have such a device which would become a super cool network but not desiered in that case.

devyte, 2017-10-03:
Closing as off-topic, see #3655 ,

closed 2017-10-03 by devyte