esp8266/Arduino #2174: Wifi disconnection handler triggers but Wifi does not connect again

djoele, 2016-06-20:
I am using latest Git version.

Lately the usage of Wifi.onEvent() was changed, see WiFi event handling refactoring #2119
The function that I am using to connect Wifi is working at boot time of ESP8266.

Then when I make Wifi disconnect by calling WiFi.disconnect(), I can see that the connectWifi() is triggered. Till so far that is as expected. 

But what I notice is that Wifi is not picking the connection again.

Am I doing something wrong in the connectWifi() function?

I have registered a disconnection handler.
```
void onDisconnected(const WiFiEventStationModeDisconnected& event)
    {
        connectWifi();
    }
```

I have the following connectWifi function.
```
void connectWifi() {
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    ESP.wdtFeed();
  }
  Serial.print(String("[WIFI] IP: "));
  Serial.println(WiFi.localIP());
}
```

djoele, 2016-06-20:
Extra information: I noticed that my ESP8266 lost Wifi connection and did not reconnect. To be able to test this situation I tried with WiFi.disconnect().

Is this indeed a good test to simulate ESP8266 losing Wifi connection?

igrr, 2016-06-21:
As it was before, you should not do any lengthy operations from WiFi event callbacks. Certainly not run a loop waiting for the connection. You can set a flag in disconnect callback and then check the value of flag in your `loop` function.
Another thing to note: by default, ESP will attempt to reconnect to WiFi network whenever it is disconnected. You don't need to handle this manually. A good way to simulate disconnection from an AP would be to reset the AP. ESP will report disconnection, and it will try to reconnect automatically.

djoele, 2016-06-21:
Thank you for the answer.
I will try to make changes the way you suggest and see if the connection pops up again.

A good note, I was not aware of the fact that ESP will automatically reconnect.

Probably the issue I created is going to be resolved because of your suggestion. I will let you knwo what I changed and if it works after that.

djoele, 2016-06-21:
I changed the code as follows:

introduced a variable disconnected which I set to 1 at init.
Then, after Wifi is connected the first time, I set it to 0.

When a disconnect is detected it is set to 1. The reconnect function then triggers.
Inside the main I check for the fact that Wifi has returned.

Now it is working. Tested it both ways:
1. call Wifi.disconnect()
2. Reboot router

```
void reconnectWifi() {
  WiFi.begin(ssid, password);
}
```

```
void onDisconnected(const WiFiEventStationModeDisconnected& event)
    {
        disconnected = 1;
        reconnectWifi();
    }
```

```
void loop() { 
    if (disconnected == 1 && WiFi.status() == WL_CONNECTED){
      Serial.print(String("[WIFI] IP: "));
      Serial.println(WiFi.localIP());
      disconnected = 0;
    }
 ....
```

djoele, 2016-06-21:
For me it's working now. As far as I'm concerned the issue can be closed.

djoele, 2016-06-21:
I also took a look at the other suggestion: without any manual function it should reconnect.

But then I do no see it reconnecting.

djoele, 2016-06-22:
Update: last night I got a real disconnection. But the connection did not recover.
I will do more testing with Serial port connected to PC, so I can really see what is going on.

djoele, 2016-06-22:
I took out the disconnection handler. Also took out the "kill wifi" function I created.

And I tested what happened at a reboot of my router.
As predicted by igrr the connection would be picked up again.

That indeed was the case. So, I guess I just misunderstood some things and it appears to be working like a charm. 

Now I will keep an eye on real world disconnection and see what happens. But I suspect it will pick up Wifi again.

boguslawb, 2016-06-22:
It may be related. I spotted such behavior : if ESP set as AP is powered down and then powered up then client easily connect to it again.No problems.  However if client is powered down during transmission to the server set on ESP  module in AP mode then when powered is returned client seems to connect but got no response from server (timeout waiting for client.print from server) then after a few successful connections (but without data handshaking) it fails even to connect to the server, even with resetting module. The only way to reconnect both modules is to reset AP module then station module. What am I missing ? The IP address is the same , because server on AP module is not turned off it just seems to hang if client disconnects abruptly during communication.

djoele, 2016-06-23:
@igrr: the issue that I found in the beginning is working for me now:
1. I can reconnect Wifi by using a disconnect handler.
2. Wifi reconnects automatically without a disconnect handler

Now I'm waiting on a real situation where Wifi is lost and reconnected again.
I think it is going work work, but I wait to confirm that.

Now boguslawb added an extra observation and possible issue.

Should we split off that observation/issue? Then we close the issue I created after I see it working as expected.

igrr, 2016-06-23:
The issue @boguslawb mentions doesn't look related to this one. I'd venture to say that the behaviour is due to failure to check if client connection is still established. It is likely that the code was copied from one of example sketches and lacks sufficient error checking.
Closing this issue per @djoele comment.

closed 2016-06-23 by igrr

dalbert2, 2018-04-20:
I turned off my wifi access point for several minutes and turned it back on; the first time I did this, the ESP8266 reconnected automatically; the second time it did not and remained in the (6) disconnected status.  Running a wifi scan shows the access point available, but it does not attempt to connect.  Fortunately I have code that will reset automatically if wifi is lost for more than 30 minutes (and then it does connect properly), but this appears to be a genuine problem.