esp8266/Arduino #2176: AP mode module get chock and continuous LmacRxBlk:1

mkeyno, 2016-06-21:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.2

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU rev 1
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

hi @igrr , I made sketch  can sweitch bettwen AP mode and Station mode , but most of time when back to AP mode, module get chock and issue endless LmacRxBlk:1, 
is there any way to break this loop and free the wifi resource for normal operation

following the debugging message and image of operation
`
 Entering the Station Mode
%%scandone
no 2010 found, reconnect after 1s
reconnect
f 0, %%%%%scandone
no 2010 found, reconnect after 1s
reconnect
f -180, %%%%scandone
no 2010 found, reconnect after 1s
reconnect
f r0, %%%%scandone
no 2010 found, reconnect after 1s
%reconnect
f 0, %%%%scandone
no 2010 found, reconnect after 1s
reconnect
f -180, scandone
del if0
usl

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

my sketch automatically  back to AP mode in case not connect to modem successfully

			 Entering the Access Mode
scandone
del if0
usl
mode : softAP(5e:cf:7f:05:fb:a8)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
AP IP address: 192.168.4.1
			HTTP server started
add 1
aid 1
station: b8:76:3f:45:64:43 join, AID = 1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
`

[image]

mkeyno, 2016-06-21:
I might add this also happened when nothings connected to AP module , so it is not related to any cause from browser multi connection trafficking which is mentioned in [this issue](https://github.com/esp8266/Arduino/issues/928)  
when I in Station mode , it is no problem but when choose to switch to AP mode  this issue come out

igrr, 2016-06-23:
Could you please post the sketch code which exhibits this issue? Thanks.

mkeyno, 2016-06-23:
ok , please check the following code, I might add for shorten the code ,some of it replace with function
`

#define DEBUG_ESP_HTTP_SERVER
#define DEBUG_ESP_PORT
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <DNSServer.h>
#include <FS.h>
#include <WebSocketsServer.h>
#include <Hash.h>
#include <EEPROM.h>
#include <ESP8266HTTPClient.h>

WebSocketsServer webSocket = WebSocketsServer(81);
ESP8266WebServer server(80);
const byte DNS_PORT = 53;
DNSServer dnsServer;

void Fail_Safe(void)
{
 Serial.print(F("ENTER SAFE MODE(Restarting.....................)"));
    EEPROM_Write("Ac",0, 1);
 EEPROM_Write("MyIOT",1 ,15);
 EEPROM_Write("KGC",15 ,30);
 EEPROM_Write("12345678", 30,45);
 EEPROM_Write("n",45 ,60);
 EEPROM_Write("12345678",60 ,75);
 EEPROM_Write("0",75 ,76);
 EEPROM_Write("0",76 ,77);
 EEPROM_Write("12",77 ,79);
 EEPROM_Write("13",79 ,81);
 EEPROM_Write("Actuator",83,100);
 EEPROM_Write("ASD",100,103);
 EEPROM_Write("Sensor",103,116);
 EEPROM_Write("SQW",116,119);
 EEPROM_Write("20160616",119,127);
 EEPROM_Write("77666",127,133);
 EEPROM_Write("S",133,134);
 EEPROM_Write("2700@4500@6300@7560@8640",134,160);
 EEPROM_Write("20@40@6@7@80",160,177);
EEPROM_Write("0",177,178);
EEPROM_Write("0",178,179);
EEPROM_Write("0",179,180);
EEPROM_Write("0",180,181);
EEPROM_Write("0",181,182);
 EEPROM_Write("2",182,183);
 EEPROM_Write("3",183,184);

 for (byte i = 0; i < 200; i++)
                  {
                  byte c=EEPROM.read(i);
                   Serial.print(i); Serial.print(" = ");Serial.println((char)c);
                  }
}
 
void setup() 
{
   
  Serial.begin(115200);

  delay(500);
  
  Serial.setDebugOutput(true);
 
pinMode(0, INPUT_PULLUP);
delay(500);
 SPIFFS.begin();
  EEPROM.begin(256); 
program_start=millis();
while((millis()-program_start )<2000 )
{

  if(!digitalRead(0)){Fail_Safe(); break;}
}
  pinMode(Sensor_pin, INPUT_PULLUP);
 pinMode(Actuater_pin, OUTPUT); 
  digitalWrite(Actuater_pin,LOW);

 
  DIR(); 
  restor_config();
  // if in access point mode  or number of fail  try to connect to access point  greater than 4 
 if(MyMode[0]=='A' || reset_number>4 || SSID.length()<3 )
                            {
                               Serial.println(F("\t\t\t Entering the Access Mode"));
                               WiFi.mode(WIFI_AP);
                               WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
                               String temp_name="Smart_Node_" +String(ESP.getChipId(), HEX);
                              
                               if(AP_SSID.length()<3) {WiFi.softAP(temp_name.c_str(),"12345678"); AP_SSID=temp_name; }
                               else                   WiFi.softAP(AP_SSID.c_str(),AP_SSIDPASS.c_str());                                
                               delay(500); // Without delay I've seen the IP address blank
                               EEPROM_Write(String(0),76,77);   
                               MyIP=WiFi.softAPIP().toString();
                               Serial.print("AP IP address: ");Serial.println(MyIP.c_str());
                               dnsServer.setErrorReplyCode(DNSReplyCode::ServerFailure);
                               dnsServer.start(DNS_PORT, "*", apIP);  
                            }
else if(reset_number<5)
              {
                Serial.println(F("\t\t\t Entering the Station Mode"));
                WiFi.mode(WIFI_STA);
                if(STATIC)   WiFi.config(str2ip(Static_IP),str2ip(Server_IP), IPAddress(255,255,255,0));   
                WiFi.begin(SSID.c_str(), SSIDPASS.c_str());             
                unsigned long startTime = millis();
                while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000)  {Serial.write('%');delay(500); } 
                if(WiFi.status() != WL_CONNECTED) {
                                                      reset_number++;
                                                      if(reset_number>4) EEPROM_Write("S",133,134);
     EEPROM_Write(String(reset_number),76,77); // store the number of reset on fail to connect                                     
                                                  ESP.restart();
                                                  }            
                 MyIP=WiFi.localIP().toString();
                 Serial.print("AP IP address: ");Serial.println(MyIP.c_str());
                if (!MDNS.begin(myHostname.c_str())) {
                                                    Serial.println(F("Error setting up MDNS responder!"));
                                                    while(1) {  delay(1000); }
                                                  }
                Serial.println(F("mDNS responder started"));
              }
 
  
 
  server.on("/process", process);  
  server.on("/data", data);   
  server.on("/update",      HTTP_POST, [](){ Serial.println("Upload post");    server.send(200, "text/plain", ""); /* Redirect("/setting.html");*/  }, handleFileUpload);
  server.on("/", handleRoot);
  server.on("/setup.html", handleSetup);
  server.on("/config.html", handleConfig);
  server.onNotFound(handleOther);
  
  server.begin();  Serial.println(F("\t\t\tHTTP server started"));  
  
  const char * headerkeys[] = {"User-Agent","Cookie"} ; //here the list of headers to be recorded
  size_t hz = sizeof(headerkeys)/sizeof(char*); 
  server.collectHeaders(headerkeys, hz ); //ask server to track these headers
  
  webSocket.begin();
  webSocket.onEvent(webSocketEvent);
 delay(500);
 

  
 // try to send registration info to server connected
  if(MyMode[0]!='A' && STANDALONE) IsRegistered=send_device_info_to_server();
  
}

loop
{
if(MyMode[0]=='A') dnsServer.processNextRequest();
  server.handleClient();  
  webSocket.loop();

in_new_sensor_data_send_it_to_server();
 yield(); 
}
`

mkeyno, 2016-06-24:
dear @igrr  do you have any suggestion? is it possible in case of chocking, internal routine or watchdog timer close all connection and flush all stack ? something like reset TCP routing

boguslawb, 2016-06-24:
I've got the same effect when client is abruptly disconnected (for example when power failure occurs). When power is restored connection to AP seems established but TCP server goto havoc like this :
err already associed!
station: 18:fe:34:e0:36:d2 leave, AID = 1
rm 1
add 1
aid 1
station: 18:fe:34:e0:36:d2 join, AID = 1
LmacRxBlk:1

last line repeating over and over, WifiServer is not working and need restart.

My question is how to detect such problem in server code and close client releasing  TCP stack ? At least wdt reset would be nice

mkeyno, 2016-06-24:
that would be good solution if @igrr  have any offer for that , however I use the EEPROM for store data and restore it in startup I think this issue related to memory  disorder , following is the decode exception 
`
ctx: cont 
sp: 3fff2920 end: 3fff2b70 offset: 01b0
ecoding 4 results
0x40212915: EEPROMClass::begin(unsigned int) at ..\2.2.0\libraries\EEPROM/EEPROM.cpp line 122
0x40209572: setup at ...PIR_1.ino.ino line 1203 (discriminator 1)
0x40214ff4: loop_wrapper at ...\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at ...\2.2.0\cores\esp8266/cont.S line 109

` 
dear @me-no-dev  would you please check this issue and let us know your opinion

mkeyno, 2016-06-25:
I disable EEPROM and use the simpler code but still LmacRxBlk:1 gonna kill me
`wifi evt: 7
add 1
aid 1
station: b8:76:3f:45:64:43 join, AID = 1
wifi evt: 5
:urn 43
:urch 43, 43
:urch 86, 30
:urch 116, 30
:urch 146, 43
LmacRxBlk:1
LmacRxBlk:1`

igrr, 2016-06-25:
There is some UDP listener in your code, and you aren't processing incoming
packets. Are you using any UDP stuff?

igrr, 2016-06-25:
Probably you have initialized DNSServer library but aren't calling
processNextRequest.

mkeyno, 2016-06-25:
I did Ivan, 
in loop `if(MyMode[0]=='A') dnsServer.processNextRequest();`

mkeyno, 2016-06-25:
oh boy you were right , somehow in my code MyMode[0] not be AP mode , thanks , very thank you dear Ivan, 
now its work
station: b8:76:3f:45:64:43 join, AID = 1
wifi evt: 5
:urn 43
:urd 43, 43, 0
:urn 33
:urd 33, 33, 0
:urn 33
:urd 33, 33, 0
:urch 33, 33
:urd 33, 33, 0
:urch 33, 33
:urd 33, 33, 0
:urn 30
:urd 30, 30, 0
:urn 30
:urd 30, 30, 0
:urch 30, 30
:urd 30, 30, 0
:urn 30
:urd 30, 30, 0
wifi evt: 7

mkeyno, 2016-06-25:
following is new debug , but still have terrified error
what would be optimize solution to let the TCP  not chocked 
Request page : /Sindex.html:wr
:sent 141
:ww
:wr
WS:ac
:rn 320
WS:ac
:rn 309
WS:ac
:rn 303
:sent 1460
:ww
:wr
:sent 1460
:ww
:wr
:sent 1460
:ww
:wr
:urch 27, 27
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
:ww
SPIFFS_close: fd=1
SPIFFS_close: fd=1
	[SENT]
:urd 27, 27, 0
:urd 27, 27, 0
:urn 27
:urd 27, 27, 0
:urn 27
:urd 27, 27, 0
:urn 33
:urd 33, 33, 0
:urn 33
:urd 33, 33, 0
:urn 33
:urd 33, 33, 0
:sent 1460
:ur 1
:close
WS:dis
:del
:ref 1
WS:av
:ref 2
:ur 2
New client
:ref 2
:ur 2

igrr, 2016-06-25:
If you need to serve lots of stuff at the same time, you probably need to use ESPAsyncWebServer library instead of ESP8266WebServer.

mkeyno, 2016-06-25:
thanks @igrr Ivan , you are right and I'm starting to go with this magnificent me-now-dev repo,  unfortunately it not has much example as your repo , however sometimes my module network icon (in windows)get busy whereas debug shows nothings  and debug level set to all

closed 2016-06-28 by mkeyno