esp8266/Arduino #2178: ESP-01 problem while powering

nitheeshmohan, 2016-06-22:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

hai.. i have been working with esp-01 for a while now.. the only prob im facing is if the gpio is connected to led while i power my esp it enters into some mode with onboard LED and gpio in ON state..and also creates some sound... -t

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

igrr, 2016-06-23:
Which pin do you connect the LED to? How exactly do you connect it? Please post the circuit diagram.

markusschweitzer, 2016-06-23:
You can get the correct wiring for serial communication and bootloading from: [ESP-Modules](https://github.com/esp8266/Arduino/blob/master/doc/boards.md#generic-esp8266-modules)

devyte, 2017-10-16:
Closing per #3655 .

closed 2017-10-16 by devyte