esp8266/Arduino #2179: client.print no send  when this statement is inside a conditional "if () {

Ciriaco08, 2016-06-22:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	2.1.0

### Description

Problem description
client.print no send to client when this statement is inside a conditional "if () {

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz?
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch
#include <ESP8266WiFi.h>

const char* ssid = "Neptuno marciana";
const char* password = "205LKJJJLKMJ95133981";

// Definimos una IP fija
IPAddress ip(192,168,1,44);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);

void setup() {
  Serial.begin(115200);
  delay(1500);

  // Preparamos los GPIO
  pinMode(2, OUTPUT);
  digitalWrite(2, LOW);
  
  // Conectando con la wifi
  Serial.println();
  Serial.println();
  Serial.print("Conectando a ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  // Escribimos la IP fija
  WiFi.config(ip, gateway, subnet);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  // Start the server
  server.begin();
  Serial.println("Server started");

  // Print the IP address
  Serial.println(WiFi.localIP());
}

void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }

  // Read the first line of the request
  String req = client.readStringUntil('\r');
  client.flush();

    // Respuesta del módulo

 if (req  != "GET / HTTP/1.1")
    {
     // Sacando datos
      String dato_1 = req.substring(10,11);
      String dato_2 = req.substring(16,17);
  
      // Los convertimos en números para poder ponerlos en los pines
      int dato_1_N = dato_1.toInt();
      int dato_2_N = dato_2.toInt();
      Serial.println("PIN en string: "+ dato_1);
      Serial.println("PIN estado string: "+ dato_2 );

      // Pone el GPIO al valor pedido
       digitalWrite(dato_1_N, dato_2_N);

   
       client.flush();

       String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO esta ahora ";
       s += (dato_2_N)?"apagado":"encencido";
       s += "</html>\n";

       // Enviando respuesta a la página
      client.print(s);
      delay(1);
   
    }

}

                                                      **HTML CLIENT CODE**

<html>
<head>
<meta charset="utf-8" />
<title>Conexión ESP286</title>

<script src="http://code.jquery.com/jquery-1.12.0.min.js"></script>

<script type="text/javascript">

	     var gpio_on = 1; 
		  var gpio_of = 0;

$(document).ready(function()
{
        $("input[name=gpio]").click(function () {    
	     var gpio_id = $(this).attr('id'); // toma el valor del atributo del id
		  var url="http://192.168.1.43:80/"

    		if( $(this).is(':checked') ){
    		
   // Hacer algo si el checkbox ha sido seleccionado
        	
         // enviando HTTP GET a la dirección url del módulo con el paranetro "id" y valor
         
       	$.get(url, {pin:gpio_id,val:gpio_on});// ejecuta el get
       	
       	// refresca el iframe y carga la respuesta del módulo
         $('#iframe').attr('src', url);
         $('#iframe').reload();
       	
    		} else {
    		
   // Hacer algo si el checkbox ha sido deseleccionado
   
			// enviando HTTP GET a la dirección url con el paranetro "id" y valor
			
         $.get(url, {pin:gpio_id,val:gpio_of});// ejecuta el get
         
         // refresca el iframe y carga la respuesta del módulo
         $('#iframe').attr('src', url);
         $('#iframe').reload();
         
    		}
         });
         
});
         

$(document).ready(function()
{
        $("input[name=gpio_2]").click(function () {    
	     var gpio_id = $(this).attr('id'); // toma el valor del atributo del id
		  var url="http://192.168.1.44:80/"
	  
    		if( $(this).is(':checked') ){
    		
   // Hacer algo si el checkbox ha sido seleccionado
   
         // enviando HTTP GET a la dirección url del módulo con el paranetro "id" y valor
         
       	$.get(url, {pin:gpio_id,val:gpio_on});// ejecuta el get enviando datos al módulo
       	
       	// refresca el iframe y carga la respuesta del módulo
         $('#iframe').attr('src', url);
         $('#iframe').reload();
         
    		} else {
    		
   // Hacer algo si el checkbox ha sido deseleccionado
   
			// enviando HTTP GET a la dirección url del módulo con el paranetro "id" y valor
			
         $.get(url, {pin:gpio_id,val:gpio_of});// ejecuta el get enviando datos al módulo
         
          // refresca el iframe y carga la respuesta del módulo
          $('#iframe').attr('src', url);
          $('#iframe').reload();
         
    		}
         });
         
});

</script>

</head>

<body>

<h2>Control del ESP2866 con Jquery</h2>
</br>
</br>

<h3>A-ESP2866 - http://192.168.1.43:80/</h3>
<form>
	<input type="checkbox" name="gpio" id="2" value="2"> GPÌO 2 </br>
</form>
</br>
</br>

<h3>B-ESP2866 - http://192.168.1.44:80/</h3>
<form>
	<input type="checkbox" name="gpio_2" id="2" value="2"> GPÌO 2 </br>
</form>
</br>
</br>
<h3>Respuesta del Módulo B-ESP2866 </h3>
<iframe src="http://192.168.1.43:80/" width="50%" height="50%" frameborder="0" scrolling="yes" id="iframe"></iframe>
</body>
</html>

### Debug Messages

```
messages here
```

igrr, 2016-06-23:
I checked your code and I'm confused. Do you get serial output from the following statement?
```c++
Serial.println("PIN en string: "+ dato_1);
```
If you see this printed, then `client.print` is also executed.

I would also suggest to use ESP8266WebServer library instead of writing your own HTTP parsing code. Library also parses request arguments for you.

Ciriaco08, 2016-06-23:
Serial.println("PIN en string: "+ dato_1);

This line is just a check, all Serial.println instruccions  are for check 

This is the send:

 String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO esta ahora ";
       s += (dato_2_N)?"apagado":"encencido";
       s += "</html>\n";

       // Enviando respuesta a la página
       client.print(s);

igrr, 2016-06-23:
This line looks okay as well. Just make sure you set Content-Type to `text/plain` if you aren't sending HTML, like in this case.

Ciriaco08, 2016-06-23:
The problem is the "if" statement. 
If we remove that instruction  the code run normally and  client.print(s), send "s" variable

igrr, 2016-06-23:
Could you please add `Serial.println(s);` line after `client.print(s);` line and post serial output please?

Ciriaco08, 2016-06-23:
Serial output:

..
WiFi connected
Server started
192.168.1.44
new client
new client
new client
new client
new client
new client
new client

HTTP/1.1 200 OK
Content-Type: text/html

<!DOCTYPE HTML>
<html>
GPIO esta ahora apagado</html>

new client

Ciriaco08, 2016-06-23:
with debug on:

Conectando a Neptuno marciana
.wifi evt: 0
wifi evt: 3
.
WiFi connected
Server started
192.168.1.44
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
WS:ac
:rn 298
:ref 1
WS:av
:ref 2
:ur 2
new client
:ur 1
:close
WS:dis
:del
WS:ac
:rn 298
:ref 1
WS:av
:ref 2
:ur 2
new client
:ur 1
:close
WS:dis
:del
WS:ac
:rn 298
:ref 1
WS:av
:ref 2
:ur 2
new client
:ur 1
:close
WS:dis
:del
WS:ac
:rn 298
:ref 1
WS:av
:ref 2
:ur 2
new client
:ur 1
:close
WS:dis
:del
WS:ac
:ref 1
WS:av
:ref 2
:ur 2
new client
:rn 298
:ur 1
:close
WS:dis
:del
WS:ac
:ref 1
WS:av
:ref 2
:ur 2
new client
:rn 298
:ur 1
:close
WS:dis
:del
WS:ac
:rn 264
:ref 1
WS:av
:ref 2
:ur 2
new client
:wr
:sent 100
:ww
HTTP/1.1 200 OK
Content-Type: text/html

<!DOCTYPE HTML>
<html>
GPIO esta ahora apagado</html>

:rcl
:abort
:ur 1
WS:dis
:del
WS:ac
:ref 1
WS:av
:ref 2
:ur 2
new client
wifi evt: 7
:rn 264
:wr
WS:ac
:rn 298
:sent 102
:ww
HTTP/1.1 200 OK
Content-Type: text/html

<!DOCTYPE HTML>
<html>
GPIO esta ahora encencido</html>

:ur 1
:close
WS:dis
:del
:ref 1
WS:av
:ref 2
:ur 2
new client

Ciriaco08, 2016-06-23:
Hi , Is there any documentation for use ESP8266WebServer library?
regards

devyte, 2017-10-16:
@Ciriaco08 no docs at this time, but there are many examples, and the object interface is pretty clear.
This looks like user error to me. Closing per #3655 .

closed 2017-10-16 by devyte