esp8266/Arduino #2180: Async connect for WiFiClient ?

pgScorpio, 2016-06-22:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	2.2.0

### Description

Problem description

At this moment WiFiClient::connect() is blocking until connected or error.

Especially when the server you are connecting to is offline this causes your sketch to freeze for a few seconds...... In a lot of cases this is not acceptable.

I think it should be possible to start a connect without blocking and check from loop if the connect succeeded or not.

chaeplin, 2016-06-23:
this ? https://github.com/me-no-dev/ESPAsyncTCP

igrr, 2016-06-23:
Also, closing as duplicate of #922.

closed 2016-06-23 by igrr

miguelpom, 2016-12-22:
Hi, guys.
@igrr: this issue is kind of related to the one posted in #922. However, the actual problem with the blocking timeout of the function WifiClient.connect() isn't really addressed or solved. Or so it seems to me.
Can we reanimate this thread, please?