esp8266/Arduino #2182: Client.connect fails with static IP Address

stavBodik, 2016-06-23:
#### Hardware
Hardware:			?ESP-12E?
Core Version:      	       ? How do i find it (^^) ? ?

### Description
Connecting with WiFiClient.connect fails when connected to router with static ip address,
When connected trough DHCP works fine .
Port 80 forward configured . 

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?512K (64K SPIFFS)?
CPU Frequency:	?80Mhz?
Flash Mode:		?dio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?nodemcu?

### Sketch

```
#include <ESP8266WiFi.h>
#include <SPI.h>

const char* ssid     = "********";
const char* password = "*******";
const char* host = "*****";

IPAddress ip(192, 168, 0, 20);    
IPAddress gateway(192,168,1,1); 
IPAddress subnet(255,255,255,0); 

void setup() {

  String Statuses[] =  { "WL_IDLE_STATUS=0", "WL_NO_SSID_AVAIL=1", "WL_SCAN_COMPLETED=2", "WL_CONNECTED=3", "WL_CONNECT_FAILED=4", "WL_CONNECTION_LOST=5", "WL_DISCONNECTED=6"};

  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.println(Statuses[WiFi.status()]);
  }

  Serial.println("WiFi connected");  
 
}

void loop() {
  delay(5000);

  Serial.print("connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed error");
    Serial.println(client.connect(host, httpPort));
    return;
  }

.....
} 
```

### Debug Messages

```
[hostByName] request IP for: jumphigher.esy.es
wifi evt: 7
[hostByName] Host: jumphigher.esy.es lookup error: -5!
[hostByName] request IP for: jumphigher.esy.es
[hostByName] Host: jumphigher.esy.es lookup error: -5!
[hostByName] request IP for: jumphigher.esy.es
[hostByName] Host: jumphigher.esy.es lookup error: -5!
connecting to jumphigher.esy.es
[hostByName] request IP for: jumphigher.esy.es
[hostByName] Host: jumphigher.esy.es lookup error: -5!
```

chaeplin, 2016-06-23:
check https://github.com/esp8266/Arduino/issues/128

```
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  WiFi.config(ip, gateway, subnet);

  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.println(Statuses[WiFi.status()]);
  }

  Serial.println("WiFi connected"); 
  Serial.println(WiFi.localIP());

```

stavBodik, 2016-06-23:
I saw this post but i need the config to work before begin all the idia is to tell the WiFi class not to wait for ip address from the router . 

Calling WiFi.config() before WiFi.begin() forces begin() to configure the WiFi shield with the network addresses specified in config().

But i will try for knowing if its works .

chaeplin, 2016-06-23:
@stavBodik check this : https://github.com/esp8266/Arduino/issues/1138
```
Links2004 commented on Dec 30, 2015
shut be fixed with #1323, there where a problem with the modes,
if WiFi STA mode where not enabled before the IP config it ignore the IP configuration.
linked to: #1316
```

chaeplin, 2016-06-23:
@stavBodik 
I have tested some config. This works to me. Looks like location of "WiFiClient client;" is problem.
```
IPAddress ip_static(192, 168, 10, 73);    
IPAddress ip_gateway(192,168,10,1); 
IPAddress ip_subnet(255,255,255,0); 
IPAddress ip_dns(192,168,10,1);

WiFiClient client;

void setup() {

  Serial.begin(115200);
  Serial.println();
  Serial.flush();
  Serial.println("Starting.....");

  //WiFi.mode(WIFI_STA);
  WiFi.config(ip_static, ip_gateway, ip_subnet);
  //WiFi.config(ip_static, ip_gateway, ip_subnet, ip_dns);
  WiFi.begin(ssid, password);

  int Attempt = 0;
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(100);
    Attempt++;
    Serial.print(" ");
    Serial.print(Attempt);
  }

  Serial.println("");
  Serial.print("[WIFI] connected : ");
  Serial.println(WiFi.localIP());
}
```

```
Starting.....
 1 2
[WIFI] connected : 192.168.10.73
```

chaeplin, 2016-06-23:
@stavBodik Is host is url ?

I have tested static ip config. Config before begin works.
To use dns, config should be like this.
```
WiFi.config(ip_static, ip_gateway, ip_subnet, ip_dns); // WiFi.config(ip_static, ip_gateway, ip_subnet, ip_dns1, ip_dns2); 
WiFi.begin(ssid, password);
```

stavBodik, 2016-06-23:
So i have tryd with the dns thing , not working same -5 look up error .
Maybe it has something to do with my router ? please take look at the next screen shoot :

[image]

In green is the input to .config

in red , this is wan right ? we are not connecting from output global internet its ok that its DHCP there ?

chaeplin, 2016-06-23:
@stavBodik 
ip and gateway is not on a same subnet.
IPAddress ip(192, 168, 0, 20); and IPAddress gateway(192,168,1,1); 
```
IPAddress ip(192, 168, 0, 20);    
IPAddress gateway(192,168,1,1); 
IPAddress subnet(255,255,255,0); 
``

If the router runs local dos cache or forwarder,
```
IPAddress ip_static(192, 168, 1, 20);    
IPAddress ip_gateway(192,168,1,1); 
IPAddress ip_subnet(255,255,255,0); 
IPAddress ip_dns(192,168,1,1);
WiFi.config(ip_static, ip_gateway, ip_subnet, ip_dns); 
WiFi.begin(ssid, password);
```

else 
```
IPAddress ip_static(192, 168, 1, 20);    
IPAddress ip_gateway(192,168,1,1); 
IPAddress ip_subnet(255,255,255,0); 
IPAddress ip_dns1(213,57,2,5);
IPAddress ip_dns1(213,57,22,5);
WiFi.config(ip_static, ip_gateway, ip_subnet, ip_dns1, ip_dns2); 
WiFi.begin(ssid, password);
```

WAN is connected to ISP using DHCP(ip : 5.29.156.184), it's not related to local LAN.
Check router config whether router only permits connection from DHCP client.

Ciriaco08, 2016-06-23:
Change the order of this:

  WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, password);

for this:
  WiFi.begin(ssid, password);
  WiFi.config(ip, gateway, subnet);

This is solved here:  https://github.com/esp8266/Arduino/issues/128

stavBodik, 2016-06-23:
Ciriaco08 please read my comment before . . .

chaeplin,With option 1 idk why the esp got very hot and the current consuption draw to very very low not working .

option 2 i already tryd ): i did not found any config(on the ui panel of the router) about allowing only DHCP there is any config file inside the router where i can check it ?

stavBodik, 2016-06-23:
Looks like my shield got crazy now wont connect on normal way only begin 

`connecting to jumphigher.esy.es
[hostByName] request IP for: jumphigher.esy.es
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4021887f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40104e67, depc=0x00000000

Exception (9):
epc1=0x4021887f epc2=0x00000000 epc3=0x00000000 excvaddr=0x40104e67 depc=0x00000000

ctx: sys 
sp: 3ffff890 end: 3fffffb0 offset: 01a0

.......

Fatal exception 28(LoadProhibitedCause):
epc1=0x402187ff, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00008000, depc=0x00000000

Exception (28):
epc1=0x402187ff epc2=0x00000000 epc3=0x00000000 excvaddr=0x00008000 depc=0x00000000

ctx: sys 
sp: 3ffff690 end: 3fffffb0 offset: 01a0
`

drmpf, 2016-09-22:
WiFi.status() != WL_CONNECTED fails for static IP in V2.3.0 but works in V2.2.0

test code

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = ".....";
const char* password = "......";

ESP8266WebServer server(80);

void setup(void) {
  Serial.begin(115200);
  for (int i = 10; i > 0; i--) { // wait 10sec with fan on at start up to indicate sucessful reset on powerup
    Serial.print(i);
    Serial.print(' ');
    delay(1000);
  }
  WiFi.begin(ssid, password);
  Serial.println("WiFi.begin called waiting for connected");

  IPAddress ip(10, 1, 1, 222);
  IPAddress gateway(ip[0], ip[1], ip[2], 1); // set gatway to ... 1
  IPAddress subnet(255, 255, 255, 0);
  Serial.print(F("Setting ip to: "));
  Serial.println(ip);
  Serial.print(F("Setting gateway to: "));
  Serial.println(gateway);
  Serial.print(F("Setting subnet to: "));
  Serial.println(subnet);
  if (!WiFi.config(ip, gateway, subnet, gateway)) {
    Serial.println("WiFi.config() failed");
  }
  Serial.println("WiFi.config() succeeded");

  // Wait for connection for 15 sec and then just continue
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

}

void loop(void) {
}
```

Results V2.3.0
5 4 3 2 1 WiFi.begin called waiting for connected
Setting ip to: 10.1.1.222
Setting gateway to: 10.1.1.1
Setting subnet to: 255.255.255.0
WiFi.config() succeeded
............................................................................................

BUT fing, shows 10.1.1.222 is on the network

Results after rolling back to V2.2.0
7 6 5 4 3 2 1 WiFi.begin called waiting for connected
Setting ip to: 10.1.1.222
Setting gateway to: 10.1.1.1
Setting subnet to: 255.255.255.0
WiFi.config() succeeded

Connected to FCC-Wfi
IP address: 10.1.1.222

Also on V2.2.0 also works calling config() BEFORE calling begin (as it should be)

devyte, 2017-10-17:
Closing as resolved.

closed 2017-10-17 by devyte

justin8, 2018-03-24:
For anyone having this issue who ends up here in the future, as ridiculous as it sounds, slowing your upload speed to 57600 baud fixed this issue for me completely. I couldn't get it working on 2.3.0 or 2.2.0 at all beforehand, now both work flawlessly.