esp8266/Arduino #2183: Recent update (2.3.0) gives some warnings

dandandandandandandan, 2016-06-23:
### Basic Infos

#### Hardware
Hardware:			NodeMCU 0.9
Core Version:      	2.3.0

### Description

During verbose compilation in the Arduino IDE the new 2.3.0 version gives me some warnings
 ### Settings in IDE

Module:  			NodeMCU 0.9
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload Using:	SERIAL
Reset Method:		nodemcu

### Debug Messages

esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp: In function 'int ax_port_read(int, uint8_t*, size_t)':

hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp:539:53: warning: suggest parentheses around '&&' within '||' [-Wparentheses]

     if (!_client || _client->state() != ESTABLISHED && !_client->getSize()) {

                                                     ^

\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp: In member function 'bool HTTPClient::beginInternal(String, const char*)':

\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp:135:10: warning: unused variable 'hasPort' [-Wunused-variable]

     bool hasPort = false;

          ^

\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\Schedule.cpp:17:13: warning: 'void init_lists()' defined but not used [-Wunused-function]

 static void init_lists()

             ^

juansta, 2016-08-26:
In addition to the above I am also seeing additional warnings on compilation

```
.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp: In member function 'void MDNSResponder::_parsePacket()':
.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:590:18: warning: unused variable 'answerPrio' [-Wunused-variable]
uint16_t answerPrio = _conn_read16(); // Read priority
^
.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:591:18: warning: unused variable 'answerWeight' [-Wunused-variable]
uint16_t answerWeight = _conn_read16(); // Read weight
^
.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:558:16: warning: unused variable 'answerClass' [-Wunused-variable]
uint16_t answerClass = _conn_read16(); // Read class
^
.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:559:16: warning: unused variable 'answerTtl' [-Wunused-variable]
uint32_t answerTtl = _conn_read32(); // Read ttl
^
```

arantius, 2017-02-26:
I also keep verbose warnings on when developing, and I want zero of them.  I get: https://gist.github.com/arantius/e786224529a17782d23fa03df250b925

Here's my attempt to address some of them: https://github.com/esp8266/Arduino/compare/master...arantius:build-warnings

devyte, 2017-10-16:
@igrr @juansta what is the status of this? I see the referenced PR is not merged, and I still see the referenced code with unused vars. OTOH, I haven't seen build warning, although I haven't really been paying attention to core messages, either.

kubark42, 2017-11-03:
This would be nice to clean up, it prevents the use of a project-wide `-Werror -Wall`. I agree with @arantius, it's very important to have all compiler warnings eliminated when deploying.

closed 2018-01-02 by igrr