esp8266/Arduino #2184: WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override.

kp372, 2016-06-23:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

/*******************************************************************
 *  this is a basic example how to program a Telegram Bot          *
 *  using TelegramBOT library on ESP8266                           *
 *                                                                 *
 *  Open a conversation with the bot, you can command via Telegram *
 *  a led from ESP8266 GPIO                                        *
 *  https://web.telegram.org/#/im?p=@FlashledBot_bot               *
 *                                                                 *
 *  written by Giancarlo Bacchio                                   *
 *******************************************************************/

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <ESP8266TelegramBOT.h>

// Initialize Wifi connection to the router
char ssid[] = "xxxxxxxxxxxxxxxxxxxxxx";              // your network SSID (name)
char pass[] = "yyyyyyyy";                              // your network key

// Initialize Telegram BOT

#define BOTtoken "134745667:AAETzUWRQdb9xbMX_s-q_50U6ffgXcW3ldg"  //token of FlashledBOT
#define BOTname "FlashledBot"
#define BOTusername "FlashledBot_bot"

TelegramBOT bot(BOTtoken, BOTname, BOTusername);

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done
bool Start = false;

/********************************************
 * EchoMessages - function to Echo messages *
 ********************************************/
void Bot_ExecMessages() {
  for (int i = 1; i < bot.message[0][0].toInt() + 1; i++)      {
    bot.message[i][5]=bot.message[i][5].substring(1,bot.message[i][5].length());
    if (bot.message[i][5] == "\/ledon") {
      digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
      bot.sendMessage(bot.message[i][4], "Led is ON", "");
    }
    if (bot.message[i][5] == "\/ledoff") {
      digitalWrite(13, LOW);    // turn the LED off (LOW is the voltage level)
      bot.sendMessage(bot.message[i][4], "Led is OFF", "");
    }
    if (bot.message[i][5] == "\/start") {
      String wellcome = "Wellcome from FlashLedBot, your personal Bot on ESP8266 board";
      String wellcome1 = "/ledon : to switch the Led ON";
      String wellcome2 = "/ledoff : to switch the Led OFF";
      bot.sendMessage(bot.message[i][4], wellcome, "");
      bot.sendMessage(bot.message[i][4], wellcome1, "");
      bot.sendMessage(bot.message[i][4], wellcome2, "");
      Start = true;
    }
  }
  bot.message[0][0] = "";   // All messages have been replied - reset new messages
}

void setup() {
  Serial.begin(115200);
  delay(3000);

  // attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  while (WiFi.begin(ssid, pass) != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);
  bot.begin();      // launch Bot functionalities
  pinMode(2, OUTPUT); // initialize digital pin 2 as an output.
}

void loop() {
  if (millis() > Bot_lasttime + Bot_mtbs)  {
    bot.getUpdates(bot.message[0][1]);   // launch API GetUpdates up to xxx message
    Bot_ExecMessages();   // reply to message with Echo
    Bot_lasttime = millis();
  }
}

```

### Debug Messages

```
/WiFiClientSecure.h:49:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override.
   size_t peekBytes(uint8_t *buffer, size_t length) override;
```

asetyde, 2016-06-23:
Giancarlo bello ! 
I not see which problem do you have ?

kp372, 2016-06-23:
THank You for quick comment !
I am getting following error while compiling in Arduino IDE
```
/WiFiClientSecure.h:49:10: error: 'size_t WiFiClientSecure::peekBytes(uint8_t*, size_t)' marked override, but does not override.
   size_t peekBytes(uint8_t *buffer, size_t length) override;
```

PS: I have already set board manager URL to this:
`http://arduino.esp8266.com/stable/package_esp8266com_index.json`

asetyde, 2016-06-23:
Why do you use a 2.1.0 rc2 ?  i can not try today your code but you are check if your issue come from bug ? Have you tried with  new 2.3.0 ?

kp372, 2016-06-23:
Thank you very much. I haven't updated the library. 2.3.0 is working like charm.
no more errors.

closed 2016-06-23 by kp372

joeleonjr, 2023-08-30:
Hi! I'm a security researcher @TruffleSecurity. I found a valid API key leaked in a comment in this Issue. I highly recommend rotating the API key immediately (not just deleting the comment). Here's a blog post I wrote on remediating these types of vulnerabilities: https://trufflesecurity.com/blog/remediate-leaked-api-keys-with-key-rotation.<br><br>If you can't find the leaked API key, you can use the following [TruffleHog](https://github.com/trufflesecurity/trufflehog) command to automatically scan for it:<br>`trufflehog github --repo https://github.com/<your_username>/<your_reponame> --issue-comments --only-verified`<br><br>If it's not clear where/what the security issue is, please let me know. I'd be happy to help. Please note that the API key might be in the past edit of a comment.<br><br><br><br>(cc @kp372)

earlephilhower, 2023-08-30:
Thanks for the notice @joeleonjr .  I've deleted revs w/the guy's key.

But, if the key was valid from 2016 when this opened until 2023 then they have bigger issues, though, I'd say. 😆