esp8266/Arduino #2186: ESP8266WiFi doesn't work with new 2.3.0

RogerSchaefer, 2016-06-24:
Installed the new 2.3.0 today and found that a sketch I have used for a long time will no longer cause the ESP8266 to connect with my WiFi

I reverted to version 2.2.0 and the sketch works fine.

ESP8266WiFi was the only ESP8266 include in the sketch

Roger

igrr, 2016-06-24:
Will you post your sketch? I can try reproducing that issue.

hoshinota, 2016-06-24:
"Millis ()" is strange?
Value of ADC was also been 20 - 30 points lower.

RogerSchaefer, 2016-06-24:
Hi igrr,
The code is below; hope it helps.
Roger
///////////////////////////////////////////
//  my dashboard
//  https://io.adafruit.com/Oldmicroguy/
//
/***************************************************
  Adafruit MQTT Library ESP8266 Example

  Must use ESP8266 Arduino from:
    https://github.com/esp8266/Arduino

  Works great with Adafruit's Huzzah ESP board & Feather
  ----> https://www.adafruit.com/product/2471
  ----> https://www.adafruit.com/products/2821

  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Tony DiCola for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
////////////////////////////
//  pick one test device  //
//                        //
#define ECOSWITCH 0       //
#define SWEETPEA 0        //
#define NODEMCU 0         //
#define ESP01 1           //
#define THING 0           // Sparkfun Thing
#define SONOFF 0          //
////////////////////////////

#if ESP01                     // testing with ESP-01
#define BUTTON_PIN 0
#define LED_PIN 2
#define RELAY 2               // only two outputs on ESP-01
#define INVERT false
#endif

#if THING                    // testing with Sparkfun THING
#define BUTTON_PIN 0
#define LED_PIN 5
#define RELAY 12
#define INVERT false
#endif

#if SWEETPEA                // testing with SweetPea
#define BUTTON_PIN 0
#define LED_PIN BUILTIN_LED
#define RELAY BUILTIN_LED
#define INVERT false
#endif

#if SONOFF                   // WiFi controlled power relay
#define BUTTON_PIN 0
#define LED_PIN 13
#define RELAY 12
#endif

//#define PULLUP true
//#define INVERT true
//#define DEBOUNCE_MS 100
//#define SINK false

[... 153 lines cut ...]
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
         while (1);
       }
  }
  Serial.println("MQTT Connected!");
}
///////////////////////////////
void blink (int i) {
  while (i) {
    digitalWrite(LED_PIN, LOW);       // Turn the LED on (Note that LOW is the voltage level
                                      // but actually the LED is on)
    delay(250);                       // Wait a bit
    digitalWrite(LED_PIN, HIGH);      // Turn the LED off by making the voltage HIGH
    delay(250);
    i--;
  }
}
///////////////////////////////

zfields, 2016-06-24:
I am using the AzureIoTHub simplesample_http. It works perfectly with 2.2, but upon grading to 2.3 I get this experience. I can connect to my router (reports assigned an IP address), but it keeps looping as though it failed and spews cryptic debugging errors.

mkeyno, 2016-06-25:
I have same problem with it , my connection get lost sometimes but module try to reconnect to Access point , and this make me problem  when using the websocket

igrr, 2016-06-25:
@RogerSchaefer
I have loaded the provided sketch, changed SSID and password and connection went fine.
Could you please provide serial output when debugging is enabled?
Please set "Debug Port: Serial" and "Debug Level: Core" in Arduino Tools menu, and add `Serial.setDebugOutput(true);` line after `Serial.begin(9600);`. This will print some info from WiFi stack.
We had an SDK update between 2.2.0 and 2.3.0. Because SDK updates sometimes don't migrate parameter storage area in flash correctly, this can cause connection issues. Debug output can give more hints at what's happening on your side.

@zfields since this is a third party library, best we can do is help you make sense of "cryptic debugging errors" (whatever that may mean). Please feel free to open an issue, and provide the exact sketch you are running, settings you have in IDE tools menu, and full serial output. Thanks.

RogerSchaefer, 2016-06-25:
Hello @igrr 
I have attached debug files from both Ver 2.2.0 and Ver 2.3.0
The object of the sketch is to get mqtt temperature data.
Ver 2.2.0 connects quickly
Ver 2.3.0 sometimes after 10 or 20 minutes

Roger

[sonoff_mqtt Ver 2.3.0.txt](https://github.com/esp8266/Arduino/files/333161/sonoff_mqtt.Ver.2.3.0.txt)
[sonoff_mqtt Ver 2.2.0.txt](https://github.com/esp8266/Arduino/files/333162/sonoff_mqtt.Ver.2.2.0.txt)

igrr, 2016-06-26:
Thanks for the logs, I was able to reproduce this with 2.3.0.

Apparently, with SDK 1.5.3, if you call WiFi.begin by the time WiFi connection is already established, WiFi connection state machine enters some weird state. SDK function `wifi_station_get_connect_status` keeps returning `STATION_CONNECTING`, even though the connection is in fact established.

This was resolved in SDK 1.5.4, which is available in current git version of ESP8266 core.
I tested the same code with git version and it worked fine.

As a workaround, you can replace `WiFi.begin(ssid, password);` with
```c++
if (WiFi.status() != WL_CONNECTED) {
  WiFi.begin(ssid, password);
}
```

or with
```c++
WiFi.persistent(false);
WiFi.mode(WIFI_OFF);   // this is a temporary line, to be removed after SDK update to 1.5.4
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
```

I will consider releasing 2.3.1 with a workaround for this issue in a couple of days.

SupotSaeEa, 2016-07-02:
I fact the same problem. 
Test on Generic ESP-01
Debug show something like not found my router.
Connect wifi with ESP8266WiFiMulti library.

**** Work fine on NodeMCU1.0 (ESP-12E Module)

Begin...
del if0
usl
mode : null
mode : sta(18:fe:34:98:6a:e2)
add if0
f r0, scandone
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f -180, scandone
no belkin.a57 found, reconnect after 1s
f r0, scandone
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f -180, scandone
no belkin.a57 found, reconnect after 1s

igrr, 2016-07-04:
@SupotSaeEa doesn't look like the same issue to me, because there are reconnect attempts visible in the log. Please check that you are setting correct flash size when compiling for you ESP-01 module. Usually they have smaller flash size than NodeMCU (512k or 1M).

SupotSaeEa, 2016-07-05:
@igrr I add WiFi.setOutputPower(0); ESP-01 is work again.

MPX4132, 2016-07-20:
@SupotSaeEa's workaround seems to fix the issue for the ESP03 as well.
```
WiFi.setOutputPower(0);
WiFi.begin(ssid, password);
```
I can also confirm the settings are correctly set on the IDE for the module, checked every time I compiled. I tried checking with the older versions (2.0.0 - 2.2.0) but apparently I can't compile with them. Keeps throwing the following:
```
Uploading 227872 bytes from to flash at 0x00000000
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
error: espcomm_upload_mem failed
```
I'd like to reiterate that it works perfectly fine with the 2.3.0 package, just needs the workaround mentioned.

PavelDorofeev, 2016-09-01:
I have the same problem on ESP-12 Lolin (Arduino 1.6.8). Reset ESP-12 ,  WiFI.begin() and WiFi.status() != WL_CONNECTED  waits forever. Only repower the WIFI router help me. 

This solves the problem fine.

`WiFi.persistent(false);`
`WiFi.mode(WIFI_OFF); ` 
`WiFi.mode(WIFI_STA);`
`WiFi.begin(ssid, password);`

Thanks iggr.

CapnNemo, 2016-10-08:
This may or may not be the same bug, but I'm using the Arduino IDE, and an Adafruit Huzzah, and the sketch I have works fine (Wifi.begin works fine and connects), until I upgrade the board from 2.1.0 to either 2.2.0 or 2.3.0 - in both cases, it no longer connects.

Also, neither of the two fixes (output power to zero, nor mode to WIFI_OFF) as shown above works (but maybe they only work when using lua?).

I haven't used the lua system, so for the moment this is only so helpful, but wanted to see if this sounds like the same bug, or a different bug, before I get too far.

Thanks!

SupotSaeEa, 2016-10-08:
I still confirm on solution to set output power to zero. Especially on
ESP-01.

*Supot Sae-Ea*
สุพจน์ แซ่เอีย
081-454-0853

19eighties, 2016-10-11:
Just to chime in with my own observations... I have been getting this bug also. Sometimes it is random, but happens the first time I try connecting a new ESP module 100% of the time with my sketch. If I upload the wificlientbasic sketch and it connects I can then upload my own sketch and it can then connect to my AP. I also find I cannot connect if I had just previously put the ESP module into AP mode. So then I have to do the "reset" as I just described.

kendo55, 2016-11-13:
i had that problem too !!!
i fixed it with: 

```
if (WiFi.status() != WL_CONNECTED) {  // FIX FOR USING 2.3.0 CORE (only .begin if not connected)
    WiFi.begin(ssid, password);       // connect to the network
}    

for ( count = 0; count < 20 ; count++ )  {
      delay(500);
      wifi_stat = WiFi.status();  
      Serial.print( count ); Serial.print( " WifiStat: " );
      Serial.println( wifi_stat );
      if ( wifi_stat == WL_CONNECTED ) {
        break;
      }
}

Serial.println("");

if ( count < 20 ) {
    Serial.println("WiFi connected");
}else{
    Serial.println("Client !!!NOT!!! connected");

   // create a SoftAP for access
   WiFi.softAP(ESPssid, ESPpassword);
   // etc..........
}
```

PrinceBalabis, 2017-01-17:
I have the same issue. I'm using an Adafruit HUZZAH ESP8266 with ESP8266Scheduler. WiFi.begin() doesn't work after uploading the code and WiFi.begin has previously been ran. I have to press the RST switch on the HUZZAH, then it works.

mlord, 2017-02-27:
I have been having similar looking issues with my NodeMCU (using Arduino) boards.  Very close to my AP, they connect to Wifi without issue, but move them 25 feet away and down one floor, and the initial "connect" takes minutes sometimes.

Finally tracked it down to DHCP.  The actual connection to the SSID takes maybe a second or less, but then it gets hung up on DHCP.  I suspect the internal RTOS isn't allowing sufficient time-outs for DHCP messages and gets it all wrong somehow.

Switching to Static IPs fixes everything for me here.  The board connects and is ready on Wifi within a second or three of power up now, regardless of distance from the AP.

ygator, 2017-05-11:
`if (WiFi.status() != WL_CONNECTED) {
  WiFi.begin(ssid, password);
}`

Worked for me.  Thanks

borch84, 2017-12-19:
Hello, I am having the same issue. Adafruit feather huzzah ESP8266 is not connecting to wifi. It used to connect but now it is stuck at the connecting step.

The code that I am using now is 2.1.0, however I've tried with 2.2 and 2.3.

This is the code I am using:

void setup() {

  //start the serial line for debugging
  Serial.begin(9600);
  Serial.setDebugOutput(true);

  //start wifi subsystem
  WiFi.disconnect();
  WiFi.persistent(false);
  WiFi.mode(WIFI_OFF);
  WiFi.mode(WIFI_STA);
  WiFi.setOutputPower(0);
  //WiFi.begin(ssid, password);
  //WiFi.config(ip,gateway,subnet);
  //attempt to connect to the WIFI network and then connect to the MQTT server
  reconnect();
  //wait a bit before starting the main loop
  //delay(2000);
}

void reconnect() {

  //attempt to connect to the wifi if connection is lost
  if(WiFi.status() != WL_CONNECTED){
    //debug printing
    Serial.print("Connecting to ");
    Serial.println(ssid);

    //loop while we wait for connection
    while (WiFi.status() != WL_CONNECTED) {
      WiFi.begin(ssid, password);
      Serial.print(".");
      delay(500);    
    }

    //print out some more debug once connected
    Serial.println("");
    Serial.println("WiFi connected");  
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
}

From the serial monitor I can see these messages:

f r0, ..0�~?�4�!�{�OAa�bcn 0
del if1
del if0
usl
mode : null
mode : sta(5c:cf:7f:00:6b:cc)
add if0
Connecting to Apartamentos villa
f r0, .....scandone
no Apartamentos villa found, reconnect after 1s
reconnect
f 0, ....scandone
no Apartamentos villa found, reconnect after 1s
reconnect
f -180, .....scandone

I've been thinking to reflash the ESP.

Any suggestions?

Thank you.

martin072, 2017-12-24:
Hi all, 

Is there any update on this issue? I am experiencing the same problems... the wemos doesn't seem to connect to the WiFi. This is my debug output:

Connecting to: FlyingDutchmen
del if0
usl
mode : null
mode : sta(60:01:94:33:00:c0)
add if0
...scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
...scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
......state: 5 -> 2 (2a0)
rm 0
.reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
......state: 5 -> 2 (2a0)
rm 0
..reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
.....state: 5 -> 2 (2a0)

closed 2018-01-02 by igrr

TobiasJiupato, 2018-01-05:
I had the same problem...It works fine to me:

WiFi.mode(WIFI_OFF);                //workaround
WiFi.mode(WIFI_AP_STA);	
WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

seblanc, 2018-02-21:
If you don't want to loose connection of the ESP8266 AP, you can use:
```
WiFi.mode(WIFI_AP); // workaround
WiFi.mode(WIFI_AP_STA);
WiFi.begin(ssid, password);
```

Works with NodeMCU 1.0 and esp8266 v2.4

NeilRShah, 2018-03-06:
I will confirm that the bug exists for me as well on a NodeMCU 1.0, using the Arduino IDE with the esp8266 board library installed. 

I used this workaround and it is working now: 
```
WiFi.persistent(false);
WiFi.mode(WIFI_OFF);   // this is a temporary line, to be removed after SDK update to 1.5.4
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
```

borqborq, 2018-04-30:
I've had the same intermittent problem with several NodeMCU v0.9 and v1.0 units. This code fixed it for me. Thanks to all that contribute!

```
#include <ESP8266WiFi.h>

const char* ssid = "YOUR-SSID";
const char* password = "YOUR-SSID-PASS";

void setup (){
  Serial.begin(115200);

  delay(3000);
  Serial.print("Connecting to ");
  Serial.println(ssid);
  Serial.println(password);

// ---------------This was the magic WiFi reconnect fix for me
  WiFi.persistent(false);
  WiFi.mode(WIFI_OFF);   // this is a temporary line, to be removed after SDK update to 1.5.4
  WiFi.mode(WIFI_STA);  
// ---------------END - WiFi reconnect fix

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  
  Serial.println("WiFi connected");
  Serial.println(WiFi.localIP());
}

void loop() {
  
  }
```

luffykesh, 2018-06-07:
I had a issue where, when ESP is assigned static IP from the router, it would connect to AP but fail to obtain IP.
The issue was persistent in lwIP v2.0 Higher Bandwidth and Low Memory.
Changing lwIP variant to "v1.4 Higher Bandwidth" solves the issue, does not matter if you WiFi.persistent(false);

d-a-v, 2018-06-07:
@luffykesh This is a bug in 2.4.1 fixed in latest git version.

luffykesh, 2018-06-07:
@d-a-v Thank you.
The git version does fix the issue of not connecting, but the connection(DHCP) time is around 21-25 seconds, whereas lwIP v1.4 establishes connection in 8-10 seconds.

d-a-v, 2018-06-07:
> IP assigned static from router.

Beeing more specific, in STA mode, getting address by DHCP server (no static IP address in the sketch), you are observing twice more delay with lwip-v2 than with lwip-v1.4 ?

luffykesh, 2018-06-07:
@d-a-v 
> you are observing twice more delay with lwip-v2 than with lwip-v1.4 ?
Yes.

luffykesh, 2018-06-08:
@d-a-v 
I have again encountered the issue. 
I have tried on both esp8266-12E and esp8266-12S with the git version, with lwIP v2, but cannot connect to the station.
lwIP v1.4 works with both git version and when installed from Arduino's board manager.

d-a-v, 2018-06-12:
@luffykesh 
Regarding connection delay, I played with the `interactive` sketch that you can find in File>examples>esp8266> and I noticed no much difference between the two versions of lwIP.
Regarding the connection issue you have, we need you to open a new issue describing the most accurately possible, by filling the issue template, what you are facing.

fabian727, 2018-06-24:
Hi alltogether,
is there already an official bug fix? I encounter connection problems too. 
I have the WEMOS D1 mini (with ESP8266MOD (so I think also called ESP12E))
and a WEMOS NodeMcu V3 of LoLin.

Both refuse to connect to my router. I already flashed them a while ago, but this won't work any more.
My test-sketch at the moment
`#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  
  WiFi.mode(WIFI_OFF);
  delay(2000);
  WiFi.mode(WIFI_AP_STA);
  WiFi.hostname("DEBUG");
  while (WiFi.status() != WL_CONNECTED) {
    WiFi.begin("WLAN_SSID",WLAN_PWD");
    delay(5000);
    Serial.print(".");
  }

    Serial.println("");
    Serial.println("WiFi connected");
   
    // Print the IP address
    Serial.print("My IP: ");
    Serial.println(WiFi.localIP());

  delay(500);
}

void loop() {
  delay(5000);
}

`

I also tested some sketches from here and to setPowerOutput to 0. Nothing helped.

My DebugOutput:
`done
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
......scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
......scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
.....scandone
state: 0 -> 2 (b0)
. and so on`

I changed between Wifi V1.4, V2 low mem and high band. Nothing...

Any further ideas?

Thanks,
Fabian

CapnNemo, 2018-06-24:
Possibly you need to turn WiFi back on again?

But this thread is mostly about being connected,  then occasionally seeing
your ESP disconnect after an arbitrary time.

mmaxus35, 2018-06-27:
I am dealing with this issue right now and i have applied to code the given solutions.
wifi.persistent(false);
wifi.begin(WIFI_OFF); 

So far, i tested my system under a 8W fan load, i made POST request instantly to cause the system drop from wifi because of high current consumption and it did. Then i coded esp to re-estabilish wifi connection . I tested the system for 50 times to drop it and re-estabilish, as result at the 51.th of the trial it did not re-estabilish the connection ( i have waited 15 min to be re-connected ) but it did not. So far this is the temporarly best solution. Moreoever, if there are permanent solution for this problem, could you share please ?

Thanks.

d-a-v, 2018-06-27:
@mmaxus35 Please try latest master. Yesterday's fixes were about WiFi disconnection and reconnection.

mmaxus35, 2018-06-27:
@d-a-v  Okey ' i will thank a lot 👍

fabian727, 2018-06-28:
as already stated I changed from WPA & WPA2 to only WPA2. This module with this flashed code worked. Now nothing works. I had to shut off and on again my router and reconfigure many devices to work again... seems to be only a lucky solution

TobiNorris, 2018-06-28:
@d-a-v , how can I use this source code to upload in my board? I got the master branch, but I don´t know how to upload this into my board.

d-a-v, 2018-06-28:
@mmaxus35 I don't know the status of PlatformIO. The fixes mentioned above are very recent and are for the arduino core with lwIP-v2.

@fabian27 please open a *new issue* and fill the issue template with all details you can.

@TobiNorris Using the git master branch (all documentation's steps must be followed) or downloading the arduino core through the board manager result in the same IDE and tools. Press the Arduino IDE upload button. 

To all, this *thread* #2186 is an *old issue*, about an old version of the core, fixed, solved and **closed**. Your issues are unrelated with the OP. Please open new issues with your specific problems.

adrionics, 2019-04-25:
This is still an issue in 2.5.0. Some device of mine got into a weird mode and was crashing after WiFi.begin or when using .disconnect().

I resolved it by using the following sequence (I wanted mode to be persistent, so the extra persistent commands):

```
  WiFiMode_t prepareWiFi_m = WiFi.getMode();
  WiFi.persistent(false);
  if (prepareWiFi_m == 3) {
    WiFi.mode(WIFI_STA);
  } else {
    WiFi.mode(WIFI_OFF);
  }
  WiFi.persistent(true);
  WiFi.mode(prepareWiFi_m);
  WiFi.persistent(false);
`
````

skywalker1979, 2019-04-30:
> This is still an issue in 2.5.0. Some device of mine got into a weird mode and was crashing after WiFi.begin or when using .disconnect().
> 
> I resolved it by using the following sequence (I wanted mode to be persistent, so the extra persistent commands):
> 
> ```
>   WiFiMode_t prepareWiFi_m = WiFi.getMode();
>   WiFi.persistent(false);
>   if (prepareWiFi_m == 3) {
>     WiFi.mode(WIFI_AP);
>   } else {
>     WiFi.mode(WIFI_OFF);
>   }
>   WiFi.persistent(true);
>   WiFi.mode(prepareWiFi_m);
>   WiFi.persistent(false);
> `
> ```

i agree...i'm on 2.5.0 as well...but i think i saw this problem this morning....when my thermometer stopped delivering temperature measurements...event if it's been working for more than a month without problems.

shrimantpatel, 2019-10-08:
I was the newest addition to this issue.....after reading thru this post below is what has worked for me, my problem was more with "WiFi.status() " in getting to know if wifi was connected before switching to AP mode , the prevailing while loop with condition " (WiFi.waitForConnectResult() != WL_CONNECTED)" used to block for ever if your defined SSID was not available and would never get past it to switch to AP mode, I finally used the temp status WiFi.status() == WL_IDLE_STATUS to get past this issue:

`	              WiFi.mode(WIFI_OFF);
			delay(100);
			WiFi.mode(WIFI_STA);
			WiFi.begin(WiFi.SSID(), WiFi.psk());
			while (WiFi.status() == WL_IDLE_STATUS) {
//**When the control is out of this loop its a guarantee that
//**1.WiFi.status() will be equal to WL_CONNECTED  if the SSID being used is available
//**2.WiFi.status() will be equal to WL_CONNECT_FAILED  if the SSID being used is NOT available or some other failures
				Serial.print(".");
			}`

alitar3, 2020-03-30:
If you use Arduino 
you must set "erase  flash" to "sketch + wifi settings".
"erase flash" in board manager.