esp8266/Arduino #2187: Crash after multiple execution of queryServices

kelcon, 2016-06-24:
### Basic Infos

#### Hardware
Hardware:		Huzzah
Core Version:      	2.?.?

### Description

ESP8266 (Huzzah) crashes after multiple (about 300) executions of queryServices

### Settings in IDE

Module:  			Huzzah
Flash Size: 		4MB
CPU Frequency      :80Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
 
const char* ssid     = "...";
const char* password = "...";

void setup() {
  Serial.begin(115200);
  delay(100);
 
  // We start by connecting to a WiFi network
 
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  if (!MDNS.begin("leaktest")) {
      Serial.println("Fatal! Cannot start discovery");
  }

}

int i = 1;

void loop() {

  Serial.println(i++);
  int n = MDNS.queryService("bitoad", "tcp"); 

}

```

### Debug Messages

```
295
296
297

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffda0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff40:  3fffdc00 000002f4 3ffeccf0 401004f4  
3fffff50:  00000000 00000000 3ffeccf0 40106b88  
3fffff60:  40104f6e 058ce05a 3ffedc28 402083f6  
3fffff70:  4021a05c 3ffe8884 402088b6 00000002  
3fffff80:  4021a082 3fffdab0 40209ab7 3fffdab0  
3fffff90:  00000000 3fffdab0 3ffeeb88 40203c2b  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

Decoding 13 results
0x401004f4: calloc at C:\Users\Bitosf\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1682
0x40106b88: pvPortZalloc at C:\Users\Bitosf\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/heap.c line 33
0x40104f6e: wdt_feed at ?? line ?
0x402083f6: pm_set_sleep_time at ?? line ?
0x4021a05c: ets_timer_handler_isr at ?? line ?
0x402088b6: pm_get_sleep_type at ?? line ?
0x4021a082: ets_timer_handler_isr at ?? line ?
0x40209ab7: pm_post at ?? line ?
0x40203c2b: loop_task at C:\Users\Bitosf\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43

```

igrr, 2016-06-25:
Possibly related to #2169

bitoadpl, 2016-06-28:
May I help with debug?

kelcon, 2016-06-28:
How can I help with debugging?

devyte, 2017-10-16:
@kelcon there was a fix merged that looks relevant. Is this issue still valid with latest git?

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte