esp8266/Arduino #2188: ESP8266WiFiMulti.cpp code inspection: Unexecutable code in ESP8266WiFiMulti::run

SteveToulouse, 2016-06-24:
While looking through the code to understand what's being done and see how issue #2117 could be resolved, [lines 51 to 63](https://github.com/esp8266/Arduino/blob/633e48f3aec5f1c3c11d4498fc90d378d49e6e9f/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L51-L63) caught my eye.

The conditions of the "if"s at lines L51 `if (scanResult > 0)` and L61 `if (scanResult <= 0)` are mutually exclusive. So the code at line 62 can never be executed. Unless I've missed something?

Not a bug as such but it adds complexity when reading the code. The enormous "else" clause, from 63 onwards to nearly the end of `::run`, could be reduced one indentation level.

devyte, 2017-10-16:
Code reference is still present in latest git. Most likely this
`if (scanResult <= 0)
`
should be
`if (scanResult == 0)`

devyte, 2017-11-20:
@SteveToulouse please take a look at the referenced PR.

devyte, 2017-11-21:
I merged the PR after some sanity checks. Any new problems will be handled in a different issue.
Closing.

closed 2017-11-21 by devyte