esp8266/Arduino #2189: WiFi.mode(WIFI_AP_STA) SSID in AP, Issues when use SPIFFS

JLC52, 2016-06-24:
### Basic Infos
ESP8266-07 
#### Hardware
Hardware:			ESP-07
Core Version:      	

### Description

I have developed an sketch in Arduino Environment 1.6.8 for MAC OSx. The basic functionality is to set up an mixed wifi mode with AP and STA. A basic sever, servers HTTP pages for configuration purposes.
When an external Push button is pressed, the SSID of the Soft AP is changed to other name in order to show the event. when the event es reseted by the html page button, the SSID is changed again to the normal name. 
I have used the Deep Sleep to save power. The system wakes up each 5 minutes or when the pushbutton is pressed. I am using SPIFFS file system to save the device state before to go  DeepSpleep and recover it After wake up. 

Problem description

At the first time, I have not used the SPIFFS. All works fine. In normal state the AP SSID was "PB-xxxx-0". (with out the ")

Then  I pushed the button and the SSID change to "PB-xxxx-1". Report the event to a server, and go to Deep Sleep

If the push button is pressed, the device is waked up pulsing CH_PD, and set the SSID to "PB-xxxx-1".
and send the message. Then go to Deep Sleep...

Each 5 minutes the device wakes up and set the SSID to "PB-xxxx-0". and If the push button is pressed, the procedure is the same. 

Then I have ADDED the SPIFFS capabilities to remember the last state of the device when the dmoule went to sleep. and the problems have stated....

The main symptom is that the SSID is changing, the firs time is "PB-xxxx-1" then "PB-xxxx-1    ", then "PB-xxxx.-1           ". 
Also I have several restarts due stack or wdt. and Some String comparations does not work (???) 

if I not run SPIFFS.begin() and I have not use the SPIFSS API the app works fine...

Question: wich is the proper SPIFFS memory configuration? How much Memory has ESP2866-07?

I have tried with 4M 1M SPIFFS and the module is no working......

### Settings in IDE

ARDUINO 1.6.8 MAC osx

Module:  			?Generic ESP8266 Module?
Flash Size: 		?512k (64k SPIFFS)?
CPU Frequency:		?80Mhz?
Flash Mode:			?DIO?
Flash Frequency:	?40Mhz?
Upload Using:		? SERIAL?
Reset Method:		?ck ?

### Sketch
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <Wire.h>
#include "FS.h"

extern "C" {
       #include "user_interface.h"
}

extern "C" {
#include "gpio.h"
}

void setup() 
{ 
  wifi_station_disconnect(); 
  initHardware();
  loadfiles();
  setupWiFi(); 
}

void loop() 
{

.........
.......
        DeviceStateFile.updatefile(DeviceStateFileName);
        
        ESP.deepSleep(SleepTime * 1000);
}

void setupWiFi()
{

//wifi_station_disconnect();

  WiFi.mode(WIFI_AP_STA);
// Build AP name

  uint8_t mac[WL_MAC_ADDR_LENGTH];
  
  WiFi.softAPmacAddress(mac);

  AP_BSSID = String (mac [0],HEX) +":"+  String (mac [1],HEX) +":"+ String (mac [2],HEX) +":"+ String (mac [3],HEX) +":"+ String (mac [4],HEX) +":"+ String (mac [5],HEX) ;

  
  String macID = String(mac[WL_MAC_ADDR_LENGTH - 2], HEX) +
                 String(mac[WL_MAC_ADDR_LENGTH - 1], HEX);
  
  macID.toUpperCase();
  DeviceName = "PB-" + macID ;

 AP_SSID = "@" + DeviceName + "-" + Attention;

  char AP_NameChar[AP_SSID.length() + 1];
  memset(AP_NameChar, 0, AP_SSID.length() + 1);

  for (int i=0; i<AP_SSID.length(); i++)
    AP_NameChar[i] = AP_SSID.charAt(i);
    
  WiFi.softAP (AP_NameChar, WiFiAPPSK);

  IPAddress IP = WiFi.softAPIP ();
  AP_LocalIP = IP.toString();

};

class DeviceState
{
  
  public: int create (String DeviceStateFileName)
  {

    if (DeviceStateFileName == "")
             return 0;
  
    // open file for writing
 
    if (!SPIFFS.exists (DeviceStateFileName)) // if the file does not exist, create the default one
    {
      Serial.println("file open failed: Not exists");
      Serial.println("====== Writing to SPIFFS file =========");
      File f = SPIFFS.open(DeviceStateFileName, "w");
      f.println("DeviceState=DeviceFreshBoot");
      f.println("AlertStatus=00");
      f.println("Attention=0");      
      f.println("LastSSIDConnected=NONE");
      f.println("CycleNumber=0");
      f.close();
      Serial.println("====== Reading from the SPIFFS file: " + DeviceStateFileName + " recently created =========");
      
      f = SPIFFS.open(DeviceStateFileName, "r");
      while (f.available())
      {
         String line = f.readStringUntil('\n');
         Serial.println(line);
      } 

      f.close();

    }else
    {
      Serial.println("file exists!!");
      File f = SPIFFS.open(DeviceStateFileName, "r");
      Serial.println("====== Reading from the SPIFFS file:" + DeviceStateFileName + " existing =========");

      while (f.available())
      {
         String line = f.readStringUntil('\n');
         Serial.println(line);
      } 

     f.close();
    }
  
  };

  public: int updatefile (String DeviceStateFileName)
  {

    if (DeviceStateFileName == "")
             return 0;
  
    // open file for writing
 
      Serial.println("====== Updating "+DeviceStateFileName + "CycleNumber: " + CycleNumber +" =========");

      File f = SPIFFS.open(DeviceStateFileName, "w");
      
      f.println("DeviceState="+DeviceStatus);
      f.println("AlertStatus="+ AlertStatus); 
      f.println("Attention="+ Attention );           
      f.println("LastSSIDConnected="+STA_ConnectedSSID);
      f.println("CycleNumber="+String (CycleNumber));
      
      f.close();

      f = SPIFFS.open(DeviceStateFileName, "r");
      
      Serial.println("====== Reading from the SPIFFS file:" + DeviceStateFileName + " updated =========");

      while (f.available())
      {
         String line = f.readStringUntil('\n');
         Serial.println(line);
      } 

     f.close();
  };

 public: int loadfile (String DeviceStateFileName)
  {

    if (DeviceStateFileName == "")
             return 0;
  
    // open file for reading
 
      Serial.println("====== loading data from "+DeviceStateFileName+" =========");

      File f = SPIFFS.open(DeviceStateFileName, "r");

      if (!f)
          Serial.println("Error opening"+DeviceStateFileName+" =========");
      
      while (f.available())
      {
         String line = f.readStringUntil('\n');
         Serial.println(line);
         
        if (line.indexOf("DeviceState=") != -1)
            DeviceStatus = line.substring (line.indexOf("=")+1); 
                     
        if (line.indexOf("AlertStatus=") != -1)
            AlertStatus = line.substring (line.indexOf("=")+1);
         if (line.indexOf("Attention=") != -1)
            Attention = line.substring (line.indexOf("=")+1);   
        if (line.indexOf("LastSSIDConnected=") != -1)
            LastSSIDConnected = line.substring (line.indexOf("=")+1);
        if (line.indexOf("CycleNumber=") != -1)
        {
          String s = line.substring ((line.indexOf("=")+1));
          CycleNumber = s.toInt() + 1 ; 
        }
      }
      f.close();  
 Serial.println ("  KeyState[0] >" + KeyState[0] +"<");
  };
};

DeviceState DeviceStateFile;

void loadfiles ()
{
 float a;
   SPIFFS.begin();
  // Next lines have to be done ONLY ONCE!!!!! if there is not the wifi data file, the file system is not formatted

  /////////////////////////////
  // WifiFileName = "/KnownWifi.dat"; this is a global variable
  // DeviceStateFileName = "/DeviceState.dat";this is a global variable
  // LastMessagesFileName = "/LastMessages.dat"; this is a global variable
  //////////////////////////////
   
   if (!SPIFFS.exists(DeviceStateFileName))
   {
      a = system_get_time (); //microseconds
      a = a/ 1000000;
      Serial.println(String (a, 6) + "--->Please wait for File: "+DeviceStateFileName+" Creation");   
      DeviceStateFile.create (DeviceStateFileName);
   } 
   
  DeviceStateFile.loadfile (DeviceStateFileName);
    
  
   
  if (!SPIFFS.exists(WifiFileName)) // if the file does not exist, create the default one
  {
    a = system_get_time (); //microseconds
    a = a/ 1000000;    
    Serial.println(String (a, 6) + "--->Please wait for File: "+WifiFileName+" Creation");      
    KnownWifiFile.create (WifiFileName);    
  }
  if (!SPIFFS.exists(LastMessagesFileName))
  {
      a = system_get_time (); //microseconds
      a = a/ 1000000;
      Serial.println(String (a, 6) + "--->Please wait for File: "+LastMessagesFileName+" Creation");
      LastMessagesFile.create (LastMessagesFileName);
  }
      
Serial.println ("  spiffs started  RAM: " + String (system_get_free_heap_size(),DEC));

    a = system_get_time (); //microseconds
    a = a/ 1000000;
     
    Serial.println (String (a, 6) +"--->File Creation ready");   

}

### Debug Messages

```
messages here
```

igrr, 2016-06-25:
Please load CheckFlashConfig example into your module (File > Examples > ESP8266 > CheckFlashConfig) and report back with serial output it prints. Also please update your question with settings you have in the IDE (like `Module`, `Flash Size`, etc.)

JLC52, 2016-06-25:
Hi igrr; thanks for your interest and fast response.

Here is the Serial output of 
CheckFlashConfig example:
----------------------------
Flash real id:   001440E0
Flash real size: 1048576

Flash ide  size: 524288
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration wrong!

Attached is the screenshot with the IDE Configuration (is the same I am using in my sketch)

[image]
:

The HW data got form my app:
-----------------------------------------
HW data:
|LTS=
|ChipID =a3314
|SdkVersion =1.5.3(aec24ac9)
|CpuFreq =80
|System Boot Version=31
|UserBinAddress= 0x0
|SystemBootMode=1
|FlashSizeMap=FLASH_SIZE_4M_MAP_256_256

the code I have used to get that HW data:
--------------------------------------------------------
  DeviceID= String (system_get_chip_id(),HEX);

          String Response = "|LTS=";
          Response += "|ChipID =" + String ( system_get_chip_id(),HEX);
          Response += "|SdkVersion =" + String ( system_get_sdk_version());
          Response += "|CpuFreq =" + String (system_get_cpu_freq(),DEC);
          Response += "|System Boot Version=" + String (system_get_boot_version(),DEC);
          Response += "|UserBinAddress= 0x" + String (system_get_userbin_addr(), HEX);
          Response += "|SystemBootMode=" + String (system_get_boot_mode(),DEC);
                            // SYS_BOOT_ENHANCE_MODE – 0
                            //SYS_BOOT_NORMAL_MODE – 1
          Response += "|FlashSizeMap=";  

          int sizeMem = system_get_flash_size_map();

          switch (sizeMem)
          {
            case 0:
            Response += "FLASH_SIZE_4M_MAP_256_256";
            break;
            case 1:
            Response += "FLASH_SIZE_2M";
            break;
            case 2:
            Response += "FLASH_SIZE_8M_MAP_512_512";
            break;
            case 3:
            Response += "FLASH_SIZE_16M_MAP_512_512";
            break;
            case 4:
            Response += "FLASH_SIZE_32M_MAP_512_512";
            break;
            case 5:
            Response += "FLASH_SIZE_16M_MAP_1024_1024";
            break;
           case 6:
            Response += "FLASH_SIZE_32M_MAP_1024_1024";
            break;

          }
          
 
    Serial.println (" HW data: " + Response); 

The include files that I used in my app:
---------------------------------------------------
ESP8266WiFi.h
ESP8266WiFiMulti.h
ESP8266HTTPClient.h
Wire.h
"FS.h"

extern "C" {
       #include "user_interface.h"
}

extern "C" {
#include "gpio.h"
}

The Module is ESP8266-07
-------------------------------------
[image]

JLC52, 2016-06-25:
Hi, I just tried with ID configs on 1M (512K SPIFFS)......up to 1M (64K SPIFFS)  and the CheckFalshConfig was ok but the symptoms on my APP are  the same as before...

igrr, 2016-06-26:
I tried to compiled your sketch and got following errors:
```
/var/folders/3y/f7lzmp5j4slft2l2f9c6_69r0000gq/T/arduino_modified_sketch_287576/sketch_jun26a.ino: In function 'void setup()':
sketch_jun26a:18: error: 'initHardware' was not declared in this scope
   initHardware();
                ^
/var/folders/3y/f7lzmp5j4slft2l2f9c6_69r0000gq/T/arduino_modified_sketch_287576/sketch_jun26a.ino: In function 'void loop()':
sketch_jun26a:26: error: expected primary-expression before '...' token
 .........
 ^
sketch_jun26a:26: error: expected ';' before '...' token
sketch_jun26a:30: error: 'SleepTime' was not declared in this scope
         ESP.deepSleep(SleepTime * 1000);
                       ^
/var/folders/3y/f7lzmp5j4slft2l2f9c6_69r0000gq/T/arduino_modified_sketch_287576/sketch_jun26a.ino: In function 'void setupWiFi()':
sketch_jun26a:45: error: 'AP_BSSID' was not declared in this scope
   AP_BSSID = String (mac [0],HEX) +":"+  String (mac [1],HEX) +":"+ String (mac [2],HEX) +":"+ String (mac [3],HEX) +":"+ String (mac [4],HEX) +":"+ String (mac [5],HEX) ;
   ^
sketch_jun26a:52: error: 'DeviceName' was not declared in this scope
   DeviceName = "PB-" + macID ;
   ^
sketch_jun26a:54: error: 'AP_SSID' was not declared in this scope
  AP_SSID = "@" + DeviceName + "-" + Attention;
  ^
sketch_jun26a:54: error: 'Attention' was not declared in this scope
  AP_SSID = "@" + DeviceName + "-" + Attention;
                                     ^
sketch_jun26a:58: error: 'AP_NameChar' was not declared in this scope
   memset(AP_NameChar, 0, AP_SSID.length() + 1);
          ^
sketch_jun26a:63: error: 'WiFiAPPSK' was not declared in this scope
   WiFi.softAP (AP_NameChar, WiFiAPPSK);
                             ^
sketch_jun26a:67: error: 'AP_LocalIP' was not declared in this scope
   AP_LocalIP = IP.toString();
   ^
/var/folders/3y/f7lzmp5j4slft2l2f9c6_69r0000gq/T/arduino_modified_sketch_287576/sketch_jun26a.ino: In member function 'int DeviceState::updatefile(String)':
sketch_jun26a:129: error: 'CycleNumber' was not declared in this scope
       Serial.println("====== Updating "+DeviceStateFileName + "CycleNumber: " + CycleNumber +" =========");
                                                                                 ^
sketch_jun26a:133: error: 'DeviceStatus' was not declared in this scope
       f.println("DeviceState="+DeviceStatus);
                                ^
sketch_jun26a:134: error: 'AlertStatus' was not declared in this scope
       f.println("AlertStatus="+ AlertStatus); 
                                 ^
sketch_jun26a:135: error: 'Attention' was not declared in this scope
       f.println("Attention="+ Attention );           
                               ^
sketch_jun26a:136: error: 'STA_ConnectedSSID' was not declared in this scope
       f.println("LastSSIDConnected="+STA_ConnectedSSID);
                                      ^
/var/folders/3y/f7lzmp5j4slft2l2f9c6_69r0000gq/T/arduino_modified_sketch_287576/sketch_jun26a.ino: In member function 'int DeviceState::loadfile(String)':
sketch_jun26a:175: error: 'DeviceStatus' was not declared in this scope
             DeviceStatus = line.substring (line.indexOf("=")+1); 
             ^
sketch_jun26a:178: error: 'AlertStatus' was not declared in this scope
             AlertStatus = line.substring (line.indexOf("=")+1);
             ^
sketch_jun26a:180: error: 'Attention' was not declared in this scope
             Attention = line.substring (line.indexOf("=")+1);   
             ^
sketch_jun26a:182: error: 'LastSSIDConnected' was not declared in this scope
             LastSSIDConnected = line.substring (line.indexOf("=")+1);
             ^
sketch_jun26a:186: error: 'CycleNumber' was not declared in this scope
           CycleNumber = s.toInt() + 1 ; 
           ^
sketch_jun26a:190: error: 'KeyState' was not declared in this scope
  Serial.println ("  KeyState[0] >" + KeyState[0] +"<");
                                      ^
/var/folders/3y/f7lzmp5j4slft2l2f9c6_69r0000gq/T/arduino_modified_sketch_287576/sketch_jun26a.ino: In function 'void loadfiles()':
sketch_jun26a:210: error: 'DeviceStateFileName' was not declared in this scope
    if (!SPIFFS.exists(DeviceStateFileName))
                       ^
sketch_jun26a:218: error: 'DeviceStateFileName' was not declared in this scope
   DeviceStateFile.loadfile (DeviceStateFileName);
                             ^
sketch_jun26a:222: error: 'WifiFileName' was not declared in this scope
   if (!SPIFFS.exists(WifiFileName)) // if the file does not exist, create the default one
                      ^
sketch_jun26a:227: error: 'KnownWifiFile' was not declared in this scope
     KnownWifiFile.create (WifiFileName);    
     ^
sketch_jun26a:229: error: 'LastMessagesFileName' was not declared in this scope
   if (!SPIFFS.exists(LastMessagesFileName))
                      ^
sketch_jun26a:234: error: 'LastMessagesFile' was not declared in this scope
       LastMessagesFile.create (LastMessagesFileName);
       ^
exit status 1
'initHardware' was not declared in this scope

```

Please provide a working sketch.

JLC52, 2016-06-27:
Hi; here is a working sketch. 
You need to connect GPIO16 to Reset pin in ordet ro wakeup from deep sleep. and a push button to ground in pin GPIO 5 to test the push event. A 10K pull up resistor is OK to that pin. Pin GPIO 4 is for a led. 

There is a constant to switch the use the SPIFFS utility on or OFF. 

#define USESPIFFS 0 // set USESPIFFS to 0 in order to skip DeviceState save before DeepSleep and recall at wakeup 

is USESPIFFS is 0, the sketch works fine. 60 seconds Working and 60 Seconds in deep sleep. if you press the button during working cycle, you will get an event and the device put e message in a buffer (you can see several events since there is no debouncing). After the event, the AP said is changed. when the device goes to Deep sleep, a message shows that it is in Alert mode. When the device wakes up, the previous state is not persistent and the device start in normal mode. 

If you se t#define USESPIFFS 1, the SPIFFS functionality is enabled. If the SPIFSS file system is blank, the sketch creates the files and load the data with default values. Every time the device goes to Deep Sleep, the device State is save in /DeviceStateFile.dat. At wake up, if the file exists, the sketch loads the previous saved state. 
Problems detected: The device does not go to deep sleep if there is not an new event registered during the working time. 
The SSID is not well presented to other devices such PC, smarphones etc. 
I not use SPIFFS.end() before  go to Deep Sleep since there is a error at compiling....

[workingsketch.zip](https://github.com/esp8266/Arduino/files/334067/workingsketch.zip)

you have an internal web server in the sketch at 192.168.4.1/HtmlPanel to see the button states.
you can send http commands to set the wifi connection http://192.168.4.1/WifiConnect=SSID,PWD

thank you for your time

best regards

devyte, 2017-10-16:
@JLC52 Assuming the issue is still valid, I suggest erasing the entire flash before reloading your (correctly built) sketch.
Not the right place to request help with your particular project. Also, this looks like a user error. Closing per #3655 .

closed 2017-10-16 by devyte