esp8266/Arduino #2190: Where does my flash go?

MCUdude, 2016-06-24:
Hi! I'm currently using an ESP-12E module with 4M flash memory. In the IDE i can choose <b>4M (1M SPIFFS)</b> or <b>4M (3M SPIFFS)</b>. No matter what I select the IDE tells me the maximum code space is about 1M.

> Sketch uses 222,141 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.

 If 1M is dedicated to SPIFFS, and 1M is dedicated to the program storage, where did the 2M go? 😕

hallard, 2016-06-25:
@MCUdude 
Conception of ESP8266 chip can't have program over 1MB (pointer size if I remember correctly) so you will never see more than 1Mb for code. 
So Why chosing 4M/1M instead of 4M/3M ? try to upload SPIFFS using serial, 1M is so much faster (yeah, at 115200 even 1M is so slow).

igrr, 2016-06-25:
The reason we can not have more than 1Mb of code in flash has to do with a hardware limitation. Flash cache hardware on the ESP8266 only allows to map 1Mb of code into the CPU address space at any given time. You can switch mapping offset, so technically you can have more than 1Mb total, but switching such "banks" on the fly is not easy and efficient, so we don't bother doing that. Besides, no one has so far complained about 1Mb of code space being insufficient for practical purposes.
The reason to have 4M/1M SPIFFS split which @hallard has given is correct — uploading 3M takes a long time so sometimes you can just use 1M. Other 2M of flash can still be used with ESP.flashRead andd ESP.flashWrite APIs if necessary.

closed 2016-06-25 by igrr