esp8266/Arduino #2192: Blink Example won't blink

Gorkde, 2016-06-25:
It worked without problems with 2.2.0
My LED is ok since it's blinking when uploading.

The Check Config Sketch can upload and is working fine.

Just the LED isn't blinking with ESP8266 Blink example.

igrr, 2016-06-25:
Please post the sketch you are using, specify which hardware you run it on, and the settings in the IDE which you use.

Gorkde, 2016-06-25:
I'm using a ESP-12F
I'm using the standard Blink Sketch of ESP8266

I did now measure Pin 1 and it's changing state as it should.

On one side (the one pinting away of pin 1) of the LED I can measure VCC
But the other side seems to be floating just touching it changes state from off to dim on all the time (0,8V)  and off again.

[image]

```
/*
 ESP8266 Blink by Simon Peter
 Blink the blue LED on the ESP-01 module
 This example code is in the public domain
 
 The blue LED on the ESP-01 module is connected to GPIO1 
 (which is also the TXD pin; so we cannot use Serial.print() at the same time)
 
 Note that this sketch uses LED_BUILTIN to find the pin with the internal LED
*/

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
```

igrr, 2016-06-25:
@Gorkde, there is no way this sketch could have triggered an LED on ESP12F with 2.2.0. 
In fact, I have a copy of 2.2.0 installed and I have just verified that this doesn't work on 2.2.0.
Reason for this is simple — unlike ESP-01, on ESP12 LED is connected to GPIO2, not GPIO1 :)
It blinks during upload because GPIO2 is configured as UART0_TXD by bootloader, so it acts in the same way as GPIO1.

I think example should be updated to note that for ESP12E/F, pin has to changed to 2.

Gorkde, 2016-06-25:
Ok, so maybe I had this messed up with my tests on the Wemos but on any of those I had 2.2.0 blink the led I'm pretty sure. But at least I know the reason now, thanks!

Gorkde, 2016-06-25:
Maybe you shoud just add a Blink_ESP-12 example and a LED_BUILTIN_12.

SteveToulouse, 2016-06-25:
Sorry to butt in. Wouldn't it be clearer (though perhaps more work) to define two "generic" variants in the boards list instead of one - one for ESP01-ESPxx where the built-in LED is on GPIO1, and another for ESP12E and others where the LED on GPIO2 (the d1, d1-mini, espino, espresso variants have LED_BUILTIN = 2 already)? There are perhaps other constants which differ too, because of the extra pins on the 12E/F?

Gorkde, 2016-06-25:
That's what I thought later on also. 
Maybe the best way.

bigbuglamer, 2016-06-27:
There exists esp12e boards with led connected to gpio1 (early release ones). So i personnaly think ... no fix there possible.

vicnevicne, 2016-09-04:
Well, "Blink" is probably the most used sample, and I guess most modules sold today (stand-alone or on a board) have their internal led on GPIO2, so I think it's important that it is made simple for newcomers.
I would suggest adding another define for the alternate led pin, and change the sample to either make it clear in a comment that it must be adapted according to the board... or better yet, make both pins blink !

ghosty-be, 2017-01-18:
darn, been looking for this for half an hour now ... read some examples where it said 1 or 3 ... never tried pin 2, was thinking that I did something wrong :)

vinogradov, 2017-03-18:
Blink example works fine for me. I have the following configuration:

[image]

pjparsons, 2017-05-08:
sorry to branch off on this topic... how do I change the (Port: "/dev/cu.SLAB_USBtoUART")  Mine only shows up as com 1 and com 3.  I know it is connect to com 3 because I get 2 squares in the serial monitor window when I connect reset to ground.  I cannot get this LED to flash... It doesn't flash when it is transferring.  Arduino IDE says it did compile

999chris, 2017-07-17:
Add `#define LED_BUILTIN 2` to line 11

dimitre, 2017-08-05:
I don't know if it adds anything but in ESP-01 it uses a constant called BUILTIN_LED instead of LED_BUILTIN

jwidht, 2017-08-16:
I had to  use the PIN 2 too. Then the LED on the board would blink.
The example may need to be updated.
LED_BUILTIN is defined for Arduino (afaik pin  13) and for this WiFI board where it is pin 2

I got 
Wireless module NodeMcu Lua Wifi  ESP8266MOD WIFI Network Development Board Based ESP8266
 
MODEL: ESP8266MOD
ISM: 2.4GHZ
PA: +25DBM
 802.11B/G/N

brendanmatkin, 2017-10-23:
@devyte @igrr I've just been adding 
`#define LED_BUILTIN 2`
to all of my sketches using ESP-12x. 

Maybe putting that behind a comment with an explanation might be the easiest way? Or possibly changing the generic ESP-8266 `LED_BUILTIN` board definition to pin 2 and then adding the define+comment for ESP-01 (seems to me the ESP-12x's are far more common now)?

On the other hand, I agree that it might be clearest to beginners (who are most likely to be using Blink), if there were Blink (ESP-01) and Blink (ESP-12) examples. Plus maybe just mentioning the uncommon generic boards like ESP-07 etc in the comments.

devyte, 2017-10-25:
I think this will likely be addressed by #3722 .

suhas101, 2017-10-26:
#define LED_BUILTIN 2
adding this line in the beginning of the code solved the problem for me :)

sonugupta4636, 2017-11-05:
Yes, even I had the same problem.
Adding **#define LED_BUILTIN 2** at the top helped me resolving this.
Finding out this is really onerous for newcomers. This should be updated somewhere in comments.

chamithchathuka, 2017-11-18:
@suhas101  and @sonugupta4636  Thank you, both! it worked for me when added #define LED_BUILTIN 2
to the top of the script

devyte, 2017-11-20:
@d-a-v is this something that should be addressed by the boards.txt generator? In the generic esp8266 module case, should the BUILTIN_LED be configurable?

d-a-v, 2017-11-20:
a led menu for generic boards is added in #3722.

devyte, 2017-11-22:
@d-a-v It is my understanding that #3722 will be merged after 2.4 release.
@igrr If you agree, I will move this to milestone 2.5. In fact, all boards.txt-related issues should be moved to 2.5.

d-a-v, 2017-11-22:
@devyte yes, and If I may, boards.txt in its current form needs fixes,
I will make a PR so you can examine them.

devyte, 2017-12-12:
@d-a-v I merged #3940 . I don't see a led menu for the Generic boards, so I assume that will be added later? If so, that's ok

d-a-v, 2017-12-12:
right, it will be added with #3722

AhmedTanvir8, 2018-01-02:
This seems to differ by which firmware version you are running. 
For me, D0 (zero) worked with the stock blink example in arduino IDE, using and external LED. Using other pins or changing the code didnt help at all. 
I just needed to check my ESP-12E nodemcu to see whether sketch uploading is working. And the built in LED doesnt work after upgrading firmware (unless when I am uploading something, then the built in LED blinks as of TX-RX) . However there are no other problems with codes so I didnt waste any more time on blinking the on board LED.

AhmedTanvir8, 2018-01-02:
@pjparsons : If your ESP doesnt have any built in usb-serial communication hardware, it will require an external one to connect to ur computers USB, such as CP2102 usb-ttl converter or RS232 FTDI adaptor. But as u already have one built-in u can directly connect a usb cable to it which shows as a direct com port. The external serial converters are like those COM ports as well, for boards without USB

d-a-v, 2018-01-08:
The LED menu is now in latest git, can you please try it and report ?

raja-supermario, 2018-04-26:
#define LED_BUILTIN 1
try this

devyte, 2018-05-29:
@d-a-v is there anything pending here? can this be closed?

manhood202, 2019-01-27:
Hi everyone;

When I upload blink example, it works.

But when I try to define pin(ex: #define LED_BUILTIN 4) D2 it doesn't give output voltage.

I can't understand and fix the problem. I tried to define D1, the result was the same, nothing changed.

Has anyone face with same problem? Or do you have any suggestions?

Thanks,

raja-supermario, 2019-01-30:
> If you select NodeMCU v.1, you don't need to define LED_BUILTIN it's
> defined inside.
>
> And the pin led are GPIO16 -- D0. Read this:
> https://github.com/esp8266/Arduino/blob/master/doc/boards.rst#nodemcu-10-esp-12e-module

i need your help can you pls help me

anastasovPetar, 2019-11-22:
> #define LED_BUILTIN 1
> try this
> 
> add this to line 11 to solve the problem
> i have solve this problem by this way

Hello,
I can confirm that it works also on ESP-01.

NieVystrel, 2020-06-12:
> #define LED_BUILTIN 1
> try this
> 
> add this to line 11 to solve the problem
> i have solve this problem by this way

@raja-supermario when you visit Mexico ill give you a Free Tour in the city, around 14 hours of twitching configuration in IDE, flashing, making auto reset circuits, and cursing the electronic shop for the lowest prices they offer, etc...my back hurts so much but today you gave it a free day (and a quiet night). With love from Mexico <3 in the near future i giving a chance to the simplest solutions (hope will find them at the begining hehehe)