esp8266/Arduino #2193: Regression: WiFi.begin is never gets connected

di9it, 2016-06-26:
### Basic Infos
WiFi.begin related code is working fine on 2.2 but doesn't work on 2.3. Regression?
See example below

#### Hardware
Hardware:		ESP-01
Core Version:      	2.2.0

### Description

======================================
l.d(String("Setting SSID as \"") + ssid + "\" and PWD as " + secrethidden);
WiFi.softAPdisconnect(true);
delay(100);
WiFi.begin(ssid, secret);

int attempts = 0; 
int waitconnect = 1000;
while (WiFi.status() != WL_CONNECTED) {
  l.i(String("WiFi.status is ") + WiFi.status() + "...");
  delay(waitconnect);
  if (attempts++ * waitconnect == 8000 ) { // 8 seconds
    break;
  }
}

if (WiFi.status() != WL_CONNECTED) {
  l.e(String("Cannot connect to specified WiFi, status is ") + WiFi.status());
  return false;
}
======================================

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL

igrr, 2016-06-26:
Possible duplicate of #2186?
Please check if the workaround in https://github.com/esp8266/Arduino/issues/2186#issuecomment-228581052 helps.

igrr, 2016-07-04:
No response, closing.

closed 2016-07-04 by igrr

ThermalEnergy, 2020-08-30:
In order to connect to Wifi from ESP32 and other boards, the WiFi connection needs to be 2.4GHz, wireless mode must be legacy. It cannot be set to Auto or N only. Go to your router box web page, and change this setting under Advanced Settings > Wireless.