esp8266/Arduino #2196: no support for external flash?

fredericplante, 2016-06-26:
It's very nice to have SPIFFS library to access the "internal flash" but it would be nice to have support external flash chip also.

igrr, 2016-06-26:
You mean this library? https://github.com/PaulStoffregen/SerialFlash

fredericplante, 2016-06-26:
Yeah there is this guy, how ever, it's not compatible with the ESP.

I can send you all the error message, if you want.

fredericplante, 2016-06-26:
Using the CopyfromSD example

> Arduino: 1.6.5 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

CopyFromSD.ino: In function 'String gestion_commande()':
CopyFromSD:198: error: no matching function for call to 'SerialFlashChip::readdir(char [15], unsigned int, long unsigned int&)'
CopyFromSD.ino:198:69: note: candidate is:
In file included from CopyFromSD.ino:1:0:
C:\Documents and Settings\gust\My Documents\Arduino\libraries\SerialFlash/SerialFlash.h:67:14: note: static bool SerialFlashChip::readdir(char*, uint32_t, uint32_t&)
  static bool readdir(char *filename, uint32_t strsize, uint32_t &filesize);
              ^
C:\Documents and Settings\gust\My Documents\Arduino\libraries\SerialFlash/SerialFlash.h:67:14: note:   no known conversion for argument 3 from 'long unsigned int' to 'uint32_t& {aka unsigned int&}'

no matching function for call to 'SerialFlashChip::readdir(char [15], unsigned int, long unsigned int&)'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File`Preferences.` >

igrr, 2016-06-26:
1st error — change type of rdlen from `unsigned long` to `unsigned` or `uint32_t`. This is a bug in the example, function argument is declared as `uint32_t` and this is not the same as `unsigned long` (size is the same, types are different).

2nd error — add one line to esp8266/libraries/SPI/SPI.h:
```c++
void transfer(void *_buf, size_t count) { writeBytes((uint8_t*) _buf, (size_t) count); }
```

fredericplante, 2016-06-26:
It did the trick. Thank you. :)
```
Sketch uses 236,941 bytes (54%) of program storage space. Maximum is 434,160 bytes.
Global variables use 32,712 bytes (39%) of dynamic memory, leaving 49,208 bytes for local variables. Maximum is 81,920 bytes.

```

fredericplante, 2016-06-26:
I don't want to be annoying :D But the lisfile example return this:

ListFiles.ino: In function 'void setup()':
ListFiles:31: error: no matching function for call to 'SerialFlashChip::readdir(char [64], unsigned int, long unsigned int&)'
ListFiles.ino:31:65: note: candidate is:
In file included from ListFiles.ino:1:0:
C:\Documents and Settings\gust\My Documents\Arduino\libraries\SerialFlash/SerialFlash.h:67:14: note: static bool SerialFlashChip::readdir(char*, uint32_t, uint32_t&)
  static bool readdir(char *filename, uint32_t strsize, uint32_t &filesize);
              ^
C:\Documents and Settings\gust\My Documents\Arduino\libraries\SerialFlash/SerialFlash.h:67:14: note:   no known conversion for argument 3 from 'long unsigned int' to 'uint32_t& {aka unsigned int&}'

  I guess this library will be fully compatible once this is settle. Except for the MP3 part, but hey!

fredericplante, 2016-06-26:
Wait, this is kinda fun, I will post the fix. :D

fredericplante, 2016-06-26:
Forget about it, All is good, I updated your esp platform, and I forgot to fix the SPI.h thing, so it made problem in the other example listfile thing..

Thank you!

closed 2016-06-26 by fredericplante