esp8266/Arduino #2197: HTTPS request fails if no fingerprint is provided

lopify, 2016-06-27:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 Mini ESP-8266
Core Version:      	2.3

### Description

http.begin("https://www.google.com") 

HTTPS request fails if no fingerprint is provided. On version 2.1 I was able to do the request without providing a fingerprint, can we go back to that?

Thanks!

igrr, 2016-06-27:
Old behaviour was a bug in HTTPClient, it was fixed in 81d3bb3.
There is a feature request for certificate chain verification, #1851. Once this is implemented, there will be an option to add an authority certificate instead of a fingerprint.

closed 2016-06-27 by igrr

lopify, 2016-06-27:
Hi,

I understand that it was a Bug, but it was working fine. What I need is an SSL connection without verifying the certificate, isn't that posible? 

Thanks!

retsifp, 2016-06-28:
A SSL connection without verifying the certificate (in any way) just doesn't make much sense. Anyone could easily play "man in the middle" and you'll never notice it. So it's not _much_ better than plain HTTP.
For now, we only have the SHA1 verification. We'll have to wait for better methods, e.g. SHA256 or the mentioned #1851.

lopify, 2016-06-29:
@igrr @retsifp 

I need to connect to an API on a server that I don't control.

That server requires all connections to go by SSL, so I'm forced to make an SSL connection.

I don't control the server, so I can't use the server fingerprint here since the server can change the SSL cert at anytime.

My only option is to create an SSL connection without verifying the certificate, I know it's insecure but at the moment I don't see any other solution.

I checked and its a common practice between networking libraries that the allow you to decide if you want to verify the certificate or not.

What can I do? 

Thank you very much guys!

retsifp, 2016-06-29:
Okay, I got your point. Since I use my own servers, that's not a problem for me.
I didn't look much into this, but I think it should be possible for you to "hack" it for your personal needs, just until #1851 is implemented.
I didn't try it, but I think a good starting point would be reverting the commit mentioned above. :wink:

(Locate the sources: Open Arduino IDE -> Preferences, enable the verbose output during compiling; There will be a lot of output if you compile now, but there should be the path of the sources in it. For me (on Linux), it's `/home/thomas/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266HTTPClient/src`, should be quite similar on other platforms (maybe `C:\Users\username\AppData\...`))