esp8266/Arduino #2198: ESP8266WebServer Performs URL Decoding on JSON body

liquidfalcon, 2016-06-27:
Regardless of whether the body is set as plain or application/json, _parseArguments will always perform URL decoding on it, leading to unexpected results. 

Example JSON sent:
{ 
  "ssid" : "My+Network",
  "psk" :"special+Password!"
}

After receiving it through ESP8266WebServer
{ 
  "ssid" : "My Network",
  "psk" :"special Password!"
}

Currently it makes it so no JSON can include a plus sign, or any other symbol the URL decoder will pick up.

igrr, 2016-06-30:
Need to check this together with #1989.

me-no-dev, 2016-06-30:
@liquidfalcon issue should be fixed now. Can you please confirm?

liquidfalcon, 2016-07-06:
So, the commits fix the first issue, which was with the plus symbol, but, it still picks up on others, such as the Ampersand, and cuts the payload after that. 

For example, here is what I send:
`{ "ssid":"SSID+&(!@","psk":"@#$%^&+" }`

And this is what I receive for plain
`{ "ssid":"SSID+`

me-no-dev, 2016-07-06:
can you please paste the full request that you are sending so I can test it here?

liquidfalcon, 2016-07-06:
Of course - Here's the raw request I tested with using cURL:

```
POST /config HTTP/1.1
Host: linez-43.local
User-Agent: curl/7.47.1
Accept: */*
Content-Type: application/json
Content-Length: 38

{ "ssid":"SSID+&(!@","psk":"@#$%^&+" }

```

Command: ` curl -X POST -d '{ "ssid":"SSID+&(!@","psk":"@#$%^&+" }' -H "Content-Type: application/json" linez-43.local/config  -v`

me-no-dev, 2016-07-06:
great! will do

me-no-dev, 2016-07-06:
btw totally different issues :) first was urldecode and here it's trying to parse parameters, which it should not

liquidfalcon, 2016-07-07:
You're a legend mate, thank you!

closed 2018-01-02 by igrr