esp8266/Arduino #2200: ESP8266WiFiScan does not terminate a 32 character long SSID

liquidfalcon, 2016-06-27:
ESP8266WiFiScan's `ssid(uint8_t i)` function assumes `bss_info`'s ssid member is null terminated with the following statement: 

` return String(reinterpret_cast<const char*>(it->ssid));`

Now, this works fine for any SSID less than 32 characters, as bss_info will have at least one null byte at the end. In the event that an SSID is exactly 32 characters, `bss_info->ssid` will not be null terminated, and the above line will include garbage in the String as it tries to find a null byte beyond the member. 

A memcpy of 32 bytes like below guarantees at most 32 characters are read and correctly terminated, and seems to work fine during my testing.

```
char ssidBuffer[33] = {0};
memcpy(&ssidBufer[0], it->ssid, 32); 
return String(ssidBuffer);
```

devyte, 2017-10-16:
@liquidfalcon the SDK uses `uint8 ssid[32]` for storing the SSIDs, and this is pretty much across the board, not just for the scanning. You mentioned testing. Do you mean that 32-char SSIDs work correctly? If so, then the implication is that the SDK doesn't use null-terminated strings to represent SSIDs. This is an important bit of information, as your code may be needed in all places where the ssids are obtained, including the station and softap ones.
Then, there is also the 64-char password.

devyte, 2018-01-08:
@liquidfalcon can you confirm that SSIDs with 32 chars work work correctly with your change?