esp8266/Arduino #2206: Firmata support for ESP8266

bkrajendra, 2016-06-30:
### Basic Infos
My board (ESP-7) after flashing only show blue led ON..
Does not connect to AP. 
I'm using StandardFirmataWiFi.ino
Also I'm using all latest libs (ESP8266 arduino 2.3.0-rc2 and firmata 2.5.3)
Any Ides.. what could be the cause.?

#### Hardware
Hardware:			ESP-12/ESP-7
Core Version:      	2.3.0-rc2

### Description

after flashing only blue led stays ON. Nothing happens.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp
Trying
StandardFirmataWiFi.ino
or also
ConfigurableFirmataWiFi.ino
```

### Debug Messages

```
ets Jan 8 2013,rst cause:2, boot mode:(3,7)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

igrr, 2016-06-30:
Works fine here.
Do you have `#define SERIAL_DEBUG` uncommented on line 85 of StandardFirmataWiFi?
If not, please uncomment and post debug output again. Baud rate is 9600.

igrr, 2016-07-05:
No response, closing. Feel free to reopen if the issue persists and you have debug output captured with `#define SERIAL_DEBUG`.

closed 2016-07-05 by igrr

ivancho523, 2016-08-13:
Hi, i got the same problem; when i try to connect my ESP like client, the blue led blink and the serial monitor show me the next message:

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

Exception (0):
epc1=0x4022f0c0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffff7d0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffff970:  4022f0c0 00000138 00000160 4022f0a9  
3ffff980:  00000000 00000000 00000000 ffffffff  
3ffff990:  4022f158 00000023 00000000 3ffee43a  
3ffff9a0:  00000138 00000000 00000002 40229be9  
3ffff9b0:  24373b30 4022ba20 00000006 00000000  
3ffff9c0:  3ffee43a 00000001 00000001 00000002  
3ffff9d0:  00000001 3ffee3f8 3ffe82f0 40229be9  
3ffff9e0:  400061cc 00000004 00000000 4022bf8c  
3ffff9f0:  00000000 3ffee3f8 00000006 00000001  
3ffffa00:  00000000 19161310 00000000 00000000  
3ffffa10:  00002437 3ffee428 4022d7bc 3ffee3f8  
3ffffa20:  00000ff4 4022d7d1 3ffee3f8 3ffee3f8  
3ffffa30:  00000000 3ffee4a4 00000001 00000000  
3ffffa40:  3ffee3f8 60000600 60000200 3ffffa70  
3ffffa50:  40229407 00000000 3ffee3f8 00000000  
3ffffa60:  40106102 60000e00 00000008 3fffff10  
3ffffa70:  4022e56e 3ffffa80 00000008 4021db0e  
3ffffa80:  00000000 00000000 00000005 6a600194  
3ffffa90:  00000000 b20005e6 00006ff8 3ffeffdc  
3ffffaa0:  00000000 40201f8d 3ffeffdc 3ffeffdc  
3ffffab0:  3ffedf6e 402329c0 00000001 00000000  
3ffffac0:  00000001 3ffeffdc 000002f4 0007c000  
3ffffad0:  40105866 0000009f 3ffedf6e 4021a697  
3ffffae0:  4021ad06 60000e00 3ffeffdc 4021aceb  
3ffffaf0:  401050ac 0000007d 0000009f 60000200  
3ffffb00:  3fffff10 001440e0 5b496873 688efeb0  
3ffffb10:  31524556 00000000 ffffffff ffffffff  
3ffffb20:  ffffffff ffffffff 00000000 00000000  
3ffffb30:  ffffffff ffffffff ffffffff ffffffff  
3ffffb40:  ffffffff ffffffff ffffffff ffffffff  
3ffffb50:  ffffffff ffffffff ffffffff ffffffff  
3ffffb60:  00000000 ffffffff ffffffff ffffffff  
3ffffb70:  ffffffff ffffffff ffffffff ffffffff  
3ffffb80:  ffffffff ffffffff ffffffff ffffffff  
3ffffb90:  ffffffff ffffffff ffffffff ffffffff  
3ffffba0:  ffffffff ffffffff ffffffff ffffffff  
3ffffbb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffbc0:  ffffffff ffffffff ffffffff ffffffff  
3ffffbd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffbe0:  ffffffff ffffffff ffffffff ffffffff  
3ffffbf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffc00:  ffffffff ffffffff ffffffff ffffffff  
3ffffc10:  ffffffff ffffffff ffffffff ffffffff  
3ffffc20:  ffffffff ffffffff ffffffff ffffffff  
3ffffc30:  ffffffff ffffffff ffffffff ffffffff  
3ffffc40:  ffffffff ffffffff ffffffff ffffffff  
3ffffc50:  ffffffff ffffffff ffffffff ffffffff  
3ffffc60:  ffffffff ffffffff ffffffff ffffffff  
3ffffc70:  ffffffff ffffffff ffffffff ffffffff  
3ffffc80:  ffffffff ffffffff ffffffff ffffffff  
3ffffc90:  ffffffff ffffffff ffffffff ffffffff  
3ffffca0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcc0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffce0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffd00:  ffffffff ffffffff ffffffff ffffffff  
3ffffd10:  ffffffff ffffffff ffffffff ffffffff  
3ffffd20:  ffffffff ffffffff ffffffff ffffffff  
3ffffd30:  ffffffff ffffffff ffffffff ffffffff  
3ffffd40:  ffffffff ffffffff ffffffff ffffffff  
3ffffd50:  ffffffff ffffffff ffffffff ffffffff  
3ffffd60:  ffffffff ffffffff ffffffff ffffffff  
3ffffd70:  ffffffff ffffffff ffffffff ffffffff  
3ffffd80:  ffffffff ffffffff ffffffff ffffffff  
3ffffd90:  ffffffff ffffffff ffffffff ffffffff  
3ffffda0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdc0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffde0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffe00:  ffffffff ffffffff ffffffff ffffffff  
3ffffe10:  ffffffff ffffffff ffffffff ffffffff  
3ffffe20:  ffffffff ffffffff ffffffff ffffffff  
3ffffe30:  ffffffff ffffffff ffffffff ffffffff  
3ffffe40:  ffffffff ffffffff ffffffff ffffffff  
3ffffe50:  ffffffff ffffffff ffffffff ffffffff  
3ffffe60:  ffffffff ffffffff ffffffff ffffffff  
3ffffe70:  ffffffff ffffffff ffffffff ffffffff  
3ffffe80:  ffffffff ffffffff ffffffff ffffffff  
3ffffe90:  ffffffff ffffffff ffffffff ffffffff  
3ffffea0:  ffffffff ffffffff ffffffff ffffffff  
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffec0:  ffffffff ffffffff ffffffff ffffffff  
3ffffed0:  ffffffff ffffffff ffffffff ffffffff  
3ffffee0:  ffffffff ffffffff ffffffff ffffffff  
3ffffef0:  ffffffff ffffffff ffffffff ffffffff  
3fffff00:  ffffffff ffffffff ffffffff ffffffff  
3fffff10:  ffffffff ffffffff ffffffff ffffffff  
3fffff20:  ffffffff ffffffff ffffffff ffffffff  
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff ffffffff ffffffff ffffffff  
3fffff60:  ffffffff ffffffff ffffffff ffffffff  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff ffffffff ffffffff ffffffff  
3fffff90:  ffffffff a4ffc537 2664925d ccfbdb0c  
3fffffa0:  000201e9 f9a163ae ffffff00 55aa55aa  
<<<stack<<<

why can i do to solve this problem??

thank you very much