esp8266/Arduino #2211: An ESP8266mDNS related build failure

jjssoftware, 2016-06-30:
I hope someone can make sense of this :) 

I develop in Eclipse with the Arduino Eclipse extension, I've updated from v2.1.0 -> v2.3.0 and code that did build previously no longer builds. 

There seems to be some problem with how ARDUINO_BOARD is being defined. I think I've worked out that this is related to what's defined in boards.txt and I'm trying to use the "Generic ESP8266 Module" profile for an ESP12F board which means the folllowing is defined in platform.txt for this profile: generic.build.board=ESP8266_ESP01

I've created a new bare bones new project bringing in the ESP8266mDNS library code and the build problem exists there too so it seems like this is an environmental problem.

This is the actual build error I see:
`/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp: In member function 'void MDNSResponder::enableArduino(uint16_t, bool)':
<command-line>:0:15: error: 'ESP8266_ESP01' was not declared in this scope
/home/joe/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:841:44: note: in expansion of macro 'ARDUINO_BOARD'
   addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);`

Does anyone have a clue what's wrong here?

Thanks
Joe

jjssoftware, 2016-06-30:
A bit more info: ARDUINO_BOARD is #defined as a fallback in ESP8266mDNS.h here: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/ESP8266mDNS.h#L49-#L52

If I comment out the #ifndef / #endif lines here leaving ARDUINO_BOARD hard defined as "generic", code then builds fine which tells me that under normal circumstances ARDUINO_BOARD is already defined as ESP8266_ESP01 and this is a symbol not recognized at build time..I expect that's because ESP8266_ESP01 is not recognized as a string literal like "generic"

me-no-dev, 2016-06-30:
that is defined at build time in platform.txt [here](https://github.com/esp8266/Arduino/blob/master/platform.txt#L73-L83) And you should maybe file a report to the maintainer of the plugin about this problem.

jjssoftware, 2016-07-01:
Thanks for the heads up I've raised a new issue over here: https://github.com/jantje/arduino-eclipse-plugin/issues/493

jjssoftware, 2016-07-02:
After a lengthy investigation around the Eclipse Arduino plug-in code I've come to the conclusion that it's actually doing the right thing and the problem is right [here](https://github.com/esp8266/Arduino/blob/master/platform.txt#L73-L83) where I was first pointed to.

With the platform.txt recipe config as it stands right now, we end up with a generated make file that writes out the gcc -D option like so:

`-DARDUINO_BOARD="ESP8266_ESP01"`
..which on the face of it looks ok, but it's not because the gcc pre-processor ends up generating code that looks like this in the .o object file:

`addServiceTxt("arduino", "tcp", "board", ESP8266_ESP01);` 
note the missing quotes around ESP8266_ESP01 which results in the build failure

It's an easy enough fix though - we just need escaped quotes in platform.txt which results in the desired effect e.g. in platform.txt:

`-DARDUINO_BOARD=\"ESP8266_ESP01\"`
results in this compiled code in the object file and a successful build:

`addServiceTxt("arduino", "tcp", "board", "ESP8266_ESP01");` 
hurray a quoted string :)

If we agree this is the correct fix I'll create the pull request

Thanks
Joe

jantje, 2016-07-02:
Mind that from experience I know the quotes are treated differently depending on OS. So you might need the .windows .linux extensions to get it working on all oses

jantje, 2016-07-02:
I double checked. The \ is needed in windows but as the arduino IDE runs in some sort of linux simulation it is not needed there.
I assume that the current platform.txt works in mac, linux and windows with the arduino IDE, but with the arduino eclipse plugin called sloeber it wil only work on mac and linux.
Adding the \ would make the Arduino IDE to fail in all OS'es.
So I think we are back to jantje/arduino-eclipse-plugin#493

jjssoftware, 2016-07-02:
I agree good spot. 

Adding escaping backslashes `-DARDUINO_BOARD=\"{build.board}\"` into platform.txt makes it work fine for the arduino eclipse plugin on linux but having tested compiling a small project in the arduino IDE, it fails to compile on linux with this error "stray '\' in program addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);" and the resulting pre-processor generated code is broken:
`addServiceTxt("arduino", "tcp", "board", \"ESP8266_ESP01\");`

I'm afraid I can't test windows at all here but I think we know enough to say that the escaping backslashes in platform.txt is not the right fix for this problem.

jantje, 2016-07-02:
I'm a bit confused here. Does it work in the Arduino IDE "out of the box"?
Because my comments are on windows (if it words with the arduino ide it should work with the arduino eclipse plugin named sloeber)
Arduino uses the same appoach for build.usb_product
```
arduino_due_x_dbg.build.usb_product="Arduino Due"
arduino_due_x_dbg.build.extra_flags=-D__SAM3X8E__ -mthumb {build.usb_flags}
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} -DUSBCON '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mcpu={build.mcu} -mthumb -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} {compiler.libsam.c.flags} {includes} "{source_file}" -o "{object_file}"
```
giving 
`recipe.c.o.pattern=..... '-DUSB_PRODUCT="Arduino Due"' ....`

The link you refer to contains
`-DARDUINO_BOARD="{build.board}"`
but that should be
`'-DARDUINO_BOARD="{build.board}"'`
For the Arduino IDE.
implying it will not work in the Arduino IDE; unless some custom hardcoding has been done in the Arduino IDE to make this work.

jjssoftware, 2016-07-02:
It does indeed work in the Arduino IDE v1.6.8 'out of the box' and I guess the Arduino IDE is where ESP8266 arduino core maintainers generally focus their testing efforts to be happy it works for the bulk of the user / dev population.

I'm not sure where the difference lies but I can say without shadow of a doubt that we definitely have different behaviour in the toolchain between how the Arduino IDE and pre-processor interprets `-DARDUINO_BOARD=\"{build.board}\"` on linux and how the Eclipse arduino plugin interprets it. 

There are no outer quotes around `-DARDUINO_BOARD=\"{build.board}\"` as you can see in the source: [https://github.com/esp8266/Arduino/blob/master/platform.txt#L73-L83](https://github.com/esp8266/Arduino/blob/master/platform.txt#L73-L83)
but this is working in the Arduino IDE and not working in the Eclipse Arduino plugin

jantje, 2016-07-02:
>I'm not sure where the difference lies but I can say without shadow of a doubt that we definitely have different behaviour in the toolchain between how the Arduino IDE and pre-processor interprets -DARDUINO_BOARD=\"{build.board}\" on linux and how the Eclipse arduino plugin interprets it. 

It is actually the command line that interprets differently.
My brain can't nail this behaviour. I'll try this on my windows system to see what it gives.

jjssoftware, 2016-07-02:
Ok thanks sounds good - good to hear you have a way to test windows there

jjssoftware, 2016-07-02:
Here's the minimal ESP8266 core sketch that doesn't compile with the eclipse arduino plugin and works fine in the arduino IDE:

```
#include "Arduino.h"
#include <ESP8266mDNS.h>

void setup()
{
	MDNS.addService("http", "tcp", 80);
}

void loop()
{

}
```

jantje, 2016-07-02:
My brain nailed it :-)
The problem does not exist in ESP8266 version 2.2, it is introduced in version 2.3.
The problem is indeed related to the 
`-DARDUINO_BOARD="{build.board}"`
the reason why there is no need for single quotes around it is because `{build.board}` does not contain spaces.
On the windows command line you'll need
`-DARDUINO_BOARD=\"[name without spaces]\"`
or
`-DARDUINO_BOARD="\"[name with spaces]\""`
I haven't tested the Linux command line but I guess in Linux it is
`-DARDUINO_BOARD="[name without spaces]"`
or
`'-DARDUINO_BOARD="[name with spaces]"'`
Because Arduino use "Arduino due" and the Arduino IDE is running in a Linux environment even under windows they used the single quotes with quotes solution.

My guess is that ESP8266 will move to string with spaces now this is nicely documented :-)

Anyway it is not a bug with ESP8266 and we are back to jantje/arduino-eclipse-plugin#493.

jjssoftware, 2016-07-03:
Agreed, this "problem" was introduced in v2.3.0 in https://github.com/esp8266/Arduino/pull/2054 where platform.txt was modified to include 
`-DARDUINO_BOARD="{build.board}"`
and this is not a problem at all when using the Arduino IDE.

{build.board} is used in a number of places in platform.txt and introducing spaces into the board string causes other build issues related to a broken command line so I can't see that happening any time soon

jjssoftware, 2016-07-03:
Confirmed as a bug in the eclipse arduino plug tracked on issue https://github.com/jantje/arduino-eclipse-plugin/issues/493. No change is being made to the esp8266 core. This is closed.

closed 2016-07-03 by jjssoftware

jjssoftware, 2016-07-06:
Confirmed as fixed in the eclipse arduino plugin. This comment highlights the issue being fixed on linux however the plugin owner / maintainer confirms the issue is also fixed on windows:
https://github.com/jantje/arduino-eclipse-plugin/issues/493#issuecomment-230584958

kindzhon, 2020-10-22:
It's 2.4.2 or older  bug.
I found way 👍 
u can edit ESP8266mDNS.cpp  like this.
**/**
 * STRINGIZE
 */
#ifndef STRINGIZE
    #define STRINGIZE(x) #x
#endif
#ifndef STRINGIZE_VALUE_OF
    #define STRINGIZE_VALUE_OF(x) STRINGIZE(x)
#endif**

  addServiceTxt("arduino", "tcp", "board", **STRINGIZE_VALUE_OF(ARDUINO_BOARD));**