esp8266/Arduino #2214: Changing SPIFFS size on each OTA update

david476, 2016-07-02:
#### Hardware
Hardware: ESP-12f
Core Version: 2.3.0

### Description

Every time I update over OTA I need to change the SPIFFS size, otherwise I get [ERROR]: No answer

### Settings in IDE

Module: Generic ESP8266 Module?
Flash Size: 		4MB/1MB and 4MB/3MB
CPU Frequency: 80Mhz
Flash Mode: dio or qio (it doesn't seem to make a difference, what is it?)
Flash Frequency: 40Mhz
Upload Using: OTA
Reset Method:	ck

### Sketch

`#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "DEB_WIFI";
const char* password = "you'rewelcome";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  pinMode(13, OUTPUT);
}

void loop() {
  ArduinoOTA.handle();
  digitalWrite(13, HIGH);
  delay(100);
  digitalWrite(13, LOW);
  delay(100);
  digitalWrite(13, HIGH);
  delay(100);
  digitalWrite(13, LOW);
  delay(700);
}`

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
Sketch uses 243,495 bytes (23%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 32,972 bytes (40%) of dynamic memory, leaving 48,948 bytes for local variables. Maximum is 81,920 bytes.
21:30:33 [ERROR]: No Answer
```

krzychb, 2016-07-02:
Hi @david476,

- Where / how do you "need to change the SPIFFS size"? Could you provide exact steps to replicate this issue?
- Does the issue occur with original [BasicOTA.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino) example?

Krzysztof

david476, 2016-07-02:
I'm using BasicOTA.ino modified to flash pin 13. I need to alternate uploading with 4MB/1MB and 4MB/3MB flash sizes or I get the [ERROR]: No Answer message. Switching to another memory setting also fails but allows me to use the same 4MB option again. I also don't think it has to do with recompilation, as changing sketches still doesn't work unless I switch to the other 4MB option.

krzychb, 2016-07-02:
Does the issue occur with original BasicOTA.ino example (without 1000 ms delay in loop())?

david476, 2016-07-02:
Yup

david476, 2016-07-02:
I don't know if this is helpful, but installing a different version of the board between uploads lets me upload with the same settings.

krzychb, 2016-07-02:
I have just loaded couple of times your sketch using each time the same settings like below:

[image].

On external serial terminal I see each time log as follows:
[image]

I wonder what is the log that you see when upload is successful and when it fails?

david476, 2016-07-02:
After playing around, it looks like I can use the same memory settings, as long I do it twice (upload, fails, upload). I also checked, it's not a timing thing. Waiting awhile still doesn't work. Failure brings up nothing, as you would expect with the No answer response. Success:
`
Start
Progress: 100%
End

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
@cp:0
     ld
                                                                                
       Booting                                                                  
Ready                                                                           
IP address: 192.168.1.103
`

david476, 2016-07-03:
Weird development, instead of doing a failed upload between successful ones, I can also just ping it. (Again, same action whether or not I have my LED blinking code) It seems the ESP8266 just needs to get connected to by something before uploading works. I hope you aren't as miffed as me!

krzychb, 2016-07-03:
Hi @david476,

I am glad you were able to remove IDE settings from the equation and narrow down the issue to connection problem. This means you are progressing by having less variables to check :+1: 

I believe symptoms you describe may indicate issue with network or PC configuration. Having another Wi-Fi router I would set up a separate network to check it out. I would use generic router configuration without any firewall, etc. After all internet connection is not required for this test at all. I would also try another PC.

As a side note, I would like to clarify the boot mode you are getting ``` rst cause:2```, that indicate you were doing [hardware reset](https://github.com/esp8266/Arduino/blob/master/doc/boards.md#rst-cause) of your board. Normally, except for the [reset after initial upload with serial](https://github.com/esp8266/Arduino/issues/1017#issuecomment-200824765), you do not need to hard reset your ESP for any OTA uploads at all. Below is a log of several subsequent uploads I have made without any retries, pinging, etc. Do you need to do any hard resets to make the OTA working?

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
@cp:0
ld
øBooting
Ready
IP address: 192.168.1.120
Start
(...)
End

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
@cp:0
ld
øBooting
Ready
IP address: 192.168.1.120
Start
(...)
End

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
@cp:0
ld
üBooting
Ready
IP address: 192.168.1.120
Start
(...)
End

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
@cp:0
ld
üBooting
Ready
IP address: 192.168.1.120
```

david476, 2016-07-03:
Well, problem isolated. It does work fine on another wifi network. I'll update my original post with a more detailed explanation of the real problem and solution. Any ideas for ways to make it work continuously on my original network?

mkeyno, 2016-10-20:
hi @krzychb I faced same result from BasicOTA.ino sketch successful OTA update , I did hard reset after each serial and OTA uploading but  the following message continue  to  come out after successful OTA update, and I have to program my module with serial again
hardware setting 
 ESP-07  set  1M(144KB SPIFF) 
FlashChipRealSize: 1.00MB
SketchSize:274.16KB
FreeSketchSpace:584.00KB
FreeHeap:42.66KB
ChipId: 14306800
CpuFreqMHz: 80
SdkVersion: 1.5.4(baaeaebb)
FlashChipSpeed: 40000000
` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
`

thomasvnl, 2017-01-18:
Having a similar problem, in my case I had to solve it by formatting the SPIFFS. After this, ArduinoOTA worked again. 

This makes me wonder if ArduinoOTA leaves temporary flash files when in the SPIFFS which slowly fills it up unless you format the SPIFFS yourself or change the size (which basically formats it too I assume?).

davisonja, 2017-01-18:
No, Arduino OTA stores it's files prior to SPIFFS (in flash) so shouldn't
have any impact on SPIFFS at all.

thomasvnl, 2017-01-18:
Okay, well I might be using the wrong terminology but I have linked it to the "Flash size" in the menu, dividing it in 4M with 1M SPIFFS or 3M SPIFFS. Changing these settings or formatting the SPIFFS resolves the issue for me at least. 

Just giving my two cents in the hope that it may help someone else.

david476, 2017-01-20:
If yours is an actual problem with the core you should open a new issue. To answer on the new issue: was that a or tube fix or do you have to keep doing it?

david476, 2017-01-20:
*one time

devyte, 2017-10-16:
@david476 I can't reproduce this on any of my ESP12 boards.
Is the issue still valid? with latest git?

david476, 2017-10-16:
@devyte If your referring to my original issue, I'm pretty sure it was just an issue with my WiFi router, not the core. As for @thomasvnl 's I don't know.

devyte, 2017-10-16:
In that case, I am closing this one.

closed 2017-10-16 by devyte