esp8266/Arduino #2215: new PROGMEM macros missing

Makuna, 2016-07-02:
Arduino/Atmel expanded the PROGMEM macros in the last year; and the added ones are missing from the Esp8266 progmem support.

pgm_read_ptr() is a known one, but there are more than likely others

These should be reviewed and brought up to date.

devyte, 2017-10-16:
pgm_read_ptr() seems to be already in.

earlephilhower, 2018-08-04:
We have all of the macros defined in AVR `pgmspace.h` already, just verified via grep:
````
earle@server:~/.arduino15$ grep "define pgm_read" ./packages/arduino/tools/avr-gcc/4.8.1-arduino5/avr/include/avr/pgmspace.h
#define pgm_read_byte_near(address_short) __LPM((uint16_t)(address_short))
#define pgm_read_word_near(address_short) __LPM_word((uint16_t)(address_short))
#define pgm_read_dword_near(address_short) \
#define pgm_read_float_near(address_short) \
#define pgm_read_ptr_near(address_short) \
#define pgm_read_byte_far(address_long)  __ELPM((uint32_t)(address_long))
#define pgm_read_word_far(address_long)  __ELPM_word((uint32_t)(address_long))
#define pgm_read_dword_far(address_long) __ELPM_dword((uint32_t)(address_long))
#define pgm_read_float_far(address_long) __ELPM_float((uint32_t)(address_long))
#define pgm_read_ptr_far(address_long) (void*)__ELPM_word((uint32_t)(address_long))
#define pgm_read_byte(address_short)    pgm_read_byte_near(address_short)
#define pgm_read_word(address_short)    pgm_read_word_near(address_short)
#define pgm_read_dword(address_short)   pgm_read_dword_near(address_short)
#define pgm_read_float(address_short)   pgm_read_float_near(address_short)
#define pgm_read_ptr(address_short)     pgm_read_ptr_near(address_short)
````

````
earle@server:~/Arduino/hardware/esp8266com/esp8266/cores/esp8266$ grep "define pgm_re" -r
pgmspace.h:#define pgm_read_with_offset(addr, res) \
pgmspace.h:#define pgm_read_byte(addr) pgm_read_byte_inlined((const void*)(addr))
pgmspace.h:#define pgm_read_word(addr) pgm_read_word_inlined((const void*)(addr))
pgmspace.h:#define pgm_read_byte(addr) pgm_read_byte_inlined(addr)
pgmspace.h:#define pgm_read_word(addr) pgm_read_word_inlined(addr)
pgmspace.h:#define pgm_read_byte(addr)     (*reinterpret_cast<const uint8_t*>(addr))
pgmspace.h:#define pgm_read_word(addr)     (*reinterpret_cast<const uint16_t*>(addr))
pgmspace.h:#define pgm_read_dword(addr) 		(*reinterpret_cast<const uint32_t*>(addr))
pgmspace.h:#define pgm_read_float(addr) 		(*reinterpret_cast<const float*>(addr))
pgmspace.h:#define pgm_read_ptr(addr) 		(*reinterpret_cast<const void* const *>(addr))
pgmspace.h:#define pgm_read_byte_near(addr) 	pgm_read_byte(addr)
pgmspace.h:#define pgm_read_word_near(addr) 	pgm_read_word(addr)
pgmspace.h:#define pgm_read_dword_near(addr) 	pgm_read_dword(addr)
pgmspace.h:#define pgm_read_float_near(addr)	pgm_read_float(addr)
pgmspace.h:#define pgm_read_ptr_near(addr) 	pgm_read_ptr(addr)
pgmspace.h:#define pgm_read_byte_far(addr) 	pgm_read_byte(addr)
pgmspace.h:#define pgm_read_word_far(addr) 	pgm_read_word(addr)
pgmspace.h:#define pgm_read_dword_far(addr) 	pgm_read_dword(addr)
pgmspace.h:#define pgm_read_float_far(addr) 	pgm_read_float(addr)
````

Closing as fixed already.

closed 2018-08-04 by earlephilhower