esp8266/Arduino #2217: EEPROMClass _dirty

pgScorpio, 2016-07-03:
#### Hardware
Hardware:		n.a.
Core Version:      	2.2.0

### Description

1:
There is no way to undo/revert any changes after a begin.
Currently the only way would be to call begin again, but... See 2:

I suggest adding:
```
bool EEPROMClass::revert() {
  if (_data && _size)
  {
    noInterrupts();
    spi_flash_read(_sector * SPI_FLASH_SEC_SIZE, reinterpret_cast<uint32_t*>(_data), _size);
    _dirty = false;
    interrupts();
    return true;
  }

  return false;
}

```
2: 
`void EEPROMClass::begin(...)` Does read _data from eeprom but does not clear the _dirty flag.
Also it would be handy if `EEPROMClass::begin(...)` would return the _data pointer.

I suggest:
```
uint8_t * EEPROMClass::begin(size_t size /* =  SPI_FLASH_SEC_SIZE */) {
  if (_data) {
    delete[] _data;
    _data = NULL;
  }
  _size = 0;
  _dirty = false;

  if ( (size > 0) && (size <= SPI_FLASH_SEC_SIZE) ) {
     size = (size + 3) & (~3); 
    _data = new uint8_t[size];
    if ( _data)  {
      _size = size;
      noInterrupts();
      spi_flash_read(_sector * SPI_FLASH_SEC_SIZE, reinterpret_cast<uint32_t*>(_data), _size);
      interrupts();
    }
  }
  return &_data[0];
}

```
3:
`EEPROMClass::getDataPtr()` Does always set _dirty. This is not wanted if we are using the pointer only for reading....

I suggest: 
```
uint8_t * EEPROMClass::getDataPtr(bool forWrite /* = true */) {
  if (forWrite) _dirty = true;
  return &_data[0];
}
```
with the default forWrite = true `getDataPtr()` would still be compatible with the current version but when just reading `getDataPtr(false)` would not set _dirty to true

devyte, 2017-10-16:
For read-only of _data, a method that returns a const ptr could be implemented, which doesn't set _dirty.

devyte, 2017-11-19:
@pgScorpio Please test PR #3849 .
EEPROM::begin() now clears dirty flag
EEPROM::getConstDataPtr() returns a pointer to const data for read-only, and it doesn't set dirty.

closed 2017-11-20 by devyte

dunk8888, 2018-01-15:
Hi,what does this EEPROM.getDataPtr(); actualy do? What is happening bellow,i want to port a sketch to Arduino uno and the uno eeprom library doesnt support this EEPROM.getDataPtr();.

All i know its a pointer but why is it needed.thanks

EEPROM.begin(EEPROM_SIZE);
memcpy(&ref,EEPROM.getDataPtr()+EEPROM_PATTERNS_START+1+patternIndex*sizeof(PatternReference),sizeof(PatternReference));
        EEPROM.end();

devyte, 2018-01-15:
In our case EEPROM is emulated on flash. To avoid rewriting on every byte, there is a buffer kept in memory, so when you write to an index in the eeprom, you really just write into the buffer. Then when you call commit, the buffer gets written to flash all at once.
The getPtr() method gets the underying pointer to the mem buffer. It's an esp-specific method.

PR #3849 (2017-11-20): EEPROM: dirty flag, reallocate only if necessary, add getConstDataPtr method (#2217)
Fixes #2217 .